espressif/esp-iot-solution #4: programming esp-01

fastlink30, 2018-06-17:
i want to use esp-prog to program esp-01

connection are:

esp-prog     esp-01

TX                 rx

RX                 tx

I00               i00

EN               RST (tryed also on CH_PD)

GND            GND

                    +V (external source)

usb connected

the blue light flash when i start to program

i try with program like esp8266flasher, inside visual studio (with visualmicro),ESPFlashDownloadTool_v3.6.4

when the green light flash, also on the esp01 flash the blue light

 

noone of this can program this, what is the trick?

thanks

xiongyumail, 2018-06-27:
@fastlink30 
Please try to exchange TX/RX:  
TX / tx  
RX / rx

xiongyumail, 2018-06-27:
@fastlink30   
Do the other modules be the same? We have tested several boards, which can be downloaded normally. You need to pay attention to TX-tx, RX-rx and GPIO2 pulldown(GPIO2 low level when dowload.).

fastlink30, 2018-06-28:
i tryed also with gpio2 to gnd but no luck.
to be clear the module (esp-01) is wired directly to esp-prog (is wrong? i want only try to program)

xiongyumail, 2018-06-29:
@fastlink30 
Is the power supply stable?You can try other serial debuggers！

fastlink30, 2018-07-01:
power is stable 3.3v  (i use +v from esp-prog), i noted that part near led become little warm, i s normal? 
i not have other serial debugger
maybe my esp-prog have hardware problem?

xiongyumail, 2018-07-03:
@fastlink30 
Can you give you a picture of your connection?

xiongyumail, 2018-07-09:
@fastlink30
I found a very good blog about esp-prog.
[esp-prog](https://blog.csdn.net/espressif/article/details/79359108)  (Chinese)

closed 2018-10-15 by costaud

fastlink30, 2018-11-17:
i've restarted trying program esp-01 with esp- prog
connection:
esp-prog    esp-01
tx               rx
rx               tx
vprog        +v
gnd           gnd
i0              i0
en             chpd

are these connections wrong? reset pin on esp8266 must be unconnected? (tryed to tie +v & gnd but nothing change)
to program i use esp download tool 3.6.4 or nodemcu esp8266flasher

Gavitron, 2019-12-06:
I had this same problem, I connected the 6 wires from esp-prog -> ESP01, as you did:
```
esp-prog -> esp-01
VPROG    -> VCC
GND      -> GND
TX       -> RXD
RX       -> TXD
IO0      -> IO0
EN       -> RESET
```

finally i got it to work, what I had to do is when esptool starts outputting the connect string like:
`Connecting........_____........_____........_____`
 
you need to *only then* manually connect
```
CH_EN    -> VCC
```
and keep it connected until programming finishes.