espressif/esp-iot-solution #6: Can't find DataScope application

webstdave, 2018-06-28:
Touch sensor readme refers to "DataScope tool can be downloaded from Espressif's official website." but I cannot find it. Could someone provide a link, please?  Thank you!

NareshDealer, 2018-07-02:
i m looking for same.. anyone has a downloaded copy.?

Cojocari, 2018-07-26:
Have you managed to find that tool ? And if not. how did you calibrated sense kit? Thank you.

InfiniteYuan1, 2018-10-09:
[ESP-Tuning Tool](http://download.espressif.com/application_release/ESP-Tuning_Tool.zip)
[esp_tuning_tool_user_guide_en.md](https://github.com/espressif/esp-iot-solution/blob/master/documents/touch_pad_solution/esp_tuning_tool_user_guide_en.md)
[ESP-Tuning Tool Code](https://github.com/espressif/esp-iot-solution/tree/master/components/features/touchpad)

closed 2018-10-15 by costaud