espressif/esp-iot-solution #11: Code not working with LAN8710A.

BHAKTIDESHPANDE, 2018-10-01:
Hello ,

I am trying to use Ethernet to WiFi data forwarding demo code on ESP32-EVB Rev D on windows 10 with msys32 and  I have added header file and C source file of LAN8710A in Ethernet components of ESP IDF, code has built successfully , but its not working, device is rebooting continuously. 
[image]

Please Help.
Thanks.

BHAKTIDESHPANDE, 2018-11-14:
Thank you for your reply. I have already resolved the issue. It was related to library path.

Xiehanxin, 2020-07-27:
@BHAKTIDESHPANDE  If your issue is now resolved, can you please close the issue. Thank you.

closed 2023-07-10 by leeebo