espressif/esp-iot-solution #15: error: wrong return type on i2s_lcd_create

nicola-lunghi, 2018-10-15:
This function is returning ESP_ERR_INVALID_ARG when it should return a pointer (NULL)

https://github.com/espressif/esp-iot-solution/blob/3cf265989d6eb3a6fc069d716a46a24da795b337/components/i2s_devices/lcd_common/i2s_lcd.c#L292-L294

also here

https://github.com/espressif/esp-iot-solution/blob/3cf265989d6eb3a6fc069d716a46a24da795b337/components/i2s_devices/lcd_common/i2s_lcd.c#L313-L315

This cause the following warning

```
/home/user/esp-iot-solution/components/i2s_devices/lcd_common/i2s_lcd.c: In function 'i2s_lcd_create':
/home/user/esp-iot-solution/components/i2s_devices/lcd_common/i2s_lcd.c:31:16: warning: return makes pointer from integer without a cast [-Wint-conversion]
         return (ret);                                                                   \
                ^
/home/user/esp-iot-solution/components/i2s_devices/lcd_common/i2s_lcd.c:294:5: note: in expansion of macro 'I2S_CHECK'
     I2S_CHECK((i2s_num < I2S_NUM_MAX), "i2s_num error", ESP_ERR_INVALID_ARG);
     ^
In file included from /home/user/esp-iot-solution/submodule/esp-idf/components/esp32/include/esp_timer.h:44:0,
                 from /home/user/esp-iot-solution/submodule/esp-idf/components/freertos/include/freertos/portmacro.h:82,
                 from /home/user/esp-iot-solution/submodule/esp-idf/components/freertos/include/freertos/portable.h:94,
                 from /home/user/esp-iot-solution/submodule/esp-idf/components/freertos/include/freertos/FreeRTOS.h:105,
                 from /home/user/esp-iot-solution/components/i2s_devices/lcd_common/i2s_lcd.c:18:
/home/user/esp-iot-solution/submodule/esp-idf/components/esp32/include/esp_err.h:28:25: warning: return makes pointer from integer without a cast [-Wint-conversion]
 #define ESP_FAIL        -1      /*!< Generic esp_err_t code indicating failure */
                         ^
/home/user/esp-iot-solution/components/i2s_devices/lcd_common/i2s_lcd.c:314:16: note: in expansion of macro 'ESP_FAIL'
         return ESP_FAIL;
                ^
```

InfiniteYuan1, 2018-10-17:
Thank you for your comments, I will fix this soon.

Xiehanxin, 2020-07-27:
@nicola-lunghi  hi, If your issue is now resolved, can you please close the issue. if not, please let us know. Thank you.

closed 2023-07-10 by leeebo