espressif/esp-iot-solution #16: warning: 'taskEXIT_CRITICAL(mux)' is deprecated in ESP-IDF, consider using 'portEXIT_CRITICAL(mux)'

nicola-lunghi, 2018-10-15:
```
/home/user/esp-iot-solution/components/hmi/ugfx_gui/ugfx/src/gqueue/gqueue.c:39:13: warning: 'taskENTER_CRITICAL(mux)' is deprecated in ESP-IDF, consider using 'portENTER_CRITICAL(mux)'
   gfxSystemLock();
             ^
/home/user/esp-iot-solution/components/hmi/ugfx_gui/ugfx/src/gqueue/gqueue.c:41:13: warning: 'taskEXIT_CRITICAL(mux)' is deprecated in ESP-IDF, consider using 'portEXIT_CRITICAL(mux)'
   gfxSystemUnlock();
             ^
/home/user/esp-iot-solution/components/hmi/ugfx_gui/ugfx/src/gqueue/gqueue.c: In function 'gfxQueueASyncPut':
/home/user/esp-iot-solution/components/hmi/ugfx_gui/ugfx/src/gqueue/gqueue.c:57:13: warning: 'taskENTER_CRITICAL(mux)' is deprecated in ESP-IDF, consider using 'portENTER_CRITICAL(mux)'
   gfxSystemLock();
             ^
/home/user/esp-iot-solution/components/hmi/ugfx_gui/ugfx/src/gqueue/gqueue.c:59:13: warning: 'taskEXIT_CRITICAL(mux)' is deprecated in ESP-IDF, consider using 'portEXIT_CRITICAL(mux)'
   gfxSystemUnlock();
             ^
/home/user/esp-iot-solution/components/hmi/ugfx_gui/ugfx/src/gqueue/gqueue.c: In function 'gfxQueueASyncPush':
/home/user/esp-iot-solution/components/hmi/ugfx_gui/ugfx/src/gqueue/gqueue.c:73:13: warning: 'taskENTER_CRITICAL(mux)' is deprecated in ESP-IDF, consider using 'portENTER_CRITICAL(mux)'
   gfxSystemLock();
             ^
/home/user/esp-iot-solution/components/hmi/ugfx_gui/ugfx/src/gqueue/gqueue.c:75:13: warning: 'taskEXIT_CRITICAL(mux)' is deprecated in ESP-IDF, consider using 'portEXIT_CRITICAL(mux)'
   gfxSystemUnlock();
             ^
/home/user/esp-iot-solution/components/hmi/ugfx_gui/ugfx/src/gqueue/gqueue.c: In function 'gfxQueueASyncInsert':
/home/user/esp-iot-solution/components/hmi/ugfx_gui/ugfx/src/gqueue/gqueue.c:86:13: warning: 'taskENTER_CRITICAL(mux)' is deprecated in ESP-IDF, consider using 'portENTER_CRITICAL(mux)'
   gfxSystemLock();
             ^
/home/user/esp-iot-solution/components/hmi/ugfx_gui/ugfx/src/gqueue/gqueue.c:88:13: warning: 'taskEXIT_CRITICAL(mux)' is deprecated in ESP-IDF, consider using 'portEXIT_CRITICAL(mux)'
   gfxSystemUnlock();
             ^
/home/user/esp-iot-solution/components/hmi/ugfx_gui/ugfx/src/gqueue/gqueue.c: In function 'gfxQueueASyncRemove':
/home/user/esp-iot-solution/components/hmi/ugfx_gui/ugfx/src/gqueue/gqueue.c:110:13: warning: 'taskENTER_CRITICAL(mux)' is deprecated in ESP-IDF, consider using 'portENTER_CRITICAL(mux)'
   gfxSystemLock();
             ^
/home/user/esp-iot-solution/components/hmi/ugfx_gui/ugfx/src/gqueue/gqueue.c:112:13: warning: 'taskEXIT_CRITICAL(mux)' is deprecated in ESP-IDF, consider using 'portEXIT_CRITICAL(mux)'
   gfxSystemUnlock();
             ^
/home/user/esp-iot-solution/components/hmi/ugfx_gui/ugfx/src/gqueue/gqueue.c: In function 'gfxQueueASyncIsIn':
/home/user/esp-iot-solution/components/hmi/ugfx_gui/ugfx/src/gqueue/gqueue.c:139:13: warning: 'taskENTER_CRITICAL(mux)' is deprecated in ESP-IDF, consider using 'portENTER_CRITICAL(mux)'
   gfxSystemLock();
             ^
/home/user/esp-iot-solution/components/hmi/ugfx_gui/ugfx/src/gqueue/gqueue.c:141:13: warning: 'taskEXIT_CRITICAL(mux)' is deprecated in ESP-IDF, consider using 'portEXIT_CRITICAL(mux)'
   gfxSystemUnlock();
             ^
/home/user/esp-iot-solution/components/hmi/ugfx_gui/ugfx/src/gqueue/gqueue.c: In function 'gfxQueueGSyncGet':
/home/user/esp-iot-solution/components/hmi/ugfx_gui/ugfx/src/gqueue/gqueue.c:172:13: warning: 'taskENTER_CRITICAL(mux)' is deprecated in ESP-IDF, consider using 'portENTER_CRITICAL(mux)'
   gfxSystemLock();
             ^
/home/user/esp-iot-solution/components/hmi/ugfx_gui/ugfx/src/gqueue/gqueue.c:176:13: warning: 'taskEXIT_CRITICAL(mux)' is deprecated in ESP-IDF, consider using 'portEXIT_CRITICAL(mux)'
   gfxSystemUnlock();
             ^
/home/user/esp-iot-solution/components/hmi/ugfx_gui/ugfx/src/gqueue/gqueue.c: In function 'gfxQueueGSyncPut':
/home/user/esp-iot-solution/components/hmi/ugfx_gui/ugfx/src/gqueue/gqueue.c:193:13: warning: 'taskENTER_CRITICAL(mux)' is deprecated in ESP-IDF, consider using 'portENTER_CRITICAL(mux)'
   gfxSystemLock();
             ^
/home/user/esp-iot-solution/components/hmi/ugfx_gui/ugfx/src/gqueue/gqueue.c:195:13: warning: 'taskEXIT_CRITICAL(mux)' is deprecated in ESP-IDF, consider using 'portEXIT_CRITICAL(mux)'
   gfxSystemUnlock();
             ^
/home/user/esp-iot-solution/components/hmi/ugfx_gui/ugfx/src/gqueue/gqueue.c: In function 'gfxQueueGSyncPush':
/home/user/esp-iot-solution/components/hmi/ugfx_gui/ugfx/src/gqueue/gqueue.c:210:13: warning: 'taskENTER_CRITICAL(mux)' is deprecated in ESP-IDF, consider using 'portENTER_CRITICAL(mux)'
   gfxSystemLock();
             ^
/home/user/esp-iot-solution/components/hmi/ugfx_gui/ugfx/src/gqueue/gqueue.c:212:13: warning: 'taskEXIT_CRITICAL(mux)' is deprecated in ESP-IDF, consider using 'portEXIT_CRITICAL(mux)'
   gfxSystemUnlock();
             ^
/home/user/esp-iot-solution/components/hmi/ugfx_gui/ugfx/src/gqueue/gqueue.c: In function 'gfxQueueGSyncInsert':
/home/user/esp-iot-solution/components/hmi/ugfx_gui/ugfx/src/gqueue/gqueue.c:224:13: warning: 'taskENTER_CRITICAL(mux)' is deprecated in ESP-IDF, consider using 'portENTER_CRITICAL(mux)'
   gfxSystemLock();
             ^
/home/user/esp-iot-solution/components/hmi/ugfx_gui/ugfx/src/gqueue/gqueue.c:226:13: warning: 'taskEXIT_CRITICAL(mux)' is deprecated in ESP-IDF, consider using 'portEXIT_CRITICAL(mux)'
   gfxSystemUnlock();
             ^
/home/user/esp-iot-solution/components/hmi/ugfx_gui/ugfx/src/gqueue/gqueue.c: In function 'gfxQueueGSyncRemove':
/home/user/esp-iot-solution/components/hmi/ugfx_gui/ugfx/src/gqueue/gqueue.c:248:13: warning: 'taskENTER_CRITICAL(mux)' is deprecated in ESP-IDF, consider using 'portENTER_CRITICAL(mux)'
   gfxSystemLock();
             ^
/home/user/esp-iot-solution/components/hmi/ugfx_gui/ugfx/src/gqueue/gqueue.c:250:13: warning: 'taskEXIT_CRITICAL(mux)' is deprecated in ESP-IDF, consider using 'portEXIT_CRITICAL(mux)'
   gfxSystemUnlock();
             ^
/home/user/esp-iot-solution/components/hmi/ugfx_gui/ugfx/src/gqueue/gqueue.c: In function 'gfxQueueGSyncIsIn':
/home/user/esp-iot-solution/components/hmi/ugfx_gui/ugfx/src/gqueue/gqueue.c:277:13: warning: 'taskENTER_CRITICAL(mux)' is deprecated in ESP-IDF, consider using 'portENTER_CRITICAL(mux)'
   gfxSystemLock();
             ^
/home/user/esp-iot-solution/components/hmi/ugfx_gui/ugfx/src/gqueue/gqueue.c:279:13: warning: 'taskEXIT_CRITICAL(mux)' is deprecated in ESP-IDF, consider using 'portEXIT_CRITICAL(mux)'
   gfxSystemUnlock();
             ^
```

InfiniteYuan1, 2018-10-17:
Thanks for your comments, uGFX will soon be consistent with ESP-IDF.

themodernlife, 2019-01-26:
Hi, is there any other information someone could share or a workaround for this issue? Couldn’t find anything on the uGFX site. 

Thank you,

closed 2023-07-10 by leeebo