espressif/esp-iot-solution #21: WroverLCD example fails at lcd_cmd()

zhivko, 2018-12-03:
## Environment

- Development Kit:      [ESP32-Wrover-Kit]
- Kit version (for WroverKit): [v1|v2|v3|v4]
- Module or chip used:  [ESP32-WROVER|ESP32-WROVER-I|ESP32-WROVER-B|ESP32-WROVER-IB]
- IDF version (run ``git describe --tags`` to find it): 
    v3.3-dev-301-gadd7c49a2
- Build System:         [Make|CMake]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
xtensa-esp32-elf-gcc.exe (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a5) 5.2.0
- Operating System:     [Windows]
- Power Supply:         [USB]
## Problem Description
Panic happens while trying wrover esp example

//Detailed problem description goes here.

### Expected Behavior
Some text should be written on wrover lcd display

### Actual Behavior
Panic because of assert at picture below. I used jtag to see where assert happens.
You get assert at this line:
[image]

### Steps to repropduce

1. Make project originally build from blink example appended with LCD code from this git repository
2. you could pull from repository:
3. build project & upload to wrover

### Code to reproduce this issue
https://github.com/zhivko/WroverLCD

InfiniteYuan1, 2018-12-03:
Ok, please modify the clk_freq of the SPI in the example to be less than 26.7 Mhz.
``` 
lcd_conf_t lcd_pins = {
        .lcd_model    = LCD_MOD_AUTO_DET,
        .pin_num_miso = GPIO_NUM_25,
        .pin_num_mosi = GPIO_NUM_23,
        .pin_num_clk  = GPIO_NUM_19,
        .pin_num_cs   = GPIO_NUM_22,
        .pin_num_dc   = GPIO_NUM_21,
        .pin_num_rst  = GPIO_NUM_18,
        .pin_num_bckl = GPIO_NUM_5,
        .clk_freq     = 26 * 1000 * 1000,
        .rst_active_level = 0,
        .bckl_active_level = 0,
        .spi_host = HSPI_HOST,
        .init_spi_bus = true,
    };
```

zhivko, 2018-12-03:
It works now :) What is the catch?  Why 26 instead 40 works? Could you explain?

InfiniteYuan1, 2018-12-03:
1. When work in full-duplex mode at frequency > 26.7MHz, device cannot read correct data.

closed 2018-12-03 by zhivko