espressif/esp-iot-solution #24: Using LittlevGL with an LCD display that doesn't have a touch screen driver

fivdi, 2018-12-26:
## Environment

- Development Kit:      ESP32-Wrover-Kit
- Kit version (for WroverKit/PicoKit/DevKitC): v3
- Module or chip used:  ESP32-WROVER
- IDF version (run ``git describe --tags`` to find it): 
    // v3.1.1-68-g4070d09
- Build System:         Make
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // 1.22.0-80-g6c4433a
- Operating System:     Linux
- Power Supply:         external 5V

## Problem Description

An ESP-WROVER-KIT V3 has an LCD display with an ILI9341 screen driver but it has no touch screen driver. If the LittlevGL library is used, it would therefore make sense to turn off the menuconfig option `IoT Solution settings > IoT Components Management > HMI Components > LVGL settings > LittlevGL Touch Screen Enable`. However, if this option is turned off, there is the following compile error:
```
/home/brian/src/esp-idf/esp-iot-solution/examples/hmi/lvgl_example/build/lvgl_gui/liblvgl_gui.a(lvgl.o):(.literal.lvgl_init+0x10): undefined reference to `lvgl_indev_init'
/home/brian/src/esp-idf/esp-iot-solution/examples/hmi/lvgl_example/build/lvgl_gui/liblvgl_gui.a(lvgl.o): In function `lvgl_init':
/home/brian/src/esp-idf/esp-iot-solution/components/hmi/lvgl_gui/lvgl.c:52: undefined reference to `lvgl_indev_init'
collect2: error: ld returned 1 exit status
/home/brian/src/esp-idf/esp-iot-solution/submodule/esp-idf//make/project.mk:406: recipe for target '/home/brian/src/esp-idf/esp-iot-solution/examples/hmi/lvgl_example/build/lvgl_example.elf' failed
make: *** [/home/brian/src/esp-idf/esp-iot-solution/examples/hmi/lvgl_example/build/lvgl_example.elf] Error 1
```
The compile error occurs because [this](https://github.com/espressif/esp-iot-solution/blob/master/components/hmi/lvgl_gui/lvgl.c#L58) line of code is attempting to call the non-existant `lvgl_indev_init` function:
```c
 lv_indev_drv_t indevdrv = lvgl_indev_init(); /* Initialize your indev */
```
Would it not be better to change this line of code to the following:
```c
#ifdef CONFIG_LVGL_DRIVER_TOUCH_SCREEN_ENABLE
 lv_indev_drv_t indevdrv = lvgl_indev_init(); /* Initialize your indev */
#endif
```
[This](https://github.com/espressif/esp-iot-solution/blob/master/components/hmi/lvgl_gui/lvgl.c#L70) line of code:
```c
 lvgl_calibrate_mouse(indevdrv, false);
```
would also need to be changed to:
```c
#ifdef CONFIG_LVGL_DRIVER_TOUCH_SCREEN_ENABLE
 lvgl_calibrate_mouse(indevdrv, false);
#endif
```

InfiniteYuan1, 2018-12-28:
OK, we‘ll fix this.

fivdi, 2018-12-29:
That was fast! Thank you for fixing this with 55a9941739fa1adbea1c36e7d12282f365b36086. I just gave it a try and it works.

closed 2018-12-29 by fivdi