espressif/esp-iot-solution #26: Some problem with i2s_lcd_write_data()

techtoys, 2019-01-14:
## Environment

- Development Kit:      [ESP32-PICO-Kit]
- Kit version(PicoKit):   Not sure
- Module or chip used:  [ESP32-PICO-D4]
- IDF version : v3.3-beta1-179-ge931fe9f5
- Build System:         [CMake]
- Compiler version: (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a5) 5.2.0

- Operating System:     [Windows]
- Power Supply:         [USB]

## repeat sending data from i2s_lcd_write_data() not correct

### expect: 0xab 0xab 0xab 0xab for the first 4 bytes at least

### actual: 0xab 0xab 0xaf 0a51 

I am trying to understand how 8080 output from i2s work. To do this I have copied 5 files from \esp-iot-solution-master\components\i2s_devices\lcd_common to the root directory of a hello-world program with structure as below. All i2s_lcd_xxx.c and .h files in the same directory as hello_world_main.c for simplicity.
- i2s_8080_lcd
  -CMakeLists
  -Makefile
  -sdkconfig
  -main 
     |- CMakelists
     |- hello_world_main.c
     |- i2s_lcd.c
     |- i2s_lcd_com.c
     |- i2s_lcd_com.h
     |- iot_i2s_lcd.h

### Code to reproduce this issue (hello_world_main.c) as below

```cpp
#include <stdio.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_system.h"
#include "esp_spi_flash.h"
#include "sdkconfig.h"
#include "../components/drv/dummy.h"
#include "i2s_lcd_com.h"
#include "iot_i2s_lcd.h"

#define I2S_PORT_NUM	(0)
#define LCD_D0_PIN  (19)
#define LCD_D1_PIN  (21)
#define LCD_D2_PIN  (0)
#define LCD_D3_PIN  (22)
#define LCD_D4_PIN  (23)
#define LCD_D5_PIN  (33)
#define LCD_D6_PIN  (32)
#define LCD_D7_PIN  (27)
#define LCD_D8_PIN  (25)
#define LCD_D9_PIN  (26)
#define LCD_D10_PIN (12)
#define LCD_D11_PIN (13)
#define LCD_D12_PIN (14)
#define LCD_D13_PIN (15)
#define LCD_D14_PIN (2)
#define LCD_D15_PIN (4)
#define LCD_WR_PIN  (18)
#define LCD_RS_PIN  (5)

void app_main()
{
    dummy_init();

    i2s_lcd_config_t d2805_conf={
    		16,	//data_width = 16
			{
			        LCD_D0_PIN, LCD_D1_PIN, LCD_D2_PIN, LCD_D3_PIN,
			        LCD_D4_PIN, LCD_D5_PIN, LCD_D6_PIN, LCD_D7_PIN,
			        LCD_D8_PIN, LCD_D9_PIN, LCD_D10_PIN,LCD_D11_PIN,
			        LCD_D12_PIN, LCD_D13_PIN, LCD_D14_PIN, LCD_D15_PIN
			},
			LCD_WR_PIN,
			LCD_RS_PIN
    };

    i2s_lcd_handle_t d2805_handle = i2s_lcd_create(I2S_PORT_NUM, &d2805_conf);
    DRAM_ATTR static const char fb[10] = {0xab, 0xab, 0xab, 0xab, 0x55, 0xaf, 0x3d, 0x6f, 0x44, 0x78};

    for(;;){
    	i2s_lcd_write_data(d2805_handle, (char *)&fb, sizeof(fb), portTICK_PERIOD_MS, false);
    }
}
```
Pins [0:15] are wired to a logic analyzer to visualize the bus. This program only send a fixed byte pattern of 10. It was expected the pattern should match the byte array defined in fb[10] but the actual waveform is shown below. 

[image]

The first two bytes are always correct. They match the first two elements 0xab 0xab, but the 3rd and 4th elements started to corrupt. It should be 0xab 0xab. I have scripted them to be the same as the 1st and 2nd byte to avoid any mistake in jumper wires. Since the 1st and 2nd bytes are correct, it means my wiring should be OK. Inconsistency is very "consistent"! I mean after a restart or power removal, the 3rd and 4th bytes are always 0xaf 0x51. So the error is a systematic one.

What is the mistake? How to achieve a simple result like repeat sending some byte array like above?

InfiniteYuan1, 2019-01-14:
I think we should use the uint16_t array to test the i2s of the 16-bit data width. For example:

```c
uint16_t fb[10] = {0xabab, 0xabab, 0x55af, 0x3d6f, 0x4478, 0xabab, 0xabab, 0x55af, 0x3d6f, 0x4478};
```

techtoys, 2019-01-14:
I have found an error from hardware. GPIO0 should not be used as LCD_D2_PIN from ESP32-PICO-D4 for some unknown reason(s). Comparing the byte mismatch it is always the GPIO 0 pulled to 0. So I have changed to GPIO9 as LCD_D2_PIN.  `#define LCD_D2_PIN  (9) // 0->9`. And I have tested with  uint16_t arrays with result below:

Src code:
```
DRAM_ATTR static const uint16_t fb[10] = {0xabcd, 0xcdba, 0xab88, 0xac23, 0x5544, 0xaf54, 0x3d2a, 0x6f56, 0x4423, 0x78fa};

    for(;;){
    	i2s_lcd_write_data(d2805_handle, (const uint16_t *)&fb, sizeof(fb), portTICK_PERIOD_MS, false);
    	vTaskDelay(500 / portTICK_PERIOD_MS);
    }
```

Waveform now matches perfectly but there is an error when compiling because it is a `const char *src`  in prototype of `i2s_lcd_write_data()`. The error as: `**expected 'const char *' but argument is of type 'const uint16_t * {aka const short unsigned int *}'** `

[image]

Another test with uint8_t array with data width changed to 8 does not sound correct:
```
void app_main()
{
    dummy_init();

    i2s_lcd_config_t d2805_conf={
    		8,	//data_width = 8
			{
			        LCD_D0_PIN, LCD_D1_PIN, LCD_D2_PIN, LCD_D3_PIN,
			        LCD_D4_PIN, LCD_D5_PIN, LCD_D6_PIN, LCD_D7_PIN,
			        //LCD_D8_PIN, LCD_D9_PIN, LCD_D10_PIN,LCD_D11_PIN,
			        //LCD_D12_PIN, LCD_D13_PIN, LCD_D14_PIN, LCD_D15_PIN
			},
			LCD_WR_PIN,
			LCD_RS_PIN
    };

    i2s_lcd_handle_t d2805_handle = i2s_lcd_create(I2S_PORT_NUM, &d2805_conf);
    DRAM_ATTR static const char fb[10] = {0xab, 0xcd, 0xab, 0x23, 0x44, 0xaf, 0x3d, 0x56, 0x42, 0xfa};

    for(;;){
    	i2s_lcd_write_data(d2805_handle, (const char *)&fb, sizeof(fb), portTICK_PERIOD_MS, false);
    	vTaskDelay(500 / portTICK_PERIOD_MS);
    }
}
```
Screen shot from Logic analyzer:
[image]

According to the probe result the byte sequence sent from LCD_D0_PIN[7:0] are `0xab 0xab 0x44 0x3d 0x42`

Can't we send with 8-bit 8080?

techtoys, 2019-01-14:
I think the reason why 8-bit didn't work is now clear. Needed to put `#define CONFIG_BIT_MODE_8BIT` right at the beginning of i2s_lcd.c to let the system know we are using 8-bit. So I believe the default is 16-bit with definition set somewhere...
` #define CONFIG_BIT_MODE_8BIT` is put right next to #include "esp_log.h"
[image]

Clean and make flash will give the right waveform for 8-bit:
[image]

Thank you

techtoys, 2019-01-14:
Some more findings. It seems there is a problem with `iot_i2s_lcd_write_data(i2s_lcd_handle_t i2s_lcd_handle, uint16_t data)` when data width is set to 8BIT. In i2s_lcd_com.c the code is :

```
#ifdef CONFIG_BIT_MODE_8BIT

void iot_i2s_lcd_write_data(i2s_lcd_handle_t i2s_lcd_handle, uint16_t data)
{
    i2s_lcd_write_data(i2s_lcd_handle, (char *)&data, 2, 100, true);
}
```

When 8 bit is defined, data type should be uint8_t, and the 3rd argument should be 1 instead of 2?

Another issue is that, RS pin is never strobed when `iot_i2s_lcd_write_cmd(i2s_lcd_handle_t i2s_lcd_handle, uint16_t cmd)` is called.

techtoys, 2019-01-14:
Some more findings. It seems there is a problem with `iot_i2s_lcd_write_data(i2s_lcd_handle_t i2s_lcd_handle, uint16_t data)` when data width is set to `8BIT`. In `i2s_lcd_com.c` the code is :

```
#ifdef CONFIG_BIT_MODE_8BIT

void iot_i2s_lcd_write_data(i2s_lcd_handle_t i2s_lcd_handle, uint16_t data)
{
    i2s_lcd_write_data(i2s_lcd_handle, (char *)&data, 2, 100, true);
}
```
When 8 bit is defined, data type should be uint8_t, and the 3rd argument should be 1 instead of 2?

Another issue is that, RS pin is never strobed when `iot_i2s_lcd_write_cmd(i2s_lcd_handle_t i2s_lcd_handle, uint16_t cmd)` is called.

techtoys, 2019-01-15:
The problem with RS pin never strobed has been solved. Mistake has been made with calling `i2s_lcd_create()` without setting the direction of RS pin which is treated as a gpio. Instead of `i2s_lcd_create` the function `iot_i2s_lcd_pin_cfg()` should be called which include both `i2s_lcd_create()` & `gpio_set_direction()` for RS pin. 

The root of another problem with `iot_i2s_lcd_write_cmd(i2s_lcd_handle_t i2s_lcd_handle, uint16_t cmd)` for `CONFIG_BIT_MODE_8BIT` seems to be a 8bit left shift (cmd>>8) as follows:

```
void iot_i2s_lcd_write_cmd(i2s_lcd_handle_t i2s_lcd_handle, uint16_t cmd)
{
    i2s_lcd_t *i2s_lcd = (i2s_lcd_t *)i2s_lcd_handle;
    i2s_port_t i2s_num = i2s_lcd->i2s_port;
    GPIO.out_w1tc = (1 << i2s_lcd->i2s_lcd_conf.rs_io_num);
    REG_WRITE(I2S_FIFO_ADD[i2s_num], (cmd >> 8));
    I2S[i2s_num]->conf.tx_start = 1;
```

This function assumed a uint16_t cmd and user should allocate the high byte for the actual command. Example,  if 0x2A is wanted one need to send like `iot_i2s_lcd_write_cmd(i2s_lcd_handle, 0x2A00)`, but it turns out that a dummy 0x00 is also sent in the tail after 0x2A. Below is a screen shot from analyzer:

[image]

techtoys, 2019-01-16:
Hi
After much effort in understanding how i2s_8080 work, it seems 8 bit parallel bus is working OK with modification to two functions in i2s_lcd_com.c. 

(1) 
```
#ifdef CONFIG_BIT_MODE_8BIT

void iot_i2s_lcd_write_data(i2s_lcd_handle_t i2s_lcd_handle, uint16_t data)
{
    //i2s_lcd_write_data(i2s_lcd_handle, (char *)&data, 2, 100, true); //original
    i2s_lcd_write_data(i2s_lcd_handle, (char *)&data, 1, 100, false); //new
}
```
(2)
```
void iot_i2s_lcd_write_cmd(i2s_lcd_handle_t i2s_lcd_handle, uint16_t cmd)
{
    i2s_lcd_t *i2s_lcd = (i2s_lcd_t *)i2s_lcd_handle;
    i2s_port_t i2s_num = i2s_lcd->i2s_port;
    GPIO.out_w1tc = (1 << i2s_lcd->i2s_lcd_conf.rs_io_num);

    REG_WRITE(I2S_FIFO_ADD[i2s_num], (uint8_t)cmd); //new
    I2S[i2s_num]->conf.tx_start = 1;
    while (!(I2S[i2s_num]->state.tx_idle)) {
        // vTaskDelay(20);
        ;
    }
    I2S[i2s_num]->conf.tx_start = 0;
    I2S[i2s_num]->conf.tx_reset = 1;
    I2S[i2s_num]->conf.tx_reset = 0;
    I2S[i2s_num]->conf.tx_fifo_reset = 1;
    I2S[i2s_num]->conf.tx_fifo_reset = 0;

    GPIO.out_w1ts = (1 << i2s_lcd->i2s_lcd_conf.rs_io_num);
}
```

Repeat sending command & data in 8-bit 8080 with snippet :
```
void app_main()
{
    dummy_init();

    i2s_lcd_config_t d2805_conf={
    		8,	//data_width = 8
			{
			        LCD_D0_PIN, LCD_D1_PIN, LCD_D2_PIN, LCD_D3_PIN,
			        LCD_D4_PIN, LCD_D5_PIN, LCD_D6_PIN, LCD_D7_PIN,
			},
			LCD_WR_PIN,
			LCD_DC_PIN
    };

    i2s_lcd_handle_t d2805_handle = iot_i2s_lcd_pin_cfg(I2S_PORT_NUM, &d2805_conf);
    for(;;){
    	iot_i2s_lcd_write_cmd(d2805_handle, 0x2A);
    	iot_i2s_lcd_write_data(d2805_handle, 0xef);
    }
}
```
Dump screen from logic analyzer is shown here. 

[image]

classic-gentleman, 2019-01-17:
Please see https://github.com/espressif/esp-iot-solution/issues/19

techtoys, 2019-01-22:
The code is working in 8-BIT 8080 mode is working. Picture is shown below for my screen.
[image]

With a logic analyzer WR# pulse frequency is measured to be ~ 13MHz. I can see that the WR# pulse rate is actually BCK clock rate. Its frequency is calculated by `Fbck = Fpll /(N + (b/a))/M.`

Inside `esp_err_t i2s_set_parallel_mode(i2s_port_t i2s_num)`, 
```
I2S[i2s_num]->clkm_conf.clkm_div_b = 1; 
I2S[i2s_num]->clkm_conf.clkm_div_a = 0;
```
I don't understand it when clkm_conf.clkm_div_a is set '0', b/a would be undefined. Why a 13MHz is obtained?

So I tried 
```
I2S[i2s_num]->clkm_conf.clkm_div_b = 0; 
I2S[i2s_num]->clkm_conf.clkm_div_a = 1;
``` hoping to get a max. bck rate = 160/2/2 = 40MHz with N=M=2 but the screen no longer works.

closed 2019-01-25 by techtoys