espressif/esp-iot-solution #27: ESP32 Sense Kit - Liquids

c2scott, 2019-01-23:
Just purchased from DigiKey the ESP32 Sense Kit.  Works nice until a little moisture gets on one of the 9 key.  Then none of the buttons work.  Is this expected operation?  I also cannot get it to work at all with any of the protective covers.  Does this touch sensor not auto-calibrate?

InfiniteYuan1, 2019-03-11:
TouchPad has [self-calibration](https://github.com/espressif/esp-iot-solution/blob/master/documents/touch_pad_solution/touch_sensor_design_en.md#43-self-calibration-solution)function. 
If you still have this problem, please attach some pictures about the touchpad to help you use it easily.

Xiehanxin, 2020-07-27:
@c2scott  hi, If your issue is now resolved, can you please close the issue. if not, please let us know. Thank you.

c2scott, 2020-07-27:
I dropped the project; you may close the case.