espressif/esp-iot-solution #31: (eth2wifi) Get a Mess after a while in Wifi SoftAP mode

ESP32DE, 2019-03-09:
Hi

the _**Ethernet to wifi station data forwarding demo**_ works very well with an extern Wifi Hotspot.
same doing to use the esp32 as own _**Wifi Hotspot and the Ethernet to wifi softap data forwarding demo**_
the things works at begin, then after a while of continous stream data 3-5 seconds the stream data becomes a mess. 

after a new refresh request the stream begin to work ok  and then it work for 3-5 seconds and the mess begins again.

i did not look deeper in the handle of FiFo or ringbuffer, which use the **Ethernet to wifi softap data forwarding demo** perhabs. 
but can you please check or say, can we flush the data ringbuffer or can we setup with more buffer that the datastream does not go confuse anymore`?

sry - the wifi things are in a blob bin
so i can't do anything on a normal way to check the things

My example Setup:
---------------------

using the ESP32 and LAN8720
works well with the setup demo, works well with the Station Demo.
the hardware setup is not the problem - it works very fine.

just only change from Station to softAP mode
works 3-5 seconds that the stream gets a mess.

i use the[ original demo code without any change](https://github.com/espressif/esp-iot-solution/tree/master/examples/eth2wifi)

Symbolic
Ethernet to wifi station data forwarding demo
"wireless adapter"

DEVICE to forward data has LAN -> connected by RJ45 to -> LAN8720 -> ESP32 as Station -> connect to a extern Wifi Hotspot
Device sends continous data stream to the connected client at extern Wifi Hotspot
there is no problem and all works fine

Ethernet to wifi softap data forwarding demo
"WiFi hotspot"

DEVICE to forward data has LAN -> connected by RJ45 to -> LAN8720 -> ESP32 as Wifi Hotspot
Device sends continous data stream to the connected client at ESP32(WifiHotspot)
first 3-5 seconds all ok, then get a mess in data stream

where can you make a flush or increase or better tune the buffer
i think the mistake is in the buffer ( read write forward goes to mess )
in Hotspot Mode.

any suggestion?
it is a very good example - i do not want to miss it

thank you
best wishes
rudi

ESP32DE, 2019-03-09:
hi
i try a step by step to increasing the DMA_RX_BUF_NUM in the MENUCONFIG
[image]

10..12..15..18..20
this helps me in first step and i think the 15 would be enough. 
i will try this procedure for now a longer time and update here for success or any.
if you have any other suggestion plz let me know.

[image]

best wishes
rudi ;-)

projectgus, 2019-03-11:
Hi @ESP32DE ,

Can you describe the stream you're sending? Is it a TCP stream or UDP or something else? How fast is the data rate?

Are you able to install Wireshark on the sender and receiver of the connection? If you can capture the stream with Wireshark (to compare in the capture files: which packets were sent, which packets were received) then this would be very useful for rapid debugging of the issue.

Thanks,

Angus

ESP32DE, 2019-03-11:
Hi Angus
@projectgus 

first, many thanks for your help.
and first, i must rename the "mess" to "always gets stuck again" cause

Angus - this happens only if the ESP32 is in Hotspot Mode, 
if i switch to Station Mode and use an extern Wifi Hotspot there is no Problem with eth2wifi.
so the more work option as a "Hotspot Mode" trigers the "stuck problem"

Hotspot Mode:
the data itself i think ( not debug just in time ) fly correct like Station Mode - 
i will do the things with Wireshark what you suggest asap

the stream data is receive by "firefox" ( windows) the stream will be interrupted sometime
a refresh let the stream then connect again and it runs a while until the stream will be interrupted again..
( stucks anyway )

Stream Info:
the "stress test" stream itself is a simply jpgeg motion stream comes in the LAB test from a medial server in the house.
the client send a request and the server response with html standard and accept and so on and send continous the stream data by TCP.  the stream it self can be configured by the client 
example the resolution "320x240" or "640x480" ... 
also the (jpeg) compression from 0..100
also the fps from 1..30..50..60

as a full test with 768x576 compression 0 and 25 fps the stream comes to 4 MB/s
as a usually test stream i take 352x288 compression 20 and 25 fps.

Strange:
if i use the Wifi Hotspot Mode and connect the "windows" client to the esp32 
after password authentification ( WPA2/PSK)  the things is connected right.
if i connect then the firefox client to the medial server, the stream will be run a while and then interrupted sometime,
it looks like, "windows" handle the Wireless connection in a "block mode sometimes"
it looks like after the authentification there comes moretimes again this authentification and
this "Blocked" the forward work of the esp32, and the stream get after a while interrupted..
( response time ...) 

but!
if i connect an "Android client" firefox  to the esp32 Wifi Hotspot, the stream works like Station Mode...
perfect, ..
and if i connect a second Client by "windows firefox" client the stream works in the "windows" system without problems. ( strange strange strange ) 

so i think - it is perhabs a (windows) "client" problem?
i will do the debug with Wireshark asap i get a longer time window for do this.

i remember me, that the "Wifi Soft Accesspoint" mode on esp32 have in the past here and there a problem on "windows" things.

what i did try:
- closed windows public firewall
- closed windows privat firewall

what i will try extra:
- i will disable wireless things on the windows client and use a esp32 station mode forwarding setup
and connect the LAN8720 on the windows client (LAN RJ45) , so the connection is then done by
esp32 station - esp32 wifi hotspot - let me try this doing next step...

My Questions are:

can we change the "Time to Live" things in the "Wifi Hotspot Mode" so, that the authenification is only one time need and the next is in few hour's? 
i think the repeat to do the authentifications things (TTL, Beacon... ) in "windows" is blocking the data stream...
..and this i think is the problem here. ( i can't see at first moment, that data are lost or leaked ) 

what was watching by the testings:

if you use DevKitC and LAN8720 i use for LAN8720 PowerSrc/Supply 3.3V from the DevKitC (3.3V)
i do not think, that this is really wrong, but i try also to get the PowerSrc for LAN8720 from a extern power supply and connect GND together. Just in time i use DevKitC as PowerSrc for LAN8720
and have connected one "windows client" and two "android clients" and the stream work just in time 
unbreakable continous ( strange ) perfect :+1: 
strange:
in the moment i switch off the android clients :) (wireless off ) .... and only one windows client is connected at esp32
the things start again with interrupting and i get stream brokens.
(fun does esp32 wifi does not like windows only clients and attacks the windows system ? /fun) :)))

hope the description helps a little.
if you need more info please let me know.
thank you for your time and read the problem.

things about the stream:

it is a jpeg motion stream.
data comes continous example..

--myboundary
Content-Type: image/jpeg
Content-Length: 82602

ÿØÿà.......

--myboundary
Content-Type: image/jpeg
Content-Length: XXXXX

ÿØÿà.......

--myboundary
Content-Type: image/jpeg
Content-Length: XXXXX

ÿØÿà.......

(XXXXX is allways the length of the data ) and so on
a standard jpeg motion stream.

best wishes
rudi ;-)

ESP32DE, 2019-03-11:
i think we must look detailed

[image]

ESP32DE, 2019-03-11:
..
trouble processes (WPA2 .. Authents ..) caused by windows?
I do not really believe it.
I made another attempt:
ESP32 as client and ESP32 as Wifi hotspot
so the Wifi story is in hand by ESP.
then
eth2wifi forwarding, the result was not good,
rather bad,
I switched off WPA2, which means open web.
no improvement.
I still have one guess:
insufficient power supply for the LAN8270?
(DevKitC very weak LDO)

But - same PowerSupply setup
and at the moment this setup is very stable:
ESP32 Wifi Hotspot,
20 DMA buffer Rx
20 DMA buffer Tx
Android clients.
No problem so far.

Once Windows - the problem starts again.
a bit baffled at the moment :)
..

ESP32DE, 2019-03-11:
..after 3h

...
Memorie_0 : 186440	****
Memorie_1 : 188084	****
Memorie_2 : 196324
Memorie_3 : 196324
Memorie_0 : 196324
Memorie_1 : 196324
Memorie_2 : 196324
Memorie_3 : 196324
Memorie_0 : 196324
Memorie_1 : 196324
Memorie_2 : 196324
Memorie_3 : 196324
Memorie_0 : 196324
Memorie_1 : 196324
Memorie_2 : 196324
Memorie_3 : 196324
Memorie_0 : 196324
Memorie_1 : 196324
Memorie_2 : 196324
Memorie_3 : 196324
Memorie_0 : 196324
Memorie_1 : 196324
Memorie_2 : 196324
Memorie_3 : 196324
Memorie_0 : 196324
Memorie_1 : 196324
Memorie_2 : 196324
Memorie_3 : 196324
Memorie_0 : 196324
Memorie_1 : 196324
Memorie_2 : 196324
Memorie_3 : 196324
Memorie_0 : 196324
Memorie_1 : 196324
Memorie_2 : 196324
Memorie_3 : 196324
Memorie_0 : 196324
Memorie_1 : 196324
Memorie_2 : 196324
Memorie_3 : 196324
Memorie_0 : 196324
Memorie_1 : 196324
Memorie_2 : 196324
Memorie_3 : 196324
Memorie_0 : 196324
Memorie_1 : 196324
Memorie_2 : 196324
Memorie_3 : 196324
Memorie_0 : 196324
Memorie_1 : 196324
Memorie_2 : 196324
Memorie_3 : 196324
Memorie_0 : 196324
Memorie_1 : 196324
Memorie_2 : 196324
Memorie_3 : 196324
Memorie_0 : 196324
Memorie_1 : 196324
Memorie_2 : 196324
Memorie_3 : 196324
Memorie_0 : 188088	****
Memorie_1 : 188088	****
Memorie_2 : 196324
Memorie_3 : 196324
Memorie_0 : 196324
Memorie_1 : 196324
Memorie_2 : 196324
[... 1393 lines cut ...]
Memorie_0 : 196324
Memorie_1 : 196324
Memorie_2 : 196324
Memorie_3 : 196324
Memorie_0 : 196324
Memorie_1 : 196324
Memorie_2 : 196324
Memorie_3 : 196324
Memorie_0 : 196324
Memorie_1 : 196324
Memorie_2 : 196324
Memorie_3 : 196324
Memorie_0 : 196324
Memorie_1 : 196324
Memorie_2 : 196324
Memorie_3 : 196324
Memorie_0 : 196324
Memorie_1 : 196324
Memorie_2 : 196324
Memorie_3 : 196324
Memorie_0 : 184792	*********
Memorie_1 : 184792	*********
Memorie_2 : 194676	*********
Memorie_3 : 196324
....

projectgus, 2019-03-11:
So the problem only happens with Windows client and ESP32 SoftAP?

Does the Windows client disconnect entirely from the ESP32 AP when the stream breaks, or does it stay connected but the stream drops?

In Wi-Fi, the AP sometimes has to buffer a lot of network data if it can't talk to the client. If the Windows client is going into some low power mode, it's possible the data is backing up in the ESP32 SoftAP - and ESP32 does not have as much RAM as a "real" AP for this case.

As it seems like you have plenty of free RAM in the ESP32 (>190KB), try increasing the [Number of Wi-Fi TX buffers](https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/kconfig.html#config-esp32-wifi-dynamic-tx-buffer-num). Possibly also enable [AMPDU TX](https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/kconfig.html#config-esp32-wifi-ampdu-tx-enabled) and increase the number of AMPDU TX buffers as well.

ESP32DE, 2019-03-12:
Hi Angus,
txs.
> 
> 
> So the problem only happens with Windows client and ESP32 SoftAP?

yes - it looks like this - 
if i connect android clients - there is no problem ( strange ) 

> 
> Does the Windows client disconnect entirely from the ESP32 AP when the stream breaks, 

(i think) no it does not disconnect .
if the "break" is not much long the firefox browser "paused" on the break and "play" again,
but on the time i can see there is a "jump" example break comes on 01:01:01 and then 
plays again and time is then 01:01:15
it does not buffering.

but:
if i use an ESP32 as Station to connect to ESP32 Hotspot
then the break is paused and there is "no" jump in time, it runs faster,
so i can speak from buffering and not "time synchron" -

> or does it stay connected to the AP but the stream drops?
>

yes this is the correct word for the ESP32Station-ESP32Hotspot connection.
it drops then.
 
> In Wi-Fi, the AP sometimes has to buffer a lot of network data if it can't talk to the client. If the Windows client is going into some low power mode or something, 

i understand, but the windows client is allways powered on - ( energy mode is off ) 

> it's possible the data is backing up in the ESP32 SoftAP - and ESP32 does not have as much RAM as a "real" AP for this case.
>
> As it seems like you have plenty of free RAM in the ESP32 (>190KB), try [...]

helps the ethernet things too`? 

[image]

Wifi looks like this just in time:
[image]

i will follow your suggestion Angus, txs!

> I think any other debugging will be helped greatly by some Wireshark packet captures, if you can get those.

ok i will try - 
thank you for your help!
Best wishes
rudi 

( what can this be "only in windows" this becomes problems ) 
it is magic but the serial sometimes goes crazzy way in windows :)
..

ESP32DE, 2019-03-12:
@projectgus 
solved Angus. 
see the first edit post.
thank you for your help!
best wishes
rudi ;-)

InfiniteYuan1, 2019-04-01:
If you have further questions about this please reopen the issue in the GitHub and let's continue there.

closed 2023-07-10 by leeebo