espressif/esp-iot-solution #32: How to assign static Ip for esp32 in Bridge mode

BHAKTIDESHPANDE, 2019-03-12:
I am using ESP32-EVB Rev.D olimex board and running Ethernet to wifi transparent bridge. (https://github.com/espressif/esp-iot-solution/tree/master/examples/eth2wifi) 
I want to run a web server in same code to change bridge modes (soft ap and station), instead of changing it from menuconfig.
In bridge code obviously my routers DHCP does not assigns ip to esp32, it is just for connection between Ethernet and wifi .
I get mac of soft AP and station but how to assign ip address to them for webserver?
I assigned static ip to eth0 in same code and changed ethernet init function, but bridge is was not working now.
is it possible to assign static ip to esp32 when it is in bridge mode?

My Ethernet initialization function is given below

#define DEVICE_IP          "192.168.10.254"
#define DEVICE_GW          "192.168.10.1"
#define DEVICE_NETMASK     "255.255.0.0"
static void initialise_ethernet(void)
{
     esp_err_t ret = ESP_OK;
     tcpip_adapter_ip_info_t ipInfo;
     inet_pton(AF_INET, DEVICE_IP, &ipInfo.ip);
     inet_pton(AF_INET, DEVICE_GW, &ipInfo.gw);
     inet_pton(AF_INET, DEVICE_NETMASK, &ipInfo.netmask);
     tcpip_adapter_init();
     ret = tcpip_adapter_dhcpc_stop(TCPIP_ADAPTER_IF_ETH); 
     ESP_LOGI(TAG, "dhcp client stop RESULT: %d", ret);
     tcpip_adapter_set_ip_info(TCPIP_ADAPTER_IF_ETH, &ipInfo);
     eth_config_t config = DEFAULT_ETHERNET_PHY_CONFIG;
    
   //  Set the PHY address in the example configuration 
    config.phy_addr = CONFIG_PHY_ADDRESS;
    config.gpio_config = eth_gpio_config_rmii;
   // config.tcpip_input = tcpip_adapter_eth_input_sta_output;
    config.tcpip_input = tcpip_adapter_eth_input;
#ifdef CONFIG_PHY_USE_POWER_PIN
     //Replace the default 'power enable' function with an example-specific
      // one that toggles a power GPIO. 
    config.phy_power_enable = phy_device_power_enable_via_gpio;
#endif
 //esp_eth_init_internal(&config) 
    esp_eth_init(&config);
    esp_eth_enable();
}

BHAKTIDESHPANDE, 2019-03-19:
Can anybody help regarding this post???

InfiniteYuan1, 2019-03-20:
```
    /* 
     * stop DHCP client on ehternet interface by default
     * */
    ESP_ERROR_CHECK(tcpip_adapter_dhcpc_stop(TCPIP_ADAPTER_IF_ETH));

    /* static ip settings */
    tcpip_adapter_ip_info_t sta_ip;
    sta_ip.ip.addr = ipaddr_addr("192.168.1.102");
    sta_ip.gw.addr = ipaddr_addr("192.168.1.1");
    sta_ip.netmask.addr = ipaddr_addr("255.255.255.0");
    tcpip_adapter_set_ip_info(ESP_IF_ETH, &sta_ip);
```

BHAKTIDESHPANDE, 2019-03-20:
Thank you very much for response.
but that didin't worked. Is it possible In ethernet and wifi bridge condition to use ethernet as independent interface as well?

InfiniteYuan1, 2019-04-01:
ESP-IDF can't support this feature in ethernet and wifi bridge condition.

If you have further questions about this please reopen the issue in the GitHub and let's continue there.

burkulesomesh43, 2020-06-03:
Hello IndiniteYuan1,
I have set an static ip to my esp32 in station mode.
but Access point does not shows esp is connected or shows hostname as unknown.
what is the issue? any guess?