espressif/esp-iot-solution #33: ESP-IDF iot_apds9960 - INT PIN isnt working using ESP-IDF Library

RafaelArkenos, 2019-03-13:
Hello there!
I'm having issues to enable the gesture interruption pin of APDS9960 using the ESP-IDF library, when the sensor starts it doesnt change the int pin to the high state. 
I already tested using the arduino library and works very well. Im using a osciloscope to analyze the signal.

Here is my gesture init:

` esp_err_t iot_apds9960_gesture_init(apds9960_handle_t sensor)
{
    /* Set default values for ambient light and proximity registers */
    iot_apds9960_set_adc_integration_time(sensor, 10);
    iot_apds9960_set_ambient_light_gain(sensor, APDS9960_AGAIN_4X);

    iot_apds9960_enable_gesture_engine(sensor, false);
    iot_apds9960_enable_proximity_engine(sensor, false);
    iot_apds9960_enable_color_engine(sensor, false);

    
    iot_apds9960_enable_proximity_interrupt(sensor, true); <- I added this
    iot_apds9960_clear_interrupt(sensor);

    iot_apds9960_enable(sensor, false);
    iot_apds9960_enable(sensor, true);

    iot_apds9960_set_gesture_dimensions(sensor, APDS9960_DIMENSIONS_ALL);
    iot_apds9960_set_gesture_fifo_threshold(sensor, APDS9960_GFIFO_4);
    iot_apds9960_set_gesture_gain(sensor, APDS9960_GGAIN_4X);
    iot_apds9960_set_gesture_proximity_threshold(sensor, 10, 10);
    iot_apds9960_reset_counts(sensor);

    iot_apds9960_set_mode(sensor, APDS9960_ALL);
    iot_apds9960_set_led_drive_boost(sensor, APDS9960_LEDDRIVE_100MA, APDS9960_LEDBOOST_100PCNT);
    iot_apds9960_set_gesture_waittime(sensor, APDS9960_GWTIME_2_8MS);

    iot_apds9960_set_gesture_pulse(sensor, APDS9960_GPULSELEN_8US, 8); // padrão tava 8US, 8

    iot_apds9960_enable_proximity_engine(sensor, true);
    iot_apds9960_enable_gesture_interrupt(sensor, true); <- I added this
    return iot_apds9960_enable_gesture_engine(sensor, true);
}`

Any one can give me an advice about this?

Thank you in advance.
Best regards
Rafael F. Silva

InfiniteYuan1, 2019-03-14:
1. In this example, no hardware interrupts are used. This example shows only some of the features of the APD9960, and you can modify the improvements yourself.
2. If you want to use hardware interrupts, check the APD9960 data sheet and use the correct configuration to use hardware interrupts.

Hope this helps.Thank you.

costaud, 2019-03-14:
I think this function(iot_apds9960_enable_gesture_interrupt) only set the interrupt for apds9960 sensor.

How did you connect the interrupt pin ? Did you initialize the GPIO in input mode ?

InfiniteYuan1, 2019-03-14:
yeah!
This example only uses I2C interface for apds9960 and not use interrupt pin.

RafaelArkenos, 2019-03-14:
@InfiniteYuan1 
Ty for ur atetion!
I implemented the interruption logic but i think somehow the i2c buffer isnt getting cleared by the library, because the int pin of apds9960 isnt getting LOW its stays HIGH all the time.
I still doing some tests and trying to figure out.

For example: when im running the arduino library, when i stay with my hand in front of sensor the interruption pin signals turns on HIGH and LOW.
Or if i do a gesture the interruption is enabled.

What im trying to do: - Use the interrupt pin in my logic when hes getting high and low my logic will count this pulses with a filter. I implemented it using arduino but now using idf im facing this problems.

My Code:
`#include <stdio.h>
#include <string.h>
#include <stdlib.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/queue.h"
#include "driver/gpio.h"
#include "driver/i2c.h"
#include "iot_i2c_bus.h"
#include "APDS9960.h"

#define INT_PIN_MAIN     23
#define INT_PIN_AUX      0
#define GPIO_INPUT_PIN_SEL  ((1ULL<<INT_PIN_MAIN) | (1ULL<<INT_PIN_AUX))
#define ESP_INTR_FLAG_DEFAULT 0

//Definições do I2C p/ sensor de gestos
#define APDS9960_I2C_MASTER_SCL_IO           (gpio_num_t)22          // I2C SCL - Serial Clock
#define APDS9960_I2C_MASTER_SDA_IO           (gpio_num_t)21          // i2C SDA - Serial Data
#define APDS9960_I2C_MASTER_NUM              I2C_NUM_1   // Numero da porta Mestre I2C
#define APDS9960_I2C_MASTER_TX_BUF_DISABLE   0           // Mestre I2C não precisa de buffer
#define APDS9960_I2C_MASTER_RX_BUF_DISABLE   0   	     // Mestre I2C não precisa de buffer
#define APDS9960_I2C_MASTER_FREQ_HZ          100000      // Frequencia do mestre I2C

volatile int isr_handler = 0;

static xQueueHandle gpio_evt_queue = NULL;

i2c_bus_handle_t i2c_bus = NULL;
apds9960_handle_t apds9960 = NULL;

/* Função que é chamada quando a interrupção é executada */
static void IRAM_ATTR gpio_isr_handler(void* arg)
{
    uint32_t gpio_num = (uint32_t) arg;
    xQueueSendFromISR(gpio_evt_queue, &gpio_num, NULL);
    isr_handler = 1;
}

void handle(void) {

	uint8_t gesture = iot_apds9960_read_gesture(apds9960);
	if (gesture == APDS9960_DOWN) {
		printf("BAIXO!\n");
	} else if (gesture == APDS9960_UP) {
		printf("CIMA\n");
	} else if (gesture == APDS9960_LEFT) {
		printf("ESQUERDA\n");
	} else if (gesture == APDS9960_RIGHT) {
		printf("DIREITA\n");
	}

}

static void APDS9960_Task(void* arg)
{

    uint32_t io_num;
    for(;;) {

    	// Se o evento de interrupção ocorrer, executa:
        if(xQueueReceive(gpio_evt_queue, &io_num, portMAX_DELAY)) {
        	if( isr_handler == 1) {
        		handle();
				vTaskDelay(2000 / portTICK_RATE_MS);
				printf("Interrupcao acionada!\n");
				printf("GPIO[%d] intr, val: %d\n", io_num, gpio_get_level(io_num));
        }
        if (isr_handler == 1){
        	printf("handler foi p/ 1\n");
        	isr_handler = 0;
        }
    }
  }
}

// Configuração IO com interrupção
void io_conf(void) {
	gpio_config_t io_conf;
	io_conf.intr_type = GPIO_PIN_INTR_NEGEDGE;
	io_conf.pin_bit_mask = GPIO_INPUT_PIN_SEL;
	io_conf.mode = GPIO_MODE_INPUT;
	io_conf.pull_up_en = 1;
	gpio_config(&io_conf);

	gpio_install_isr_service(ESP_INTR_FLAG_DEFAULT);
	gpio_isr_handler_add(INT_PIN_MAIN, gpio_isr_handler, (void*) INT_PIN_MAIN);

}

void i2c_conf(void) {
	int i2c_master_port = APDS9960_I2C_MASTER_NUM;
		i2c_config_t conf;
		conf.mode = I2C_MODE_MASTER;
		conf.sda_io_num = APDS9960_I2C_MASTER_SDA_IO;
		conf.sda_pullup_en = GPIO_PULLUP_ENABLE;
		conf.scl_io_num = APDS9960_I2C_MASTER_SCL_IO;
		conf.scl_pullup_en = GPIO_PULLUP_ENABLE;
		conf.master.clk_speed = APDS9960_I2C_MASTER_FREQ_HZ;

		i2c_bus = iot_i2c_bus_create(i2c_master_port, &conf);
		apds9960 = iot_apds9960_create(i2c_bus, APDS9960_I2C_ADDRESS);

}

void app_main(){

	io_conf();
	i2c_conf();

	gpio_evt_queue = xQueueCreate(10, sizeof(uint32_t)); //Evento de interrupção
	iot_apds9960_gesture_init(apds9960);
	xTaskCreate(APDS9960_Task, "APDS9960_Task", 2048, NULL, 10, NULL);
	printf("Configurações executadas e task criada\n");

}
`

InfiniteYuan1, 2019-03-16:
I think you should look at the register configuration of the apds9960, because the interrupt pin related registers may not be properly configured using the code provided here. Alternatively, you can refer to the register configuration of arduino.

RafaelArkenos, 2019-03-18:
> 
> 
> I think you should look at the register configuration of the apds9960, because the interrupt pin related registers may not be properly configured using the code provided here. Alternatively, you can refer to the register configuration of arduino.

Do you know if i can use the arduino librarie directly to sdk idf?( i know if i re-write in idf struct may work) because i was comparing them they are different at some points.

InfiniteYuan1, 2019-03-20:
You can discuss the usage of arduino here. [arduino-esp32](https://github.com/espressif/arduino-esp32)

InfiniteYuan1, 2019-04-01:
If you have further questions about this please reopen the issue in the GitHub and let's continue there.

closed 2023-07-10 by leeebo