espressif/esp-iot-solution #35: I2C and SPI thread-safe

colesnicov, 2019-03-19:
## Problem Description
Hello.
If it is a components `SPI` and `I2C` **thread-safe?**

For now, i'll use:
```cpp
BaseType_t I2C_take() {
	return xSemaphoreTake(i2c_semphr, 1000 / portTICK_RATE_MS);
}

BaseType_t I2C_give() {
	return xSemaphoreGive(i2c_semphr);
}
```

and in the code

```
if (I2C_take() == pdTRUE) {
	float t = iot_bme280_read_temperature(dev);
	I2C_give();
	xQueueSend(queue_, &t, 0);
}
```

If you known about [This Man](https://github.com/UncleRus/esp-idf-lib) and his i2c_dev component thats is thread-safe.

Sorry for bad english :)

InfiniteYuan1, 2019-03-20:
- I2C driver in esp-idf prevents multiple tasks from accessing I2C peripherals.
- If your application has exclusive operations, you can use semaphores to mutually exclusive.

fivdi, 2019-03-23:
As mentioned in the documentation for the `i2c_master_cmd_begin` function [here](https://docs.espressif.com/projects/esp-idf/en/v3.1.3/api-reference/peripherals/i2c.html#_CPPv220i2c_master_cmd_begin10i2c_port_t16i2c_cmd_handle_t10TickType_t), the I2C APIs are not thread-safe.

As documented [here](https://docs.espressif.com/projects/esp-idf/en/v3.1.3/api-reference/peripherals/spi_master.html#thread-safety), the SPI APIs can be used in a way that is thread-safe.

colesnicov, 2019-03-24:
If you wanna implement safe mode for I2C?

fivdi, 2019-03-24:
Use semaphores for mutual exclusion as suggested by @InfiniteYuan1 in the above [comment](https://github.com/espressif/esp-iot-solution/issues/35#issuecomment-474667781).

@InfiniteYuan1 Let us say that esp-iot-solution is being used to control a touch screen with an FT5X06 touch screen controller and that the FT5X06 is connected to I2C0. Does this mean that it's no longer possible to safely connect another I2C device to I2C0 as the FT5X06 driver will access I2C0 whenever it wants without using synchronization primitives like semaphores?

InfiniteYuan1, 2019-03-28:
Yeal, It's ok that connect another I2C device to I2C0. But should add mutual exclusion mechanism on application layer.

closed 2019-03-28 by colesnicov