espressif/esp-iot-solution #36: Esp32 capacitive touch sensitivity

bardiabarai, 2019-03-24:
Hello, I want to now about the sensitivity of  ESP31 internal capacitive touch sensor. I read this article:
https://github.com/espressif/esp-iot-solution/blob/master/documents/touch_pad_solution/touch_sensor_design_en.md
It mentioned  that between finger and electrode, It is possible to have a protective cover, But it's thickness and material are important points, Is a normal glass or plastice surface which is less than 1 mm thick, fine? I tried ESP-WROOM-32 but even with a piece of paper between finger and electrode it doesn't detect anything at all.

InfiniteYuan1, 2019-03-28:
Hi，
Do you have some picture about you question?If you have, attached it.

[3.1. Mechanical Design](https://github.com/espressif/esp-iot-solution/blob/master/documents/touch_pad_solution/touch_sensor_design_en.md#31-mechanical-design) chapter mentions how the protective cover should be selected.

If you have any question about touchpad, please create a issue based on the template.

InfiniteYuan1, 2019-04-01:
If you have further questions about this please reopen the issue in the GitHub and let's continue there.

imrozesyed, 2020-03-29:
Hi, 

I have started developing a new product with esp32, but I'm facing issue in setting the pad sensitivity by using the threshold level from 40 - 100 (till 100 its's not sending) 
After the thresh >100 all the pins are reading on its own. 
1. After placing the acrylic sheet it's not sending the touch ( 2mm -3mm thickness) 

Can anyone help me with this?

imrozesyed, 2020-03-29:
> Hello, I want to now about the sensitivity of ESP32 internal capacitive touch sensor. I read this article:
> https://github.com/espressif/esp-iot-solution/blob/master/documents/touch_pad_solution/touch_sensor_design_en.md
> It mentioned that between finger and electrode, It is possible to have a protective cover, But it's thickness and material are important points, Is a normal glass or plastice surface which is less than 1 mm thick, fine? I tried ESP-WROOM-32 but even with a piece of paper between finger and electrode it doesn't detect anything at all.

Hey Hi, 
Did you resolve the issue?

closed 2020-03-30 by BardiaBaraei