espressif/esp-iot-solution #37: LittlevGL and thread safety

fivdi, 2019-03-24:
## Environment

- Development Kit:      ESP32-DevKitC
- Kit version (for WroverKit/PicoKit/DevKitC): v4
- Module or chip used:  ESP32-WROVER-B
- IDF version (run ``git describe --tags`` to find it): esp-idf v3.1.3
- Build System:         Make
- Compiler version : 1.22.0-80-g6c4433a
- Operating System:     Linux
- Power Supply:         USB

## Problem Description

According to the [LittlevGL Documentation](https://docs.littlevgl.com/) LittlevGL is not thread-safe. Here's the relevant section of the documentation:

>LittlevGL is not thread-safe. Despite it, it's quite simple to use LittlevGL inside an operating system.

>The simple scenario is to don't use the operating system's tasks but use lv_tasks. An _lv_task_ is a function called periodically in lv_task_handler. In the _lv_task_ you can get the state of the sensors, buffers etc and call LittlevGL functions to refresh the GUI. To create an _lv_task_ use: lv_task_create(my_func, period_ms, LV_TASK_PRIO_LOWEST/LOW/MID/HIGH/HIGHEST, custom_ptr)

>If you need to use other task or threads you need one mutex which should be taken before calling lv_task_handler and released after it. In addition, you have to use to that mutex in other tasks and threads around every LittlevGL (lv_...) related code. This way you can use LittlevGL in a real multitasking environment. Just use a mutex to avoid concurrent calling of LittlevGL functions.

On the other hand, some of the LittlevGL examples in the esp-iot-solution use LittlevGL in a way that is not thread safe. They create FreeRTOS tasks that call LittlevGL functions while LittlevGL may itself be running in the background in it's own task.

For example, lvgl_example creates a FreeRTOS task [here](https://github.com/espressif/esp-iot-solution/blob/master/examples/hmi/lvgl_example/main/app_main.cpp#L146-L152) which calls LittlevGL functions in a way that is not thread safe [here](https://github.com/espressif/esp-iot-solution/blob/master/examples/hmi/lvgl_example/main/app_main.cpp#L124-L125).

Another example is lvgl_wificonfig which creates a FreeRTOS task [here](https://github.com/espressif/esp-iot-solution/blob/master/examples/hmi/lvgl_wificonfig/main/app_main.c#L790) which can end up calling LittlevGL functions in a way that is not thread safe in [refresh_wifi_list](https://github.com/espressif/esp-iot-solution/blob/master/examples/hmi/lvgl_wificonfig/main/app_main.c#L485-L505).

I had a program that occasionally crashed with errors similar to the one shown below when LittlevGL functions were called in a way that was not thread safe. Switching from a FreeRTOS task to a lv_task resolved the issue. 

If needed, I can publish a repository on GitHub that shows how to reproduce the below error quickly. The same repository also show how to resolve the issue with an lv_task.

```
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x400ec01e  PS      : 0x00060130  A0      : 0x800eb671  A1      : 0x3ffaf6a0  
0x400ec01e: lv_disp_is_mem_fill_supported at /home/brian/src/esp-idf/esp-iot-solution/components/hmi/lvgl_gui/lvgl/lv_hal/lv_hal_disp.c:228

A2      : 0x00000001  A3      : 0x3ffc3ce0  A4      : 0x0000004a  A5      : 0x3ffc3ce0  
A6      : 0x00000280  A7      : 0x3ffc87e0  A8      : 0xffffffff  A9      : 0x00000135  
A10     : 0x0000004a  A11     : 0x0000013f  A12     : 0x00000280  A13     : 0x3ffc87e0  
A14     : 0xffffffff  A15     : 0xffffffff  SAR     : 0x00000001  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x0000000f  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  

Backtrace: 0x400ec01e:0x3ffaf6a0 0x400eb66e:0x3ffaf6c0 0x400e552a:0x3ffaf6f0 0x400e6a99:0x3ffaf870 0x400e40dd:0x3ffaf890 0x400e9dae:0x3ffaf8c0 0x400e9e50:0x3ffaf910 0x400e9ee4:0x3ffaf930 0x400e9fd5:0x3ffaf960 0x400ea041:0x3ffaf990 0x400ea081:0x3ffaf9b0 0x400e7255:0x3ffaf9d0 0x400e72f3:0x3ffaf9f0 0x400e28fb:0x3ffafa10 0x400d1c33:0x3ffafa30 0x400d1c97:0x3ffafa50
0x400ec01e: lv_disp_is_mem_fill_supported at /home/brian/src/esp-idf/esp-iot-solution/components/hmi/lvgl_gui/lvgl/lv_hal/lv_hal_disp.c:228

0x400eb66e: lv_vfill at /home/brian/src/esp-idf/esp-iot-solution/components/hmi/lvgl_gui/lvgl/lv_draw/lv_draw_vbasic.c:161

0x400e552a: lv_draw_rect_main_corner at /home/brian/src/esp-idf/esp-iot-solution/components/hmi/lvgl_gui/lvgl/lv_draw/lv_draw_rect.c:316

0x400e6a99: lv_draw_rect at /home/brian/src/esp-idf/esp-iot-solution/components/hmi/lvgl_gui/lvgl/lv_draw/lv_draw_rect.c:78

0x400e40dd: lv_obj_design at /home/brian/src/esp-idf/esp-iot-solution/components/hmi/lvgl_gui/lvgl/lv_core/lv_obj.c:1495

0x400e9dae: lv_refr_obj at /home/brian/src/esp-idf/esp-iot-solution/components/hmi/lvgl_gui/lvgl/lv_core/lv_refr.c:149

0x400e9e50: lv_refr_obj_and_children at /home/brian/src/esp-idf/esp-iot-solution/components/hmi/lvgl_gui/lvgl/lv_core/lv_refr.c:149

0x400e9ee4: lv_refr_area_part_vdb at /home/brian/src/esp-idf/esp-iot-solution/components/hmi/lvgl_gui/lvgl/lv_core/lv_refr.c:149

0x400e9fd5: lv_refr_area_with_vdb at /home/brian/src/esp-idf/esp-iot-solution/components/hmi/lvgl_gui/lvgl/lv_core/lv_refr.c:149

0x400ea041: lv_refr_areas at /home/brian/src/esp-idf/esp-iot-solution/components/hmi/lvgl_gui/lvgl/lv_core/lv_refr.c:149

0x400ea081: lv_refr_task at /home/brian/src/esp-idf/esp-iot-solution/components/hmi/lvgl_gui/lvgl/lv_core/lv_refr.c:149

0x400e7255: lv_task_exec at /home/brian/src/esp-idf/esp-iot-solution/components/hmi/lvgl_gui/lvgl/lv_misc/lv_task.c:262

0x400e72f3: lv_task_handler at /home/brian/src/esp-idf/esp-iot-solution/components/hmi/lvgl_gui/lvgl/lv_misc/lv_task.c:262

0x400e28fb: lv_task_timercb at /home/brian/src/esp-idf/esp-iot-solution/components/hmi/lvgl_gui/lvgl.c:37

0x400d1c33: timer_process_alarm at /home/brian/src/esp-idf/esp-iot-solution/submodule/esp-idf/components/esp32/esp_timer.c:176

0x400d1c97: timer_task at /home/brian/src/esp-idf/esp-iot-solution/submodule/esp-idf/components/esp32/esp_timer.c:176

CPU halted.
```

fivdi, 2019-03-25:
A test case for reproducing the issue can be found at [lcd_matrix](https://github.com/fivdi/lcd_matrix).

One change was made to `lv_config.h`, the amount of memory available to lv_mem_alloc was increased from 32kB to 64kb [here](https://github.com/fivdi/lcd_matrix/blob/master/lv_conf.h#L21).

If the test program is run as is it should crash in less that one minute.

If [this line](https://github.com/fivdi/lcd_matrix/blob/master/main/main.cpp#L5) is commented out, an lv_task will be used instead of a FreeRTOS task an the program will run flawlessly.

InfiniteYuan1, 2019-03-28:
Thank you for your feedback.
We will improve the relevant documentation as soon as possible and give corresponding solutions. 
Do you have any advice？

colesnicov, 2019-03-28:
Sometime, I has have this problem too.

fivdi, 2019-03-30:
>Thank you for your feedback.

You're welcome.

>We will improve the relevant documentation as soon as possible and give corresponding solutions.

That's good news.

>Do you have any advice？

I'm not LittlevGL expert but if I understand the [documentation](https://docs.littlevgl.com/) correctly there are two options available for working with LittlevGL in a multithreaded environment.

#### Option 1

> The simple scenario is to don't use the operating system's tasks but use lv_tasks. An _lv_task_ is a function called periodically in lv_task_handler. In the _lv_task_ you can get the state of the [...]

#### Option 2

> If you need to use other task or threads you need one mutex which should be taken before calling lv_task_handler and released after it. In addition, you have to use to that mutex in other tasks and [...]

You'll have to decide which option to use.

If you choose option 1, the hmi examples using LittlevGL will need to be refactored so that they only call LittlevGL functions when this is allowed. LittlevGL functions can be called in callback functions or in an lv_task registered by calling lv_task_create. An lv_task will need to do it's work as quickly as possible and return.

An example of a callback function being registered is [this](https://github.com/espressif/esp-iot-solution/blob/master/examples/hmi/lvgl_thermostat/main/app_main.cpp#L166):
```c
 lv_btn_set_action(btn, LV_BTN_ACTION_CLICK, day_click);
```
The [day_click](https://github.com/espressif/esp-iot-solution/blob/master/examples/hmi/lvgl_thermostat/main/app_main.cpp#L299-L304) function can safely call LittlevGL functions:
```c
static lv_res_t day_click(lv_obj_t *btn)
{
    lv_img_set_src(wp, &lv_day);

    return LV_RES_OK;
}
```
An example of code that will not always function correctly can be seen in [lvgl_example](https://github.com/espressif/esp-iot-solution/tree/master/examples/hmi/lvgl_example). In lvgl_example a [FreeRTOS task is created](https://github.com/espressif/esp-iot-solution/blob/master/examples/hmi/lvgl_example/main/app_main.cpp#L146-L152) and it [calls LittlevGL functions](https://github.com/espressif/esp-iot-solution/blob/master/examples/hmi/lvgl_example/main/app_main.cpp#L118-L128) whenever it wants. This isn't a good idea and if the program is allowed to run for long enough it is likely to fail. Instead of calling xTaskCreate to create a FreeRTOS task that updates the UI an lv_task should be created by calling lv_task_create.

For example, replace [this](https://github.com/espressif/esp-iot-solution/blob/master/examples/hmi/lvgl_example/main/app_main.cpp#L146-L152) with something like this:
```c
 lv_task_create(lv_user_task, 500, LV_TASK_PRIO_MID, NULL);
```
And replace [this](https://github.com/espressif/esp-iot-solution/blob/master/examples/hmi/lvgl_example/main/app_main.cpp#L118-L128) with something like this:
```c
static void lv_user_task(void *pvParameter) {
    static uint8_t value = 0;

    value = esp_random() % 100;
    lv_chart_set_next(chart, series, value);
    lv_gauge_set_value(gauge, 0, value);
}
```
The modifications that need to made to the other examples are likely to be more complex.

InfiniteYuan1, 2019-04-01:
Thank you!

If you hava any idea about this, please free to comment!

embeddedt, 2019-11-22:
I suggest that you just wrap all the LittlevGL function calls with a global mutex lock/unlock (including the call to `lv_task_handler`). That will solve the issue and should involve little to no refactoring.

fivdi, 2019-11-28:
> I suggest that you just wrap all the LittlevGL function calls with a global mutex lock/unlock (including the call to `lv_task_handler`). That will solve the issue and should involve little to no refactoring.

If the esp-iot-solution  (Espressif IoT Library) was modified to function this way and provided access to the mutex allowing it to be locked and unlocked then this would be possible.

monkeytronics, 2020-01-23:
I'm failing to get this to work with ESP IDF. It's crashing every couple of minutes with a very similar log to that shown above!

```
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400953f0  PS      : 0x00060433  A0      : 0x80097abd  A1      : 0x3ffdbb10
0x400953f0: uxPortCompareSet at C:/msys32/home/thebi/esp/esp-idf/components/freertos/tasks.c:3507
 (inlined by) vPortCPUAcquireMutexIntsDisabledInternal at C:/msys32/home/thebi/esp/esp-idf/components/freertos/portmux_impl.inc.h:86
 (inlined by) vPortCPUAcquireMutexIntsDisabled at C:/msys32/home/thebi/esp/esp-idf/components/freertos/portmux_impl.h:98
 (inlined by) vTaskEnterCritical at C:/msys32/home/thebi/esp/esp-idf/components/freertos/tasks.c:4201

A2      : 0x9fffcffc  A3      : 0x00000004  A4      : 0x40097430  A5      : 0x3ffdc000
0x40097430: vPortYieldFromInt at C:/msys32/home/thebi/esp/esp-idf/components/freertos/portasm.S:583

A6      : 0x3ffdc330  A7      : 0x00000000  A8      : 0x0000abab  A9      : 0x0000abab
A10     : 0xb33fffff  A11     : 0x0000cdcd  A12     : 0x00060423  A13     : 0x00000001
A14     : 0x00060423  A15     : 0x00000003  SAR     : 0x0000001b  EXCCAUSE: 0x0000001d
EXCVADDR: 0x9fffcffc  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff

ELF file SHA256: 049c46fbc24d243432c19132ab89c743ca317e4a85864c52b8d0e04682304d71

Backtrace: 0x400953f0:0x3ffdbb10 0x40097aba:0x3ffdbb30 0x400858b9:0x3ffdbb50 0x400858ea:0x3ffdbb70 0x40085e3d:0x3ffdbb90 0x4000beaf:0x
3ffdbbb0 0x40096b86:0x3ffdbbd0 0x40096db8:0x3ffdbbf0 0x40085c8a:0x3ffdbc10 0x40085cb1:0x3ffdbc30 0x40085df1:0x3ffdbc60 0x4010e63f:0x3f
fdbc80 0x401087f5:0x3ffdbf40 0x401086bd:0x3ffdbf90 0x40096886:0x3ffdbfc0 0x400950ee:0x3ffdbfe0 0x4009739c:0x3ffdc000 0x4009742d:0x3ffd
c020 0x40142d70:0x3ffdc050 0x4013bd5d:0x3ffdc080 0x4013bddb:0x3ffdc0a0 0x4013bf05:0x3ffdc0d0 0x4013bf51:0x3ffdc100 0x4013bf9d:0x3ffdc1
20 0x401418fe:0x3ffdc150 0x401419d0:0x3ffdc170 0x40107a77:0x3ffdc190 0x40107bc5:0x3ffdc280 0x40096721:0x3ffdc310
0x400953f0: uxPortCompareSet at C:/msys32/home/thebi/esp/esp-idf/components/freertos/tasks.c:3507
 (inlined by) vPortCPUAcquireMutexIntsDisabledInternal at C:/msys32/home/thebi/esp/esp-idf/components/freertos/portmux_impl.inc.h:86
 (inlined by) vPortCPUAcquireMutexIntsDisabled at C:/msys32/home/thebi/esp/esp-idf/components/freertos/portmux_impl.h:98
 (inlined by) vTaskEnterCritical at C:/msys32/home/thebi/esp/esp-idf/components/freertos/tasks.c:4201

0x40097aba: multi_heap_internal_lock at C:/msys32/home/thebi/esp/esp-idf/components/heap/multi_heap.c:380
 (inlined by) multi_heap_malloc_impl at C:/msys32/home/thebi/esp/esp-idf/components/heap/multi_heap.c:416

0x400858b9: heap_caps_malloc at C:/msys32/home/thebi/esp/esp-idf/components/heap/heap_caps.c:232

0x400858ea: heap_caps_malloc_default at C:/msys32/home/thebi/esp/esp-idf/components/heap/heap_caps.c:232

0x40085e3d: _malloc_r at C:/msys32/home/thebi/esp/esp-idf/components/newlib/syscalls.c:37

0x40096b86: xQueueGenericCreate at C:/msys32/home/thebi/esp/esp-idf/components/freertos/queue.c:2038

0x40096db8: xQueueCreateMutex at C:/msys32/home/thebi/esp/esp-idf/components/freertos/queue.c:2038

0x40085c8a: lock_init_generic at C:/msys32/home/thebi/esp/esp-idf/components/newlib/locks.c:79

0x40085cb1: lock_acquire_generic at C:/msys32/home/thebi/esp/esp-idf/components/newlib/locks.c:134

0x40085df1: _lock_acquire_recursive at C:/msys32/home/thebi/esp/esp-idf/components/newlib/locks.c:171

0x4010e63f: _vfiprintf_r at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdio/../../../../.
/newlib/libc/stdio/vfprintf.c:860 (discriminator 2)

0x401087f5: fiprintf at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdio/../../../.././new
lib/libc/stdio/fiprintf.c:50

0x401086bd: __assert_func at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdlib/../../../..
/./newlib/libc/stdlib/assert.c:59 (discriminator 8)

0x40096886: vPortCPUAcquireMutexIntsDisabledInternal at C:/msys32/home/thebi/esp/esp-idf/components/freertos/port.c:403
 (inlined by) vPortCPUAcquireMutexIntsDisabled at C:/msys32/home/thebi/esp/esp-idf/components/freertos/portmux_impl.h:98
 (inlined by) vPortCPUAcquireMutex at C:/msys32/home/thebi/esp/esp-idf/components/freertos/port.c:394

0x400950ee: vTaskSwitchContext at C:/msys32/home/thebi/esp/esp-idf/components/freertos/tasks.c:3507

0x4009739c: _frxt_dispatch at C:/msys32/home/thebi/esp/esp-idf/components/freertos/portasm.S:406

0x4009742d: vPortYield at C:/msys32/home/thebi/esp/esp-idf/components/freertos/portasm.S:555

0x40142d70: ili9341_flush at C:/msys32/home/thebi/esp/esp-jumpstart/components/lvgl_ili9341/ili9341.c:130
[... 41 lines cut ...]
        drawData(data1_string, data2_string, data3_string);
        ....

}

void draw_Welcome( void ) 
{
    xSemaphoreTake(xSemaphore_LittleVGL, (TickType_t) 0);
    
    lv_ code....
    
    xSemaphoreGive(xSemaphore_LittleVGL);
}

void drawData( void ) 
{
    xSemaphoreTake(xSemaphore_LittleVGL, (TickType_t) 0);
    
    lv_ code....
    
    xSemaphoreGive(xSemaphore_LittleVGL);
}
```

So, I'm trying to wrap all the lv_.... code in a mutex, as per the guidance, but doesn't seem to be doing the business. Does it look like the same failure mechanism, or maybe a red herring? Appreciate any thoughts...

fivdi, 2020-01-24:
@monkeytronics The code posted doesn't appear to be based on esp-iot-solution. As such, you may be experiencing an issue similar to the one for which this ticket was originally opened but it will not be the same issue.