espressif/esp-iot-solution #38: lvgl_example does not activate backlight of LCD

timredfern, 2019-05-02:
## Environment

- Development Kit: ESP32-Wrover-Kit
- Kit version (for WroverKit): v3
- Module or chip used: ESP32-WROVER
- IDF version v3.1.3
- Build System:  Make
- Compiler version : 1.22.0-80-g6c4433a
- Operating System: macOS
- Power Supply: USB

## Problem Description

lvgl_example does not activate backlight of LCD

### Expected Behavior

backlight would turn on to view example

### Actual Behavior

screen remains black

### Steps to reproduce

set correct options in menuconfig:

 (25) LittlevGL LCD MISO GPIO    
 (23) LittlevGL LCD MOSI GPIO  
(19) LittlevGL LCD CLK GPIO     
(22) LittlevGL LCD CS GPIO    
(21) LittlevGL LCD DC GPIO  
(18) LittlevGL LCD RESET GPIO  
(5) LittlevGL LCD BL GPIO     

& disable touch screen options

### Code to reproduce this issue

If I add this line to app_main.cpp, after lvgl_init();, the screen lights up:

    ///Enable backlight
    gpio_set_level((gpio_num_t)CONFIG_LVGL_LCD_BL_GPIO, 0);

## Debug Logs

## Other items if possible

InfiniteYuan1, 2019-05-05:
Hi,
You can set backlight active level in menuconfig as follow:
[image]
[image]

timredfern, 2019-05-07:
ah, LOW is on. 

Sorry for the confusion. It might be an idea to make LOW the default

closed 2019-05-07 by timredfern