espressif/esp-iot-solution #39: ssd1306 oled on i2c not showing up ( esp Azure iot kit )

manjuk41, 2019-05-02:
I used  esp32_azure_iot_kit example code from this repo. 
I tried using esp_tool to scan for available devices,  Except OLED ( addr: 0x3c ) other sensors are showing up. Below is the result on i2cdetect

 Scanning I2C Addresses
.. .. .. .. .. .. .. .. .. .. .. .. .. .. 0E .. 
.. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. 
.. .. .. 23 .. .. .. .. .. .. .. .. .. .. .. .. 
.. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. 
.. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. 
.. .. .. .. .. .. .. .. .. .. .. .. .. .. .. 5F 
.. .. .. .. .. .. .. .. 68 .. .. .. .. 6D .. .. 
.. .. .. .. .. .. .. .. .. .. .. .. .. .. ..

InfiniteYuan1, 2019-05-05:
Hi,
I don't understand your issue through your comments.Please describe your issue by template issue.

manjuk41, 2019-05-05:
@InfiniteYuan1 I dont know about template issue

InfiniteYuan1, 2019-05-05:
[new issue](https://github.com/espressif/esp-iot-solution/issues/new)

manjuk41, 2019-05-07:
@InfiniteYuan1 new issue is what i created.

Board : Esp Azure IOT kit
when i scan for i2c device using i2ctool, i am getting this now

esp32> i2cconfig --port=1 --freq=100000 --sda=25 --scl=26

esp32> i2cdetect
        0    1    2    3    4     5    6    7    8    9     a    b    c    d    e     f
00: UU UU UU UU UU UU UU UU UU UU UU UU UU UU UU UU 
10: UU UU UU UU UU UU UU UU UU UU UU UU UU UU UU UU 
20: UU UU UU UU UU UU UU UU UU UU UU UU UU UU UU UU 
30: UU UU UU UU UU UU UU UU UU UU UU UU UU UU UU UU 
40: UU UU UU UU UU UU UU UU UU UU UU UU UU UU UU UU 
50: UU UU UU UU UU UU UU UU UU UU UU UU UU UU UU UU 
60: UU UU UU UU UU UU UU UU UU UU UU UU UU UU UU UU 
70: UU UU UU UU UU UU UU UU UU UU UU UU UU UU UU UU 

esp32> i2cget -c 0x5F -r 0x0F
W (136645) cmd_i2ctools: Bus is busy

ChristosPalisidis, 2019-08-06:
Hello,
i am using the same board, and i want to have access to the sensors' data. I am trying to make it through the I2C bus. Although, the scanning proccess on I2C does not detect any device.
I am coding in arduino IDE. Is there any advices?
Thank you!

manjuk41, 2019-08-07:
@ChristosPalisidis i had exactly the same issue. If using esp-idf there will not be any issues, device works properly. 
Have you set the SDA and SCL pin correctly?

ChristosPalisidis, 2019-08-07:
Yes, i print the SDA and SCL pins so as to be sure, and i use these printed pins for the I2C interface.
I made a lot of progress and development of my project, working on ARDUINO IDE, so a migration to ESP-IDF at this point, would be very time consuming. Is this possible to continue with ESP-IDF and use my current work as component, or something like that?
Thank you for your response @manjuk41 !

manjuk41, 2019-08-07:
@ChristosPalisidis I don't know.

tawalke, 2019-08-07:
Hi, I have this board and I am having the same issue.  I am using the sample as is without any changes as yet but I am planning to enhance this application to write to Azure IoT using the esp-azure sdk.  But before I do this I need to ensure that the board and all sensors are working correctly.  Looking at the monitor the application is working correctly and I can see the data being captured from sensors and shown in monitor.  The issue is that it is not showing up on the OLED. Note before I used the example from this repo the OLED was working and displaying data/output.

tawalke, 2019-08-07:
Running i2cdetect gives me: 
[image]

tawalke, 2019-08-07:
@InfiniteYuan1 I did open a case: https://github.com/espressif/esp-iot-solution/issues/49 this seems to be pattern of an issue with the OLED/ss1306.  Please have someone look into it as these tests are crucial to some of our work related releases.

ChristosPalisidis, 2019-08-08:
I personally figured it out. It was the pins settings finally @manjuk41 . I will be more carefull with the datasheets next time!

tawalke, 2019-08-08:
@ChristosPalisidis I wish mine where that easy. Pins were set correctly. Hoping they review this issue and/or mine for resolution.

manjuk41, 2019-08-08:
@tawalke Same issue here. Pins are all set correctly. 
Do you have any idea on how to debug with a logic analyzer?

closed 2023-07-10 by leeebo