espressif/esp-iot-solution #40: assert fails for nvs_get_blob() in lvgl_wificonfig example

timredfern, 2019-06-06:
## Environment

- Development Kit:      [ESP32-DevKitC]
- Kit version (for DevKitC): [v4]
- Module or chip used:  [ESP32-WROOM-32]
- IDF version (run ``git describe --tags`` to find it): 
    // v3.1.3
- Build System:         [Make]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // 5.2.0
- Operating System:     [macOS]
- Power Supply:         [USB]

## Problem Description

assert failed running lvgl_wificonfig example - fails to load blob from NVS

### Expected Behavior

should be possible to build and run the example

### Actual Behavior

firmware halts with assert failure:

E (841) param: /Users/tim/embedded/esp/esp-iot-solution/components/general/param/param.c:61 (iot_param_load)

### Steps to repropduce

make defconfig
make menuconfig 
- set serial params
make erase_flash flash simple_monitor

### Code to reproduce this issue

## Debug Logs

Hard resetting via RTS pin...
--- forcing DTR inactive
--- forcing RTS inactive
--- Miniterm on /dev/tty.SLAB_USBtoUART  115200,8,N,1 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
I (202) esp_image: segment 1: paddr=0x0005c1ec vaddr=0x3ffb0000 size=0x032a0 ( 12960) load
I (208) esp_image: segment 2: paddr=0x0005f494 vaddr=0x40080000 size=0x00400 (  1024) load
I (210) esp_image: segment 3: paddr=0x0005f89c vaddr=0x40080400 size=0x00774 (  1908) load
I (219) esp_image: segment 4: paddr=0x00060018 vaddr=0x400d0018 size=0x7db34 (514868) map
I (407) esp_image: segment 5: paddr=0x000ddb54 vaddr=0x40080b74 size=0x1001c ( 65564) load
I (445) boot: Loaded app from partition at offset 0x10000
I (445) boot: Disabling RNG early entropy source...
I (446) cpu_start: Pro cpu up.
I (449) cpu_start: Starting app cpu, entry point is 0x40080e48
I (0) cpu_start: App cpu up.
I (460) heap_init: Initializing. RAM available for dynamic allocation:
I (466) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (472) heap_init: At 3FFD09F8 len 0000F608 (61 KiB): DRAM
I (478) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (485) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (491) heap_init: At 40090B90 len 0000F470 (61 KiB): IRAM
I (497) cpu_start: Pro cpu start user code
I (68) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (473) gpio: GPIO[5]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
E (841) param: /Users/tim/embedded/esp/esp-iot-solution/components/general/param/param.c:61 (iot_param_load)

## Other items if possible

timredfern, 2019-06-11:
I tried the other lvgl examples.

I see the NVS read assert fail and halt with lvgl_example also.

lvgl_coffee doesn't build:

esp-iot-solution/examples/hmi/lvgl_coffee/main/app_main.cpp:362:64: error: invalid conversion from 'void (*)(lv_obj_t*, uint16_t) {aka void (*)(_lv_obj_t*, short unsigned int)}' to 'lv_tabview_action_t {aka unsigned char (*)(_lv_obj_t*, short unsigned int)}' [-fpermissive]

the Makefile in lvgl_thermostat is broken but it builds with the Makefile from lvgl_example.

When it runs it also halts with the assert failure.

InfiniteYuan1, 2019-06-18:
Thanks，I‘ll fix it as soon as possible.

InfiniteYuan1, 2019-06-20:
> E (841) param: /Users/tim/embedded/esp/esp-iot-solution/components/general/param/param.c:61 (iot_param_load)

This error indicates that there is no calibration data in the flash. Please recalibrate on the screen.

> esp-iot-solution/examples/hmi/lvgl_coffee/main/app_main.cpp:362:64: error: invalid conversion from 'void ()(lv_obj_t, uint16_t) {aka void ()(_lv_obj_t, short unsigned int)}' to 'lv_tabview_action_t {aka unsigned char ()(_lv_obj_t, short unsigned int)}' [-fpermissive]

Please modify the following::
```
static lv_res_t tabview_load_action(lv_obj_t *tabview, uint16_t tab_action_id)
{
    tab_id = tab_action_id;
    return LV_RES_OK;
}
```

Aiouan, 2019-06-22:
Thanks Yuan, I was having the same issue, the above edit fixed it.

Xiehanxin, 2020-07-27:
@timredfern  hello, If your issue is now resolved, can you please close the issue. Thank you.

ajsb85, 2021-10-02:
```
--- WARNING: GDB cannot open serial ports accessed as COMx
--- Using \\.\COM16 instead...
--- idf_monitor on \\.\COM16 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6992
load:0x40078000,len:14308
ho 0 tail 12 room 4
load:0x40080400,len:3716
0x40080400: _init at ??:?

entry 0x40080680
I (29) boot: ESP-IDF v4.3.1 2nd stage bootloader
I (29) boot: compile time 05:54:31
I (29) boot: chip revision: 3
I (32) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (39) boot.esp32: SPI Speed      : 40MHz
I (44) boot.esp32: SPI Mode       : DIO
I (48) boot.esp32: SPI Flash Size : 4MB
I (53) boot: Enabling RNG early entropy source...
I (58) boot: Partition Table:
I (62) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (76) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (84) boot:  2 factory          factory app      00 00 00010000 00200000
I (91) boot: End of partition table
I (96) boot_comm: chip revision: 3, min. application chip revision: 0
I (103) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=137e0h ( 79840) map
I (140) esp_image: segment 1: paddr=00023808 vaddr=3ffb0000 size=02ea8h ( 11944) load
I (145) esp_image: segment 2: paddr=000266b8 vaddr=40080000 size=09960h ( 39264) load
I (162) esp_image: segment 3: paddr=00030020 vaddr=400d0020 size=573b4h (357300) map
I (292) esp_image: segment 4: paddr=000873dc vaddr=40089960 size=04fach ( 20396) load
I (300) esp_image: segment 5: paddr=0008c390 vaddr=50000000 size=00010h (    16) load
I (308) boot: Loaded app from partition at offset 0x10000
I (308) boot: Disabling RNG early entropy source...
I (322) cpu_start: Pro cpu up.
I (322) cpu_start: Starting app cpu, entry point is 0x40081344
0x40081344: call_start_cpu1 at C:/Users/ajsb85/.esp-idf/components/esp_system/port/cpu_start.c:141

I (0) cpu_start: App cpu up.
I (337) cpu_start: Pro cpu start user code
I (337) cpu_start: cpu freq: 160000000
I (337) cpu_start: Application information:
I (341) cpu_start: Project name:     lvgl_example
I (347) cpu_start: App version:      87e76fd-dirty
I (352) cpu_start: Compile time:     Oct  2 2021 05:54:16
I (358) cpu_start: ELF file SHA256:  6e562ba2938e45b2...
I (364) cpu_start: ESP-IDF:          v4.3.1
I (369) heap_init: Initializing. RAM available for dynamic allocation:
I (376) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (382) heap_init: At 3FFB3F50 len 0002C0B0 (176 KiB): DRAM
I (389) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (395) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (401) heap_init: At 4008E90C len 000116F4 (69 KiB): IRAM
I (409) spi_flash: detected chip: generic
I (412) spi_flash: flash io: dio
I (417) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (434) spi_bus: SPI2 bus created
I (437) Board: Board Info: ESP32-LCDKit
I (441) Board: Board Init Done ...
I (446) spi_bus: SPI2 bus device added, CS=5 Mode=0 Speed=20000000
I (873) lcd ili9341: MADCTL=28
I (873) spi_bus: SPI2 bus device added, CS=32 Mode=0 Speed=10000000
I (873) XPT2046: Touch panel size width: 240, height: 320
I (878) XPT2046: Initial successful | GPIO INT:-1 | GPIO CS:32 | dir:4
E (989) param save: param_load(58): nvs get blob failed
I (1132) lcd ili9341: MADCTL=28
I (1149) lcd ili9341: MADCTL=8
```

ajsb85, 2021-10-02:
Dearest.
I got the same issue even with the changes of the branch 1.1

```console
I (607) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (1026) gpio: GPIO[5]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
E (1351) param: /home/user/esp/esp-iot-solution/components/general/param/param.c:61 (iot_param_load)
```

closed 2023-07-10 by leeebo