espressif/esp-iot-solution #41: unit-test-app don't work.

nopnop2002, 2019-06-12:
I tried unit-test-app, but error occure.

## Environment
	freeRTOS version:V8.2.0
	NEWLIB version:2.2.0
	lwIP version:2-0-3-255
	ESP-IDF version:v4.0-dev-311-g70eda3d22-dirty

## Problem Description

unit-test-app don't work.

## Steps to repropduce

	$ cd MY_IOT_SOLUTION_PATH/tools/unit-test-app
	$ make erase_flash
	$ make defconfig
	$ make flash IOT_TEST_ALL=1 monitor

	--- idf_monitor on /dev/ttyUSB0 115200 ---
	--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
	Enabling RNG e?ets Jun  8 2016 00:22:57

	rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
	configsip: 0, SPIWP:0xee
	clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
	mode:DIO, clock div:2
	load:0x3fff0018,len:4
	load:0x3fff001c,len:5888
	ho 0 tail 12 room 4
	load:0x40078000,len:9188
	load:0x40080000,len:6084
	0x40080000: _WindowOverflow4 at /home/robotics/esp-iot-solution/submodule/esp-idf/components/freertos/xtensa_vectors.S:1779

	entry 0x4008032c
	0x4008032c: _KernelExceptionVector at ??:?

	I (30) boot: ESP-IDF v3.1.3 2nd stage bootloader
	I (30) boot: compile time 10:12:34
	I (30) boot: Enabling RNG early entropy source...
	I (34) boot: SPI Speed      : 40MHz
	I (39) boot: SPI Mode       : DIO
	I (43) boot: SPI Flash Size : 4MB
	I (47) boot: Partition Table:
	I (50) boot: ## Label            Usage          Type ST Offset   Length
	I (58) boot:  0 nvs              WiFi data        01 02 00009000 00004000
	I (65) boot:  1 otadata          OTA data         01 00 0000d000 00002000
	I (73) boot:  2 phy_init         RF data          01 01 0000f000 00001000
	I (80) boot:  3 ota_0            OTA app          00 10 00010000 00100000
	I (87) boot:  4 ota_1            OTA app          00 11 00110000 00100000
	I (95) boot: End of partition table
	I (99) boot: No factory image, trying OTA 0
	I (104) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x269354 (2528084) map
	I (998) esp_image: segment 1: paddr=0x0027937c vaddr=0x3ffc0000 size=0x04350 ( 17232) load
	I (1005) esp_image: segment 2: paddr=0x0027d6d4 vaddr=0x40080000 size=0x00400 (  1024) load
	0x40080000: _WindowOverflow4 at /home/robotics/esp-iot-solution/submodule/esp-idf/components/freertos/xtensa_vectors.S:1779

	I (1006) esp_image: segment 3: paddr=0x0027dadc vaddr=0x40080400 size=0x02534 (  9524) load
	I (1018) esp_image: segment 4: paddr=0x00280018 vaddr=0x400d0018 size=0xfdc08 (1039368) map
	0x400d0018: _flash_cache_start at ??:?

	I (1387) esp_image: segment 5: paddr=0x0037dc28 vaddr=0x40082934 size=0x17580 ( 95616) load
	0x40082934: _xt_medint3 at /home/robotics/esp-iot-solution/submodule/esp-idf/components/freertos/xtensa_vectors.S:1323

	I (1427) esp_image: segment 6: paddr=0x003951b0 vaddr=0x400c0000 size=0x00064 (   100) load
	I (1427) esp_image: segment 7: paddr=0x0039521c vaddr=0x50000200 size=0x00008 (     8) load
	E (1434) esp_image: Image length 3691088 doesn't fit in partition length 1048576
	E (1442) boot: OTA app partition slot 0 is not bootable
	E (1448) esp_image: image at 0x110000 has invalid magic byte
	W (1454) esp_image: image at 0x110000 has invalid SPI mode 141
	W (1461) esp_image: image at 0x110000 has invalid SPI size 7
	E (1467) boot: OTA app partition slot 1 is not bootable
	E (1473) boot: No bootable app partitions in the partition table
	Fatal exception (0): IllegalInstruction
	epc1=0x40080354, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000
	0x40080354: _UserExceptionVector at ??:?

InfiniteYuan1, 2019-06-20:
please reflash bin :
```
$ make erase_flash
$ make defconfig
$ make IOT_TEST_ALL=1 
$ make flash monitor
```

nopnop2002, 2019-06-20:
Thank you for reply.
I tried your step.
But unit-test-app don't work.
I think No Factory partition is the cause.

	I (30) boot: ESP-IDF v3.1.3 2nd stage bootloader
	I (30) boot: compile time 13:13:10
	I (30) boot: Enabling RNG early entropy source...
	I (34) boot: SPI Speed      : 40MHz
	I (38) boot: SPI Mode       : DIO
	I (42) boot: SPI Flash Size : 16MB
	I (46) boot: Partition Table:
	I (50) boot: ## Label            Usage          Type ST Offset   Length
	I (57) boot:  0 nvs              WiFi data        01 02 00009000 00004000
	I (65) boot:  1 otadata          OTA data         01 00 0000d000 00002000
	I (72) boot:  2 phy_init         RF data          01 01 0000f000 00001000
	I (80) boot:  3 ota_0            OTA app          00 10 00010000 00100000
	I (87) boot:  4 ota_1            OTA app          00 11 00110000 00100000
	I (95) boot: End of partition table
	I (99) boot: No factory image, trying OTA 0
	I (104) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x269354 (2528084) map
	I (998) esp_image: segment 1: paddr=0x0027937c vaddr=0x3ffc0000 size=0x04350 ( 17232) load
	I (1005) esp_image: segment 2: paddr=0x0027d6d4 vaddr=0x40080000 size=0x00400 (  1024) load
	0x40080000: _WindowOverflow4 at /home/robotics/esp-iot-solution/submodule/esp-idf/components/freertos/xtensa_vectors.S:1779

	I (1006) esp_image: segment 3: paddr=0x0027dadc vaddr=0x40080400 size=0x02534 (  9524) load
	I (1018) esp_image: segment 4: paddr=0x00280018 vaddr=0x400d0018 size=0xfdc08 (1039368) map
	0x400d0018: _flash_cache_start at ??:?

	I (1387) esp_image: segment 5: paddr=0x0037dc28 vaddr=0x40082934 size=0x17580 ( 95616) load
	0x40082934: _xt_medint3 at /home/robotics/esp-iot-solution/submodule/esp-idf/components/freertos/xtensa_vectors.S:1323

	I (1427) esp_image: segment 6: paddr=0x003951b0 vaddr=0x400c0000 size=0x00064 (   100) load
	I (1427) esp_image: segment 7: paddr=0x0039521c vaddr=0x50000200 size=0x00008 (     8) load
	E (1434) esp_image: Image length 3691088 doesn't fit in partition length 1048576
	E (1442) boot: OTA app partition slot 0 is not bootable
	E (1448) esp_image: image at 0x110000 has invalid magic byte
	W (1454) esp_image: image at 0x110000 has invalid SPI mode 141
	W (1460) esp_image: image at 0x110000 has invalid SPI size 7
	E (1467) boot: OTA app partition slot 1 is not bootable
	E (1473) boot: No bootable app partitions in the partition table
	Fatal exception (0): IllegalInstruction
	epc1=0x40080354, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000
	0x40080354: _UserExceptionVector at ??:?

InfiniteYuan1, 2019-06-20:
Please provide you solution commit id.or you can update lastest version.

I tried the unit test is ok.

I assume your app bin too larger to hard for app partition config.

`Image length 3691088 doesn't fit in partition length 1048576`

so you can try following:

```
make TEST_COMPONENTS="bh1750"
make flash monitor
```

nopnop2002, 2019-06-21:
Hello.

My Environment:
ESP-IDF version:v4.0-dev-311-g70eda3d22-dirty

I renamed my esp-idf to other name.

	$mv esp-idf esp-idf-back

I installed esp-idf newly.
But i got make error.

	$ make
	The following Python requirements are not satisfied:
	pyparsing>=2.0.3,<2.4.0
	Please refer to the Get Started section of the ESP-IDF Programming Guide for setting up the required packages.
	Alternatively, you can run "/usr/bin/python -m pip install --user -r /home/robotics/esp-idf/requirements.txt" for resolving the issue.

I execute this command.
But make error don't fix.
/usr/bin/python -m pip install --user -r /home/robotics/esp-idf/requirements.txt

So i renamed old one.

	$mv esp-idf esp-idf-new
	$mv esp-idf-back esp-idf

I tried this.

	make TEST_COMPONENTS="bh1750"
	make flash monitor

Monitor output:

	Here's the test menu, pick your combo:
	(1)     "Sensor BH1750 test" [bh1750][iot][sensor]
	(2)     "Sensor BH1750 obj test" [bh1750_cpp][iot][sensor]

InfiniteYuan1, 2019-06-21:
Yeah, This‘s right.

Reason:

Image length 3691088 doesn't fit in partition length 1048576

nopnop2002, 2019-06-24:
I understood that we can't to work unit-test-app with IOT_TEST_ALL=1.
If my understand is right, please close this issues.
Thank you for your support.

Xiehanxin, 2020-07-27:
@nopnop2002  hello. If your issue is now resolved, can you please close the issue. Thank you.

nopnop2002, 2020-07-27:
Thank you for your help.

closed 2020-07-27 by nopnop2002