espressif/esp-iot-solution #42: lvgl_example current problem

InfiniteYuan1, 2019-06-20:
> /esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: /esp-iot-solution/examples/hmi/lvgl_example/build/lvgl_example.elf section `.dram0.bss' will not fit in region `dram0_0_seg'
/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: region `dram0_0_seg' overflowed by 10072 bytes
collect2: error: ld returned 1 exit status
/esp-iot-solution//submodule/esp-idf//make/project.mk:406: recipe for target '/esp-iot-solution/examples/hmi/lvgl_example/build/lvgl_example.elf' failed
make: *** [/esp-iot-solution/examples/hmi/lvgl_example/build/lvgl_example.elf] Error 1

when you encounter this problem, please modify the following:

`lv_conf.h`:
```
/*================
 *  THEME USAGE
 *================*/
#define LV_THEME_LIVE_UPDATE    1       /*1: Allow theme switching at run time. Uses 8..10 kB of RAM*/

#define USE_LV_THEME_TEMPL      0       /*Just for test*/
#define USE_LV_THEME_DEFAULT    1       /*Built mainly from the built-in styles. Consumes very few RAM*/
#define USE_LV_THEME_ALIEN      1       /*Dark futuristic theme*/
#define USE_LV_THEME_NIGHT      0       /*Dark elegant theme*/
#define USE_LV_THEME_MONO       0       /*Mono color theme for monochrome displays*/
#define USE_LV_THEME_MATERIAL   0       /*Flat theme with bold colors and light shadows*/
#define USE_LV_THEME_ZEN        0       /*Peaceful, mainly light theme */
#define USE_LV_THEME_NEMO       0       /*Water-like theme based on the movie "Finding Nemo"*/

/*==================
 *    FONT USAGE
 *===================*/

/* More info about fonts: https://docs.littlevgl.com/#Fonts
 * To enable a built-in font use 1,2,4 or 8 values
 * which will determine the bit-per-pixel. Higher value means smoother fonts */
#define USE_LV_FONT_DEJAVU_10              0
#define USE_LV_FONT_DEJAVU_10_LATIN_SUP    0
#define USE_LV_FONT_DEJAVU_10_CYRILLIC     0
#define USE_LV_FONT_SYMBOL_10              0

#define USE_LV_FONT_DEJAVU_20              4
#define USE_LV_FONT_DEJAVU_20_LATIN_SUP    4
#define USE_LV_FONT_DEJAVU_20_CYRILLIC     4
#define USE_LV_FONT_SYMBOL_20              4

#define USE_LV_FONT_DEJAVU_30              0
#define USE_LV_FONT_DEJAVU_30_LATIN_SUP    0
#define USE_LV_FONT_DEJAVU_30_CYRILLIC     0
#define USE_LV_FONT_SYMBOL_30              0

#define USE_LV_FONT_DEJAVU_40              0
#define USE_LV_FONT_DEJAVU_40_LATIN_SUP    0
#define USE_LV_FONT_DEJAVU_40_CYRILLIC     0
#define USE_LV_FONT_SYMBOL_40              0
```

`lv_test_theme.c`
```
/**********************
 *   STATIC FUNCTIONS
 **********************/
static lv_res_t roller_action(lv_obj_t * roller)
{
    lv_theme_t *th = lv_theme_alien_init(100, NULL);
    switch (lv_ddlist_get_selected(roller))
    {
        case 0:
            th = lv_theme_default_init(100, NULL);
            break;
    
        case 1:
            th = lv_theme_alien_init(100, NULL);
            break;
    
        case 2:
            // th = lv_theme_night_init(100, NULL);
            break;
    
        case 3:
            // th = lv_theme_mono_init(100, NULL);
            break;
    
        case 4:
            // th = lv_theme_material_init(100, NULL);
            break;
    
        case 5:
            // th = lv_theme_zen_init(100, NULL);
            break;
    
        case 6:
            // th = lv_theme_nemo_init(100, NULL);
            break;
    
        default:
            break;
    }
    lv_theme_set_current(th);
    return LV_RES_OK;
}
```

linyaqiang, 2019-08-17:
I want to ask you the lvgl_example can be builded on idf 4.0 / gcc 8.2 tool chain? but I build it then report some error.

darrinbolden, 2019-09-16:
Has this issue been resolved? appears to be open. If solution found please share. I have encountered the same issue.
Thanks

linyaqiang, 2019-09-17:
You try the lower version of the compilation tool, I can use V5.2 version of the tool to compile successfully, with V8.2 version is not successful compilation.

leeebo, 2023-07-10:
The ESP-IDF has integrated LVGL from v5.0, please use the component https://components.espressif.com/components/espressif/esp_lvgl_port instead, and you can find more examples in https://github.com/espressif/esp-idf/tree/master/examples/peripherals/lcd and https://github.com/espressif/esp-bsp/tree/master/esp32_s3_lcd_ev_board

closed 2023-07-10 by leeebo