espressif/esp-iot-solution #43: Issue getting temperature from hts221 sensor (ESP32-Azure IoT KiT)

carlosalvite, 2019-06-26:
I'm running the provided example to test the hts221 using ESP32-Azure IoT KiT but the temperature and humidity value is always 0.00 and I cannot see any error message in the console

It is necessary to execute a previous configuration before running the example?

Thanks in advance

carlosalvite, 2019-06-26:
hts221 device ID is: 00
avg_h is: 00
avg_t is: 00
odr is: 00
bdu_status is: 00
heater_status is: 00
irq_level is: 00
irq_output_type is: 00
irq_enable is: 00

********HTS221 HUMIDITY&TEMPERATURE SENSOR********
humidity value is: 0.00
temperature value is: 0.00
**************************************************
heap: 301184

fivdi, 2019-06-30:
It looks like the goal is to get [this test](https://github.com/espressif/esp-iot-solution/blob/master/components/i2c_devices/sensor/hts221/test/hts221_test.c) to function on an ESP32-Azure IoT KiT. If so, were the values of `I2C_MASTER_SCL_IO` and `I2C_MASTER_SDA_IO` [here](https://github.com/espressif/esp-iot-solution/blob/master/components/i2c_devices/sensor/hts221/test/hts221_test.c#L21-L22) changed to the appropriate values? To function on an ESP32-Azure IoT KiT these values will need to be changed from 21 and 15 to 26 and 25 respectively (see [here](https://github.com/espressif/esp-iot-solution/blob/master/examples/esp32_azure_iot_kit/main/main.c#L27-L28).)

Does the [esp32_azure_iot_kit example](https://github.com/espressif/esp-iot-solution/tree/master/examples/esp32_azure_iot_kit) function correctly and display the correct humidity and temperature on an ESP32-Azure IoT KiT?

ChristosPalisidis, 2019-07-30:
Hello,
I also want to use this library, but i get a problem even trying to include it in the libraries folder. I get an "no headers files (.h) found in C:\......\Arduino\libraries\esp-iot-solution" message. Can you tell me how did you manage to run the example?
Thank you

fivdi, 2019-07-30:
Information about usage of the library and building examples is provided in README.md. To the best of my knowledge the library isn't designed for usage with the Arduino IDE.

ChristosPalisidis, 2019-07-31:
So which platform do you use @fivdi ? Although i think it is not the issue.
If someone could clarify this, it would be very helpful.
Thank you for your answer.

fivdi, 2019-07-31:
> So which platform do you use @fivdi

esp-idf.

The version of esp-idf used by esp-iot-solution is included in the esp-iot-solution as a submodule. See [here](https://github.com/espressif/esp-iot-solution/tree/master/submodule).

closed 2023-07-10 by leeebo