espressif/esp-iot-solution #48: i2c bus usage in arduino IDE (esp32 azure IoT kit)

ChristosPalisidis, 2019-08-06:
Hello,
I would like to have access to the  sensors' data of esp32 azure IoT kit. I am trying to make it through the I2C bus. Although, the scanning proccess on I2C does not detect any device.
I am coding in arduino IDE. Is there any advices?
Thank you!

colesnicov, 2019-08-07:
Hello fellow. This way I use i2c on esp8266 with esp-iot:

Routine for initialize I2C bus:

```
 
#define OLED_IIC_NUM        I2C_NUM_0
#define OLED_IIC_SCL_NUM    4
#define OLED_IIC_SDA_NUM    5

static void i2cBusInit() {
	m_i2c_bus = new CI2CBus((i2c_port_t) OLED_IIC_NUM,
			(gpio_num_t) OLED_IIC_SCL_NUM, (gpio_num_t) OLED_IIC_SDA_NUM,
			100000);
}
```

Routinne for scanning I2C addresses:
```
void i2cScanner(i2c_port_t i2c_port) {
	uint8_t address;
	printf("     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f\r\n");
	for (int i = 0; i < 128; i += 16) {
		printf("%02x: ", i);
		for (int j = 0; j < 16; j++) {
			fflush(stdout);
			address = i + j;
			i2c_cmd_handle_t cmd = i2c_cmd_link_create();
			i2c_master_start(cmd);
			i2c_master_write_byte(cmd, (address << 1) | I2C_MASTER_WRITE,
			true);
			i2c_master_stop(cmd);
			esp_err_t ret = i2c_master_cmd_begin(i2c_port, cmd,
					(TickType_t) (50 / portTICK_RATE_MS));
			i2c_cmd_link_delete(cmd);
			if (ret == ESP_OK) {
				printf("%02x ", address);
			} else if (ret == ESP_ERR_TIMEOUT) {
				printf("UU ");
			} else {
				printf("-- ");
			}
		}
		printf("\r\n");
	}
}
```

And this is the mr. Task what can do it:

```
void task_i2c_scanner(void *ignore) {
    i2c_bus_t* i2c_bus = (i2c_bus_t*) bus;
    i2cScanner(i2c_bus->i2c_port); // Simillar to => i2cScanner(I2C_NUM_0);
    vTaskDelete(NULL);
}
```
And sequence to start:
```
 i2cBusInit() ;
vTaskCreate(&task_i2c_scanner, ...);
vTaskCreate(&task_ssd1306, ...);
vTaskCreate( ... );
...
```

For now, I can to use APDS-9960 and ssd1306. 
BME280 so far (although successfully initialize) does not display the value but I plan to solve it. However, I had to rewrite something somewhat to make it work for ESP8266 with ESP8266_RTOS_SDK.

**But now I think how best to make I2CBUS thread safe...**

colesnicov, 2019-08-07:
If you are interesting about thread-safe, [see](https://github.com/espressif/esp-iot-solution/issues/35)

ChristosPalisidis, 2019-08-08:
Thank you @colesnicov for you reply.
I managed to reach the solution, being more carefull, reading the datasheets, and creating the I2C inteface with the right way, as far as the pins and the adresses concern. I used the Wire Library.

closed 2019-08-08 by ChristosPalisidis