espressif/esp-iot-solution #49: No output on OLED on i2c using ESP32-Azure-IoT Kit (ESP32-WROVER-B)

tawalke, 2019-08-07:
----------------------------- Delete below -----------------------------

If your issue is a general question, starts similar to "How do I..", or is related to 3rd party development kits/libs, please discuss this on our community forum at esp32.com instead.

INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [ ] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/stable/) and the issue is not addressed there.
- [ ] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [ ] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs on the monitor](https://docs.espressif.com/projects/esp-idf/en/stable/get-started/idf-monitor.html#automatically-decoding-addresses) or the [coredump](https://docs.espressif.com/projects/esp-idf/en/stable/api-guides/core_dump.html).
4. Provide more items under **Other items if possible** can help us better locate your problem.
5. Use markup (buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

----------------------------- Delete above -----------------------------

## Environment

- Development Kit:      ESP32-Azure IoT Kit
- Kit version: 
- Module or chip used: ESP32-Wrover-B
- IDF version (run ``git describe --tags`` to find it): v3.2-2
- Build System:         Make
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): 
xtensa-esp32-elf-gcc (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a) 5.2.0
- Operating System:     Linux (via WSL) 
- Power Supply:         USB

## Problem Description

Hi, I have the Azure IoT board and I am having an issue with the OLED displaying the output from the example application from esp32_azure_iot_kit from the examples in the esp-iot-solution repo. I am using the sample as is without any changes as yet but I am planning to enhance this application to write to Azure IoT using the esp-azure sdk. 

Before I do this I need to ensure that the board and all sensors are working correctly. Looking at the monitor the application is working correctly and I can see the data being captured from sensors and shown in monitor. The issue is that this data  it is not showing up on the OLED. Please Note before I used the example from this repo and powered the board the OLED was working and displaying data/output as it should. It was only when I started using the example so that I could develop against it (have not as yet) did I start having issues. 

I have run i2cdetect and it is not showing anything.  (see screenshots I have attached)

### Expected Behavior
Application to run and sensor data to display on OLED

### Actual Behavior
Application runs but no sensor data is displayed on OLED

### Steps to repropduce

1. Clone esp-iot-solution repo and make flash monitor application
2. Review monitor to see data is coming through accurately, push button to see that application changes to different sensor data and monitor displays data
3. Notice that no data is displayed at all on OLED. 

// It helps if you attach a picture of your setup/wiring here.

### Code to reproduce this issue

```cpp
// the code should be wrapped in the ```cpp tag so that it will be displayed better.
#include "esp_log.h"

void app_main()
{
    
}

```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs

```
Debug log goes here, should contain the backtrace, as well as the reset source if it is a crash.
Please copy the plain text here for us to search the error log. Or attach the complete logs but leave the main part here if the log is *too* long.
```

## Other items if possible

- [X ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ X] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.) 

![esp-iot-solution
[image]
-monitor2](https://user-images.githubusercontent.com/2299606/62659323-09c9b800-b939-11e9-8880-551a6fb43195.png)
[image]
[image]

[FileforDebug.zip](https://github.com/espressif/esp-iot-solution/files/3478994/FileforDebug.zip)

tawalke, 2019-08-08:
When I use the i2ctools and manually config (again) `i2cconfig --port=1 --sda=25 --scl=26 --freq=100000`
I get this but rebuilding and flashing solution still doesn't show output on OLED: 
[image]

colesnicov, 2019-08-08:
Hello. Without looking into your code, I see in your console listing that you have not initialized the I2C driver pin. Here is my console listing from ESP8266:
[image]
 The red arrow shows that I have configured the I2C pin configuration. More precisely, I use pins 4 and 5. In your console listing, you can see only 1 pins with number 0 initialization. I assume this is 1 wire? you must perform the initialization pin for I2C and install drivers for I2C .. Sorry for the English, I use Google Translate.

colesnicov, 2019-08-08:
May be [this](https://github.com/espressif/esp-iot-solution/issues/48)?

tawalke, 2019-08-08:
Hi @colesnicov My code is just using the esp-iot-solution example for the board with no changes. I did set the pins using the esptool.py and then reran the code and got the same experience. I didn't attached the console after doing this as it was late.  
My I2C config was/is: SDA=25 SCL=26 pin# 1 and 2. I just did an erase flash and going to redo all my steps to ensure I didn't miss anything but I did set the pins but still didn't get a response.

colesnicov, 2019-08-08:
OK, as mentioned [here](https://github.com/espressif/esp-iot-solution/issues/48#issuecomment-519383010), try to use another library or Oscilloscope and check the I2C line?

tawalke, 2019-08-08:
@colesnicov Yes the ssd1306 is switched on in the components menu via make menuconfig. It was one of the first things I checked when the OLED was not displaying output.  

And yes that's what is weird about this, the issue is only that the data will not write to the OLED.  All other sensors are working which is why I filed this ticket.  Before I used the esp-azure-iot-kit solution code (as it was) and just wrote out a simple Hello World program to test the board the OLED was working fine and I could see output.

I will try your code to see if the board reacts as well first before I try another library.

tawalke, 2019-08-09:
This is definitely an issue with the something in the solution. I have the device connected and sending data to Azure from the sensors but still no output to OLED.  Anyone from espressif going to look at this?????

ESP32DE, 2019-08-09:
@tawalke 
can i ask you which board version you have?
V0.8 like these ?

[image]

[image]

[image]

[image]

you use [this demo](https://github.com/espressif/esp-iot-solution/tree/master/examples/esp32_azure_iot_kit) code right?

tawalke, 2019-08-09:
Hi @ESP32DE Yes this is the exact board I have and yes that is the demo was running.  I also tried to run this demo for OLED (https://github.com/espressif/esp-iot-solution/tree/master/examples/oled_screen_module) but this trashes my serial port connection and I have to an erase_flash to get it back. 

Before using this sample, the OLED was working out the box.

ESP32DE, 2019-08-09:
@tawalke 
txs.
i study just in time the shematic.
[image]

I2C0_SDA is going to ESP32 IO25 
I2C0_SCL  is going to ESP32 IO26
but in code it is set 
I2C0_SDA [IO17](https://github.com/espressif/esp-iot-solution/blob/master/components/i2c_devices/others/ssd1306/include/iot_ssd1306.h#L54)
I2C0_SCL [IO4](https://github.com/espressif/esp-iot-solution/blob/master/components/i2c_devices/others/ssd1306/include/iot_ssd1306.h#L50)

also 
OLED SCL [IO4](https://github.com/espressif/esp-iot-solution/blob/master/components/i2c_devices/others/ssd1306/include/iot_ssd1306.h#L81)
OLED SDA [IO17](https://github.com/espressif/esp-iot-solution/blob/master/components/i2c_devices/others/ssd1306/include/iot_ssd1306.h#L82)

i will check the other I2C devices... and come back to the post with append..
( can be there is a code problem but let me check )

tawalke, 2019-08-09:
Hi @ESP32DE: 

First when I look at menuconfig there are no options to set SDA and SCL from code. I used the I2C tools to do so when testing (see screenshots below).  
[image]

[image]

Using their code, with NO changes: 
The OLED doesn't work and this is output from monitor: 
[image]

About to rerun i2ctools and provide output from that for you.

ESP32DE, 2019-08-09:
@tawalke 
txs
ok i suggest now 2 possibles to do:
a) we check each sensor code for not use pins ( CS Pins ) from Oled Display 
b) we change together the OLED header to the right pins ( SDA 25, SCL 26) like it is on ESP32 shematic

.-just in time i run in a) 
..

ESP32DE, 2019-08-09:
@tawalke 
in the meantime you can perhabs [download the PDF](https://www.espressif.com/sites/default/files/documentation/esp32-azure_iot_kit_hardware_design_guide__en.pdf) from espressif
or [here](https://www.espressif.com/en/support/download/documents?keys=Azure)
..

tawalke, 2019-08-09:
So in looking at their example code here, it seems as if the SDA and SCL are being set correctly based upon schematic. Unless it's being overwritten. Thoughts?
[image]

[image]

The OLED/SS1306 is a sensor on the board AND it's enabled via menuconfig.  There is just not an option to set SDA/SCL in menuconfig. Setting SDA/SCL is handled in code as you can see above.

ESP32DE, 2019-08-09:
@tawalke 
ok try this if is possible:
first make a backup of the [header file from OLED SSD1306](https://github.com/espressif/esp-iot-solution/blob/master/components/i2c_devices/others/ssd1306/include/iot_ssd1306.h), that you can use the origin later again
load the [header file](https://github.com/espressif/esp-iot-solution/blob/master/components/i2c_devices/others/ssd1306/include/iot_ssd1306.h) from OLED SSD1306 in a code editor

now change all CLK and SDA pins in OLED Header to the shematic ESP32 PINS

[image]

which are 
for SDA I2C0_SDA = 25
for SCL I2C0_SCL = 26

from OLED Shematic the CS Pins goes to GND
[image]

so i think it is Active Low and pulled to GND no CS Pin need from ESP32..

so change only the lines in header for CLK and SDA ( DIN ) 

we start here to edit the pin numbers :
[SDD1306_CLK_NUM 4 ](https://github.com/espressif/esp-iot-solution/blob/master/components/i2c_devices/others/ssd1306/include/iot_ssd1306.h#L50) must be your I2C_SCL example then 26
[SSD1306_DIN_NUM 17](https://github.com/espressif/esp-iot-solution/blob/master/components/i2c_devices/others/ssd1306/include/iot_ssd1306.h#L54) must be your I2C_SDA example 25

[OLED_IIC_SCL_NUM 4](https://github.com/espressif/esp-iot-solution/blob/master/components/i2c_devices/others/ssd1306/include/iot_ssd1306.h#L81) must be your I2C_SCL exmaple again 26
[OLED_IIC_SDA_NUM 17](https://github.com/espressif/esp-iot-solution/blob/master/components/i2c_devices/others/ssd1306/include/iot_ssd1306.h#L82) must be your I2C_SDA example again 25

vola - hope this helps
don't forgett to clean your Demo Project

btw ( perhabs this must be edit by espressif later too but for testing now not important just in time ) 
[SSD1306_RES_NUM 18](https://github.com/espressif/esp-iot-solution/blob/master/components/i2c_devices/others/ssd1306/include/iot_ssd1306.h#L58) 

is pulled high and not conected to ESP32 pins ( like ESP32 Shematic says ) 

[image]

also
and [SSD1306_DC_NUM 5](https://github.com/espressif/esp-iot-solution/blob/master/components/i2c_devices/others/ssd1306/include/iot_ssd1306.h#L62) are not used like the shematic  from OLED SSD1306 says
[image]

is pulled to GND.

A note:

the demo code runs and bump the sensor data  .,  so the sensor all ok with 25, 26  i think.
the Oled could not run cause there was wrong Pins in the header
not your faul .- good catch - i think this is a bug :+1: 

hope this helps and hope this can be resoltion for you ( and Azure users ) 
Let me know how can go on in this and if there is any further problem

best wishes
rudi ;-)

tawalke, 2019-08-09:
Hi @ESP32DE Let me try this and get back to you on results.  
Thanks so much.

ESP32DE, 2019-08-09:
@tawalke 
the header file must be changed more
the PERIPHS_IO_MUX pins are not right and points not to the right GPIO_MUX too

// CLK old 4
// #define SSD1306_CLK_MUX             PERIPHS_IO_MUX_GPIO4_U
// #define SSD1306_CLK_NUM             4
// #define SSD1306_CLK_FUNC            FUNC_GPIO4_GPIO4

// CLK new 26 
#define SSD1306_CLK_MUX             PERIPHS_IO_MUX_GPIO26_U
#define SSD1306_CLK_NUM             26
#define SSD1306_CLK_FUNC            FUNC_GPIO26_GPIO26

// SDA (DIN) old 17
// #define SSD1306_DIN_MUX             PERIPHS_IO_MUX_GPIO17_U
// #define SSD1306_DIN_NUM             17
// #define SSD1306_DIN_FUNC            FUNC_GPIO17_GPIO17

// SDA (DIN) new 25
#define SSD1306_DIN_MUX             PERIPHS_IO_MUX_GPIO25_U
#define SSD1306_DIN_NUM             25
#define SSD1306_DIN_FUNC            FUNC_GPIO25_GPIO25

//I2C
// SCL, SDA  old 4 and 17
// #define OLED_IIC_SCL_NUM            (gpio_num_t)4       /*!< gpio number for I2C master clock IO4*/
// #define OLED_IIC_SDA_NUM            (gpio_num_t)17      /*!< gpio number for I2C master data IO17*/

// SCL, SDA new 26 and 25
#define OLED_IIC_SCL_NUM            (gpio_num_t)26       /*!< gpio number for I2C master clock IO26*/
#define OLED_IIC_SDA_NUM            (gpio_num_t)25      /*!< gpio number for I2C master data IO25*/

One point i do not 100 procent know, is, 
perhabs we must change the I2C_NUM from I2C_NUM_0 to I2C_NUM_1 
onetime in the Oled SSD1306 header file..
cause **I2C_MASTER_NUM** is defined as **I2C_NUM_1** 
..i am searching just in time in code and will append ( edit ) 
so long test it with this:

// #define OLED_IIC_NUM                I2C_NUM_0           /*!< I2C number >*/
#define OLED_IIC_NUM                I2C_NUM_1           /*!< I2C number >*/

btw check the I2C frequenz init, i think it is ok 100000
#define OLED_IIC_FREQ_HZ            100000              /*!< I2C colock frequency >*/

hope this helps..

tawalke, 2019-08-09:
Wow yes this helps a LOT, I was definitely about to save with only the SDA/SCL changes. You rock!! 

Now that you have shared this, I'm going to really double down on reviewing the schematic now to make sure everything is correct, but I think you really did the work for me.

ESP32DE, 2019-08-09:
@costaud  @projectgus @anujdeshpande  
perhabs there are old [OLED SSD1306 header](https://github.com/espressif/esp-iot-solution/blob/master/components/i2c_devices/others/ssd1306/include/iot_ssd1306.h) in the driver ( 2 Years old ) 
can someone plz check the SCL, SDA pins (4,17) and unused pins (23,18,5) from CS, DC and RST, cause [shematic](https://www.espressif.com/sites/default/files/documentation/esp32-azure_iot_kit_hardware_design_guide__en.pdf) says, 
they are 25,26 and CS RST DC not used GPIO'S but used in the OLED SSD1306 h/c file..
also the I2C_NUM_1 is for MASTER defined and used but header is I2C_NUM_0
txs

@tawalke 
hope you can go on now
happy coding 
best wishes
rudi ;-)

tawalke, 2019-08-10:
Thanks @ESP32DE for everything. I'm going to implement and update ticket appropriately with results. I hope this helps others and that it gets fixed.

fivdi, 2019-08-22:
The [esp32_azure_iot_kit example](https://github.com/espressif/esp-iot-solution/tree/master/examples/esp32_azure_iot_kit) functions correctly for me on an ESP32 Azure IoT Kit. No modifications to the code were needed to get the example to function.

The GPIO numbers for the I2C clock and data lines are set correctly [here](https://github.com/espressif/esp-iot-solution/blob/master/examples/esp32_azure_iot_kit/main/main.c#L27-L28).

The GPIO numbers defined in [iot_ssd1306.h](https://github.com/espressif/esp-iot-solution/blob/master/components/i2c_devices/others/ssd1306/include/iot_ssd1306.h#L81-L82) are not used by the esp32_azure_iot_kit example and therefore shouldn't cause any issues.

tawalke, 2019-08-22:
We've (3 engineers not incl myself) flashed 3 different Esp32 Azure IoT kit boards without code changes using this exact code and gotten this same behavior. If  the internal changes we made to the ssd1306.h noted by Rudi is used it works for all of us. So there is some issue that's happening that needs to be reviewed in using the OLED on the board. If this is not the culprit we would love insight into what it is, so we can consistent results in testing.

Thanks.

fivdi, 2019-08-23:
One interesting point is that there appears to be several different versions of the ESP32-Azure IoT Kit. Or at least different boards have different information printed on them which may indicate that the hardware is also different. Perhaps the hardware is so different that the esp32_azure_iot_kit example functions unmodified on some boards but not on others.

For example, the text printed on [this board](https://user-images.githubusercontent.com/2299606/62659324-09c9b800-b939-11e9-8b58-4af5a11e8907.jpg) from [this post](https://github.com/espressif/esp-iot-solution/issues/49#issue-478150968) is "ESP32-Azure IoT Kit".

The text printed on [this board](https://user-images.githubusercontent.com/16070445/62810759-5101a580-baff-11e9-9fa6-8af75b5dec7b.png) from [this post](https://github.com/espressif/esp-iot-solution/issues/49#issuecomment-520074989) is "ESP32-Azure IoT Kit V0.8 2018.9.28"

The board I'm using has "ESP32 Azure IoT Kit 1845" printed on it.

[image]

The I2C bus for the esp32_azure_iot_kit example is configured by `i2c_bus_init` [here](https://github.com/espressif/esp-iot-solution/blob/master/examples/esp32_azure_iot_kit/main/main.c#L74-L84). This means that the [defines related to initializing the I2C bus and SSD1306 in iot_ssd1306.h](https://github.com/espressif/esp-iot-solution/blob/master/components/i2c_devices/others/ssd1306/include/iot_ssd1306.h#L45-L86) and [the code related to to initializing the I2C bus and SSD1306 in ssd1306.c](https://github.com/espressif/esp-iot-solution/blob/master/components/i2c_devices/others/ssd1306/ssd1306.c#L248-L258) shouldn't be needed for the esp32_azure_iot_kit example. If the defines and code are commented out, the esp32_azure_iot_kit example still functions on the board I'm using.

All that being said, there is an issue. The [code in ssd1306.c that is there to initialize the I2C bus and SSD1306](https://github.com/espressif/esp-iot-solution/blob/master/components/i2c_devices/others/ssd1306/ssd1306.c#L248-L258) isn't needed for the esp32_azure_iot_kit example. The code configures 5 pins, resets two pins (CS, DC) and sets one pin (RES). On an ESP32-Azure IoT Kit there are actually no pins on the ESP32 connected to CS, DC or RES.

tawalke, 2019-09-25:
@fivdi The code I posted [here](https://github.com/espressif/esp-iot-solution/issues/49#issuecomment-520086714) shows the i2c_bus_init code and the associated pins setting as you noted. The OLED still does not show anything on the screen.  

Additionally, I checked my team's other boards. We ordered a bunch of these in groups but it seems they all look like my board (numbers, etc) and not like the boards you showed above. So yes, I concur maybe there is different hardware for this board?? Additionally, we have run the OLED sample updating the SDA, SCL setting on multiple boards and we all are getting the same errors/results. 

I would love to have a definitely answer and a concrete solution to this problem, and other folks may order this board and it should be a consistent experience with the board and all the onboard sensors including the OLED. Please advise.

tawalke, 2019-10-21:
Used feedback from @ESP32DE as solution. Closing since no other response.

closed 2019-10-21 by tawalke