espressif/esp-iot-solution #51: Is it possible to compile the lvgl examples on idf 4.0 / gcc 8.2 tool chain?

daduke, 2019-08-16:
## Environment

- Development Kit:      none
- Kit version
- Module or chip used:  ESP32-WROVER-B
- IDF version (run ``git describe --tags`` to find it): v4.0-dev-1443-g39f090a4f
   
- Build System:        both
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): 8.2.0

- Operating System:    Fedora 30
- Power Supply:         external 3.3V

## Problem Description

trying to compile the lvgl examples on  idf 4.0 / gcc 8.2 tool chain results in dozens of errors. Is this even supported?

### Expected Behavior
successful make

### Actual Behavior
hundreds of errors like
/opt/esp32-idf/projects/esp-iot-solution/submodule/esp-idf/components/newlib/include/sys/reent.h:166:8: error: '_VOID' does not name a type
 extern _VOID   _EXFUN(__sinit,(struct _reent *));

/opt/esp32-idf/esp-idf/tools/xtensa-esp32-elf/esp32-2019r1-8.2.0/xtensa-esp32-elf/xtensa-esp32-elf/sys-include/stdlib.h:155:44: error: expected initializer before '__result_use_check'
 void *reallocarray(void *, size_t, size_t) __result_use_check __alloc_size(2)

etc
other projects build fine.

### Steps to repropduce

1. install current idf + current toolchain according to docs
2. make menuconfig
3. make

### Code to reproduce this issue

## Debug Logs

## Other items if possible

C47D, 2019-11-22:
Hi @daduke,

There's a example project on the LVGL repo which is running using both IDF3.3 and 4.0, maybe it can help you to get up and running, its available here: [lv_port_esp32_ili9341](https://github.com/littlevgl/lv_port_esp32_ili9341).

We're also trying to find a easy way to integrate lvgl into esp32 projects in [here](https://forum.littlevgl.com/t/lvgl-on-the-esp-idf-ecosystem/1232), maybe you can help us with some suggestions.

Regards

InfiniteYuan1, 2020-01-22:
IDF submodule will be updated to 3.3 or 4.0 soon.

leeebo, 2020-08-13:
Hi @daduke , you can use [esp-iot-solution/tree/release/v1.1](https://github.com/espressif/esp-iot-solution/tree/release/v1.1) branch in your idf 4.0 project. You can also use standalone lvgl repo here: https://github.com/lvgl/lv_port_esp32 .

closed 2023-07-10 by leeebo