espressif/esp-iot-solution #52: Is it possible to compile the empty_project on idf 4.0 / gcc 8.2 tool chain?

linyaqiang, 2019-08-17:
## Environment

- Development Kit: ESP32-DevKitC
- Kit version
- Module or chip used:  [ESP32-WROOM-32
- IDF version (run ``git describe --tags`` to find it): v4.0-dev-1287-gd7e659df2
- Build System: Make
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):xtensa-esp32-elf-gcc.exe (crosstool-NG esp32-2019r1) 8.2.0
- Operating System: Windows
- Power Supply: USB

## Problem Description
      building the empty project, then report some errors!
## Debug Logs
Administrator@DE-0011 MSYS /d/lin_work/esp32_work/esp32_prj/empty_project
# make all
/d/lin_work/esp32_work/esp32_prj/esp/esp-iot-solution/submodule/esp-idf//make/project.mk:60: esp-idf build system only supports MSYS2 in "MINGW32" mode. Consult the ESP-IDF documentation for details.
WARNING: Failed to find Xtensa toolchain, may need to alter PATH or set one in the configuration menu
/d/lin_work/esp32_work/esp32_prj/esp/esp-iot-solution/submodule/esp-idf/make/project.mk:60: esp-idf build system only supports MSYS2 in "MINGW32" mode. Consult the ESP-IDF documentation for details.
WARNING: Failed to find Xtensa toolchain, may need to alter PATH or set one in the configuration menu
WARNING: Missing submodule components/bootloader/subproject/components/micro-ecc/micro-ecc...
Attempting 'git submodule update --init components/bootloader/subproject/components/micro-ecc/micro-ecc' in esp-idf root directory...
error: 路径规格 'components/bootloader/subproject/components/micro-ecc/micro-ecc' 未匹配任何 git 已知文件
make[1]: *** [/d/lin_work/esp32_work/esp32_prj/esp/esp-iot-solution/submodule/esp-idf/make/project.mk:531：/d/lin_work/esp32_work/esp32_prj/esp/esp-iot-solution/submodule/esp-idf/components/bootloader/subproject/components/micro-ecc/micro-ecc/.git] 错误 1
make: *** [/d/lin_work/esp32_work/esp32_prj/esp/esp-iot-solution/submodule/esp-idf/components/bootloader/Makefile.projbuild:41：/d/lin_work/esp32_work/esp32_prj/empty_project/build/bootloader/bootloader.bin] 错误 2

InfiniteYuan1, 2019-08-19:
> Failed to find Xtensa toolchain, may need to alter PATH or set one in the configuration menu

Please correct set path refer to [setup-toolchain](https://docs.espressif.com/projects/esp-idf/en/v3.2.2/get-started/index.html#setup-toolchain)

linyaqiang, 2019-08-19:
I find the example of esp-iot-solution cannot to build by the compiler version of 8.2.0, then can be build by the compiler version of 5.2.0.

InfiniteYuan1, 2019-08-20:
IDF 版本和 工具链有一定关联，所以会出现问题，你需要选择你使用 IDF 对应的工具链，可以在在线文档中找到。

leeebo, 2020-08-13:
Hi @liyaqiang , you can use [esp-iot-solution/tree/release/v1.1](https://github.com/espressif/esp-iot-solution/tree/release/v1.1) branch for your idf 4.0 projects.

closed 2023-07-10 by leeebo