espressif/esp-iot-solution #53: When using an 128 x 64 OLED display the SSD1306 OLED driver displays the 16th character on a row incorrectly

fivdi, 2019-08-25:
## Environment

- Development Kit:      ESP32-Azure IoT Kit
- Kit version (for WroverKit/PicoKit/DevKitC): Unknown. Unfortunately no version information is printed on board. The number 1845 is printed on the board. See below pictures.
- Module or chip used:  ESP32-WROVER-B
- IDF version (run ``git describe --tags`` to find it): v3.1.3
- Build System:         Make
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): 1.22.0-80-g6c4433a
- Operating System:     Linux
- Power Supply:         USB

## Problem Description

The ESP32-Azure IoT Kit has a 128 x 64 OLED display. If text is displayed on the OLED display using the 16 x 8 font, it should be possible to display 16 characters per row on the OLED display.

The X pixel position of the 1st character is 0
The X pixel position of the 2nd character is 8
The X pixel position of the 3rd character is 16
...
The X pixel position of the 16th character is 120

If a character is displayed at X pixel position 120 using the 16 x 8 font it is not displayed correctly. The character is 8 pixels wide. As can be seen in the below picture, the first 6 columns of pixels for the character are displayed on the right hand side of the display and the last 2 columns of pixels are displayed on the left hand side of the display.

[image]

### Expected Behavior

A character displayed at X pixel position 120 using the 16 x 8 font should appear on the right hand side of the display.

### Actual Behavior

A character displayed at X pixel position 120 using the 16 x 8 font only partially appears on the right hand side of the display. The first 6 columns of pixels for the character are displayed on the right hand side of the display and the last 2 columns of pixels are displayed on the left hand side of the display.

### Steps to repropduce

The issue can be reproduced with the code provided below.

### Code to reproduce this issue

```cpp
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "iot_i2c_bus.h"
#include "iot_ssd1306.h"

#define I2C_MASTER_SCL_IO           26
#define I2C_MASTER_SDA_IO           25
#define I2C_MASTER_NUM              I2C_NUM_1
#define I2C_MASTER_FREQ_HZ          100000

static i2c_bus_handle_t i2c_bus = NULL;
static ssd1306_handle_t ssd1306_dev = NULL;

static void i2c_bus_init() {
    i2c_config_t conf;
    conf.mode = I2C_MODE_MASTER;
    conf.sda_io_num = (gpio_num_t)I2C_MASTER_SDA_IO;
    conf.sda_pullup_en = GPIO_PULLUP_ENABLE;
    conf.scl_io_num = (gpio_num_t)I2C_MASTER_SCL_IO;
    conf.scl_pullup_en = GPIO_PULLUP_ENABLE;
    conf.master.clk_speed = I2C_MASTER_FREQ_HZ;
    i2c_bus = iot_i2c_bus_create(I2C_MASTER_NUM, &conf);
}

static void ssd1306_init() {
    ssd1306_dev = iot_ssd1306_create(i2c_bus, SSD1306_I2C_ADDRESS);
    iot_ssd1306_refresh_gram(ssd1306_dev);
    iot_ssd1306_clear_screen(ssd1306_dev, 0x00);
}

extern "C" void app_main() {
    i2c_bus_init();
    ssd1306_init();

    iot_ssd1306_draw_string(ssd1306_dev, 120, 0, (const uint8_t *) "6", 16, 0);
    iot_ssd1306_refresh_gram(ssd1306_dev);
}
```

### Steps to fix

This can be fixed by setting the lower column start address for page addressing mode to 0 rather than 2 as is currently the case.

In other words, changing [this line of code](https://github.com/espressif/esp-iot-solution/blob/master/components/i2c_devices/others/ssd1306/include/iot_ssd1306.h#L34) from:
```c
#define SSD1306_SET_LOWER_ADDRESS   0x02
```
to:
```c
#define SSD1306_SET_LOWER_ADDRESS   0x00
```
resolves the issue.

Here's a picture of what it looks like after this modification is made.

[image]

I can create a pull request with this change but I don't know why `SSD1306_SET_LOWER_ADDRESS` is currently set to 2. I also don't know if setting `SSD1306_SET_LOWER_ADDRESS` to 0 will have a negative impact on something else.

If I had to guess why `SSD1306_SET_LOWER_ADDRESS` is currently set to 2, I'd say it has something to do with supporting 132 x 64 OLED displays that have a SH1106 chip. If is this is the case, it has a negative impact on 128 x 64 displays as the 16th character on a row is no longer displayed correctly. An additional negative impact of this is that the text wrap functionality that provided by the SSD1306 driver can be relied on as it will also display the 16th character on a row incorrectly.

cima, 2019-12-22:
It is not just about the letter is wrapping around. If I runaway from display it totally falls appart.
 I suspect that author of this library bought display advertised as SSD1306 but obtained SH1106 that does have the two column shift. There's a couple of threads in arduino forum:
- [problem with a 128X64 oled i2c display](https://forum.arduino.cc/index.php?topic=492076.0)
- [\[SOLVED\] SSD1306 can only write to part of screen (U8Glib works, AdaFruit not)](https://forum.arduino.cc/index.php?topic=265557.0)
- [Problem with SSD1306 LCD and U8glib](https://forum.arduino.cc/index.php?topic=219419.90)

I have my firts two columns fill with "random mess" (new within each boot).

[image]

Totaly cures my problem.

[image]

*Offtopic:* Some polishing would be needid for this library, because On my Lilygo Oled display is at different pins and without modifying the code of this library I couldn't use it. Wires are hardcoded.

fivdi, 2019-12-22:
> Thank you so much @fivdi

@cima you're welcome :smile:

simonliu009, 2020-06-17:
@fivdi Thank you, you save my day.

LucasCanete, 2021-05-24:
@fivdi why is iot_sdd1306.h no longer ib this project? I can only find it from others people bifurcations but it seems it is no longer supported

fivdi, 2021-05-24:
It looks like it got moved to a different location. See: https://github.com/espressif/esp-iot-solution/tree/master/components/display/screen/controller_driver/ssd1306

LucasCanete, 2021-05-25:
@fivdi Yes thanks for your answer. Kind of weird that now that it has been moved it lacks some useful functions like iot_ssd1306_draw_string() and iot_ssd1306_draw_char() right?. Probably they will update it later

fivdi, 2021-05-25:
> Kind of weird that now that it has been moved it lacks some useful functions like iot_ssd1306_draw_string() and iot_ssd1306_draw_char() right?.

Yes, that is strange. The implementation doesn't look finished. There doesn't appear to be any examples using the SSD1306 anymore. I can't say if it works, I've never tried it.

> Probably they will update it later

Perhaps.

commit 2477d1da (2024-06-03): Merge branch 'fix/ci_push_to_github' into 'master'

fix(ci): specify runner tags for push_to_github job

Closes #53

See merge request ae_group/esp-iot-solution!1014
closed 2023-07-10 by leeebo