espressif/esp-iot-solution #54: ESP-MDF

GitHubForPP, 2019-08-28:
I am using esp-mdf now, but how should I use ssd1306 driver to light up my OLED?

haasc, 2019-09-13:
According to the home page, using this project as a set of components requires adding

`include($ENV{IOT_SOLUTION_PATH}/CMakeLists.txt)
include($ENV{IDF_PATH}/tools/cmake/project.cmake)`

to your project's CMakeLists.txt.  However, the first line rewrites the value of IDF_PATH to force the use of the built-in IDF version, which seems to defeat the purpose.

An attempt to switch like

`include($ENV{IOT_SOLUTION_PATH}/components/CMakeLists.txt)
include($ENV{IDF_PATH}/tools/cmake/project.cmake)`

sort of work - files are available for compilation purposes, but 'idf.py menuconfig' has none of the IOT components available for configuration, which seems like a bug.

InfiniteYuan1, 2020-01-22:
Please refer to [buddy](https://github.com/espressif/esp-mdf/tree/master/examples/development_kit/buddy) example.

Xiehanxin, 2020-07-27:
@GitHubForPP  hello, If your issue is now resolved, could you please close the issue. if you still have problem, please tell us . Thank you.

closed 2020-07-27 by GitHubForPP