espressif/esp-iot-solution #57: make: *** No rule to make target '//Makefile'.  Stop.

Abishek05, 2019-09-15:
## Environment

- Development Kit:    ESP32-DevKitC
- Kit version (for WroverKit/PicoKit/DevKitC): v1
- Module or chip used:  ESP32-WROOM-32
- IDF version : v3.3
- Build System:    Make
- Compiler version :  (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a5) 5.2.0
- Operating System:   Windows
- Power Supply:    USB

## Problem Description

When an example ulp project(esp-idf-solutions repository) is built it fails by saying "make: *** No rule to make target '//Makefile'.  Stop." However, I have installed and added the ulp toolchain in path. The ulp examples in the main esp-idf repository builds and I can flash successfully. The ulp examples of esp-idf-solutions do not work.

### Expected Behavior
Expected Behavior is that Build should be successful.
### Actual Behavior
Gives build error: make: *** No rule to make target '//Makefile'.  Stop.
### Steps to repropduce

1. navigate to example project
2. run make flash

## Debug Logs
abish@DESKTOP-S465RFF MINGW32 /d/esp/ulp_tsens

```
$ make flash
Toolchain path: /opt/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: crosstool-ng-1.22.0-80-g6c4433a5
Compiler version: 5.2.0
make: *** No rule to make target '//Makefile'.  Stop.

```

## Other items if possible

[sdkconfig.txt](https://github.com/espressif/esp-iot-solution/files/3613495/sdkconfig.txt)

zztiger123, 2019-12-24:
Hi, Abishek05

If you change the location of the folder `ulp_tsens`, you need to modify the `Makefile` (modify the path like esp-iot-solution)in the folder `ulp_tsens` when you use make. Like the following:
```
#
# This is a project Makefile. It is assumed the directory this Makefile resides in is a
# project subdirectory.
#

PROJECT_NAME := ulp-tsens

#If IOT_SOLUTION_PATH is not defined, use relative path as default value
# IOT_SOLUTION_PATH ?= $(abspath $(shell pwd)/../../../)

IOT_SOLUTION_PATH=C:/esp-iot-solution
IDF_PATH=C:/esp-iot-solution/submodule/esp-idf

include $(IOT_SOLUTION_PATH)/Makefile
include $(IDF_PATH)/make/project.mk
```
Be careful: You need to use your own environment path. The above snippet uses my environment path as an example.

Xiehanxin, 2020-07-27:
@Abishek05   hello, If your issue is now resolved, could you please close the issue. if you still have problem, please tell us . Thank you.

closed 2023-07-10 by leeebo