espressif/esp-iot-solution #58: lvgl_example and IDF3.3 not working

brownx, 2019-09-23:
IDF v3.3 (basic kit was installed with https://dl.espressif.com/dl/esp-idf-tools-setup-2.0.exe)
ESP32_Core_board_v2 (with ESP32 WROOM 32) 
littlevgl installed: https://blog.littlevgl.com/2019-01-31/esp32

Since I use latest idf instead of "make menuconfig" and "make" I use:
idf.py menuconfig
idf.py build
from esp-iot-solution\examples\hmi\lvgl_example directory.

The result of build is:
_CMake Error at D:/Work/ESP32/esp-iot-solution/submodule/esp-idf/tools/cmake/project.cmake:57 (idf_set_global_variables):
  Unknown CMake command "idf_set_global_variables".
Call Stack (most recent call first):
  CMakeLists.txt:20 (project)_

The hello_world example from IDF is working fine, I can compile and run it in the same way, I have issues only with the IOT package.

C47D, 2019-11-23:
I have the same issue when trying to build the alink_smart_device_demo, but this repo seems to be in dead silence. Have you come out with a solution for this?

InfiniteYuan1, 2020-01-22:
@brownx Please refer to [perpare](https://github.com/espressif/esp-iot-solution#preparationmake)

InfiniteYuan1, 2020-01-22:
@C47D This may be encounter problem that other example can't complie in esp-iot-solution.

wolfeidau, 2020-01-22:
I was looking to try even bumping esp-idf up to 3.2.3 but alas I get the same `idf_set_global_variables` issue when building :cry:

InfiniteYuan1, 2020-01-22:
idf_set_global_variables this is located in idf 3.2.2

wolfeidau, 2020-01-22:
@InfiniteYuan1 thanks, I figured out how to get idf 3.2.3 working.

after the initial clone of esp-iot-solution.
```
git submodule update --init --recursive
cd submodule/esp-idf
git checkout v3.2.3
git submodule update --init --recursive
```

And then I navigate to `examples/esp32_azure_iot_kit` and the project builds with make :+1: 

Cheers

Alexlizx, 2020-05-08:
> Then i went to the esp-idf directory and pulled master
> 
> ```
> esp-iot-solutions $ cd submodule/esp-idf
> esp-idf $ git pull origin master
> ...
> ```
@C47D  I tried your solution to update the esp-idf submodule, but encountered the following errors:
`Unlink of file 'components/micro-ecc/micro-ecc' failed. Should I try again? (y/n) y
Unlink of file 'components/micro-ecc/micro-ecc' failed. Should I try again? (y/n) n
Unlink of file 'components/aws_iot/aws-iot-device-sdk-embedded-C' failed. Should I try again? (y/n) y
Unlink of file 'components/aws_iot/aws-iot-device-sdk-embedded-C' failed. Should I try again? (y/n) n`
There are hundreds of CONFLICT messages after this, so I'm not going to post them all here.  An example is `CONFLICT (content): Merge conflict in tools/unit-test-app/unit_test.py`.

@InfiniteYuan1 The esp-idf submodule branch in esp-iot-solution is v3.2.2, which does not recognize "idf_set_global_variables".  `git submodule update --init --recursive` does not help me at all because it just reverts the idf to 3.2.2. Is there another way to update it to 4.0? I tried to do a straight forward git clone of the esp-idf repository but then cmake failed.

C47D, 2020-05-08:
@Alexlizx Hi, I haven't been able to work on this repo since I posted my proposal, but if you are trying to get a quick demo of LVGL running on the ESP32 we are working on [this](https://github.com/littlevgl/lv_port_esp32) repo. Hope it helps!.

Alexlizx, 2020-05-08:
@C47D Thank you for the quick reply. I am working to get the aws_iot_demo example running on my board but I keep running into `Unknown CMake command "idf_set_global_variables"` like the OP. Just curious about how you were able to update the idf submodule.

leeebo, 2020-08-13:
New [esp-iot-solution/tree/release/v1.1](https://github.com/espressif/esp-iot-solution/tree/release/v1.1) branch based on esp-idf v4.0.1 becomes available now.

leeebo, 2023-07-10:
The ESP-IDF has integrated LVGL from v5.0, please use the component https://components.espressif.com/components/espressif/esp_lvgl_port instead, and you can find more examples in https://github.com/espressif/esp-idf/tree/master/examples/peripherals/lcd and https://github.com/espressif/esp-bsp/tree/master/esp32_s3_lcd_ev_board

closed 2023-07-10 by leeebo