espressif/esp-iot-solution #59: Industry Protocols on Azure Kit ESP32

Architk08, 2019-10-01:
On this link 
https://catalog.azureiotsolutions.com/details?title=AzureKit-ESP32&source=all-devices-page

It is written that Azure ESP 32 kit supports the following industrial protocols

Industry Protocols
CAN bus ,   OPC UA ,   PPMP ,   Modbus ,   ZigBee ,   PROFINET ,   OPC Classic ,   EtherCAT

I am unable to find any relevant software stack regarding their implementation. Are they under development?. Any help would be appreciated.

Thanks.

zztiger123, 2019-12-24:
[esp-idf](https://github.com/espressif/esp-idf)

zztiger123, 2019-12-24:
Hello, Architk08

Maybe you can find the information you want in [esp-idf](https://github.com/espressif/esp-idf). 

For example : [CAN BUS](https://github.com/espressif/esp-idf/blob/master/components/driver/can.c). 

If you don't find some Protocols, then it may be under development for the time being.

BTW, Here is the [official esp-azure github documentation](https://github.com/espressif/esp-azure), which may also be helpful for your development.

Best regards,

Xiehanxin, 2020-07-27:
hi, If your issue is now resolved, could you please close the issue. if you still have problem, please tell us . Thank you.

closed 2020-07-27 by Architk08