espressif/esp-iot-solution #60: Unable to compile ESP32_Azure_IoT_Kit Example

serenagrl, 2019-10-08:
## Environment

- Development Kit:      [ESP32-Azure IoT Kit]
- Kit version 
- Module or chip used:  [ESP32-WROVER-B]
- IDF version: v4.1-dev-369-g4dac7c7df
- Build System: [CMake]
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp32-2019r1) 8.2.0
- Operating System:     [Windows]
- Power Supply:         [USB]

## Problem Description

idf.py build was not able to compile the example in the "\esp-iot-solution\examples\esp32_azure_iot_kit\" folder.

### Expected Behavior

idf.py build should be able to compile the example as mentioned in documentation

### Actual Behavior

idf.py build will start throwing errors after [178/848]

### Steps to repropduce

1. Install ToolChain
2. Get the latest esp-iot-solution repository from github
3. Run ESP-IDF Command Prompt and navigate to the esp32_azure_iot_kit folder
4. Run idf.py build

## Snapshot of errors. 

The error is too long. Below is just a snapshot of the start of the errors:

Running ninja in directory d:\esp\githubs\esp-iot-solution\examples\esp32_azure_iot_kit\build
Executing "ninja all"...
[172/848] Building CXX object cxx/CMakeFiles/cxx.dir/cxx_exception_stubs.cpp.obj
FAILED: cxx/CMakeFiles/cxx.dir/cxx_exception_stubs.cpp.obj
ccache D:\ESP\.espressif\tools\xtensa-esp32-elf\esp32-2019r1-8.2.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-g++.exe  -DESP_PLATFORM -DGCC_NOT_5_2_0=1 -DHAVE_CONFIG_H -DIDF_VER=\"v3.2.2\" -Iconfig -ID:/ESP/githubs/esp-iot-solution/submodule/esp-idf/components/esp32/include -ID:/ESP/githubs/esp-iot-solution/submodule/esp-idf/components/driver/include -ID:/ESP/githubs/esp-iot-solution/submodule/esp-idf/components/esp_ringbuf/include -ID:/ESP/githubs/esp-iot-solution/submodule/esp-idf/components/tcpip_adapter/include -ID:/ESP/githubs/esp-iot-solution/submodule/esp-idf/components/lwip/include/apps -ID:/ESP/githubs/esp-iot-solution/submodule/esp-idf/components/lwip/lwip/src/include -ID:/ESP/githubs/esp-iot-solution/submodule/esp-idf/components/lwip/port/esp32/include -ID:/ESP/githubs/esp-iot-solution/submodule/esp-idf/components/lwip/port/esp32/include/arch -ID:/ESP/githubs/esp-iot-solution/submodule/esp-idf/components/lwip/include_compat -ID:/ESP/githubs/esp-iot-solution/submodule/esp-idf/components/vfs/include -ID:/ESP/githubs/esp-iot-solution/submodule/esp-idf/components/esp_event/include -ID:/ESP/githubs/esp-iot-solution/submodule/esp-idf/components/log/include -ID:/ESP/githubs/esp-iot-solution/submodule/esp-idf/components/newlib/platform_include -ID:/ESP/githubs/esp-iot-solution/submodule/esp-idf/components/newlib/include -ID:/ESP/githubs/esp-iot-solution/submodule/esp-idf/components/freertos/include -ID:/ESP/githubs/esp-iot-solution/submodule/esp-idf/components/app_trace/include -ID:/ESP/githubs/esp-iot-solution/submodule/esp-idf/components/heap/include -ID:/ESP/githubs/esp-iot-solution/submodule/esp-idf/components/soc/esp32/include -ID:/ESP/githubs/esp-iot-solution/submodule/esp-idf/components/soc/include -Og -std=gnu++11 -fno-rtti -fno-exceptions -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -mlongcalls -nostdlib -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -ggdb -MD -MT cxx/CMakeFiles/cxx.dir/cxx_exception_stubs.cpp.obj -MF cxx\CMakeFiles\cxx.dir\cxx_exception_stubs.cpp.obj.d -o cxx/CMakeFiles/cxx.dir/cxx_exception_stubs.cpp.obj -c D:/ESP/githubs/esp-iot-solution/submodule/esp-idf/components/cxx/cxx_exception_stubs.cpp
In file included from d:\esp\.espressif\tools\xtensa-esp32-elf\esp32-2019r1-8.2.0\xtensa-esp32-elf\xtensa-esp32-elf\sys-include\stdlib.h:18,
                 from d:\esp\.espressif\tools\xtensa-esp32-elf\esp32-2019r1-8.2.0\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\8.2.0\cstdlib:75,
                 from D:/ESP/githubs/esp-iot-solution/submodule/esp-idf/components/cxx/cxx_exception_stubs.cpp:1:
D:/ESP/githubs/esp-iot-solution/submodule/esp-idf/components/newlib/include/sys/reent.h:166:8: error: '_VOID' does not name a type
 extern _VOID   _EXFUN(__sinit,(struct _reent *));
        ^~~~~
D:/ESP/githubs/esp-iot-solution/submodule/esp-idf/components/newlib/include/sys/reent.h:193:3: error: '_PTR' does not name a type
   _PTR _cookie; /* cookie passed to io functions */
   ^~~~
D:/ESP/githubs/esp-iot-solution/submodule/esp-idf/components/newlib/include/sys/reent.h:195:36: error: '_read' has not been declared
   _READ_WRITE_RETURN_TYPE _EXFNPTR(_read, (struct _reent *, _PTR,
                                    ^~~~~
D:/ESP/githubs/esp-iot-solution/submodule/esp-idf/components/newlib/include/sys/reent.h:195:43: error: expected identifier before '(' token
   _READ_WRITE_RETURN_TYPE _EXFNPTR(_read, (struct _reent *, _PTR,
                                           ^
D:/ESP/githubs/esp-iot-solution/submodule/esp-idf/components/newlib/include/sys/reent.h:195:61: error: '_PTR' has not been declared
   _READ_WRITE_RETURN_TYPE _EXFNPTR(_read, (struct _reent *, _PTR,
                                                             ^~~~
D:/ESP/githubs/esp-iot-solution/submodule/esp-idf/components/newlib/include/sys/reent.h:197:36: error: '_write' has not been declared
   _READ_WRITE_RETURN_TYPE _EXFNPTR(_write, (struct _reent *, _PTR,
                                    ^~~~~~
D:/ESP/githubs/esp-iot-solution/submodule/esp-idf/components/newlib/include/sys/reent.h:197:44: error: expected identifier before '(' token
   _READ_WRITE_RETURN_TYPE _EXFNPTR(_write, (struct _reent *, _PTR,
                                            ^
D:/ESP/githubs/esp-iot-solution/submodule/esp-idf/components/newlib/include/sys/reent.h:197:62: error: '_PTR' has not been declared
   _READ_WRITE_RETURN_TYPE _EXFNPTR(_write, (struct _reent *, _PTR,
                                                              ^~~~
D:/ESP/githubs/esp-iot-solution/submodule/esp-idf/components/newlib/include/sys/reent.h:200:20: error: '_seek' has not been declared
   _fpos_t _EXFNPTR(_seek, (struct _reent *, _PTR, _fpos_t, int));

The error will continue until it hit this:

[185/848] Building C object touchpad/CMakeFiles/touchpad.dir/touchpad.c.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

Mair, 2019-12-22:
I'm having the same issue when compiling the hmi/lvgl examples

InfiniteYuan1, 2020-01-22:
Plese refer to https://github.com/espressif/esp-iot-solution/issues/58#issuecomment-576979265

eddco, 2020-05-18:
having the same issue. Tried with many idf versions but didnt work

JackieLi1011, 2020-09-01:
Did you find the solution with your issue?I am facing the same situation.I have tried to git release V1.1 and overwrite master,it is still not working.

espenairmine, 2021-01-31:
I'm having the same issue, tried with multiple idf versions as well as v1.1 (where the example is missing), any plans to update the code for Azure IoT to make it usable as a dev kit (currently not convincing for demoing azure + esp32 = True)?

leeebo, 2023-07-10:
The Azure support has been migrated to [esp-azure](https://github.com/espressif/esp-azure). We just close this issue, if you have further questions please add new issues there.

closed 2023-07-10 by leeebo