espressif/esp-iot-solution #61: new operator kills logging (stack issue?)

brownx, 2019-10-11:
Noticed while compiling lvgl_example.

Log works fine until this call:
_lcd_obj = new CEspLcdAdapter(&lcd_pins);_ from esp-iot-solution\components\hmi\gdrivers\gdisp\ST7789\ST7789_adapter.cpp

Added ESP_LOGI("Something", "1"); lines and the are popping up in the monitor until that line. The last line of _esp-iot-solution\components\spi_devices\lcd\iot_lcd.cpp\CEspLcd::CEspLcd_ added a log, that is visible.

The first log what is not visible is this one:
_CEspLcdAdapter(lcd_conf_t *lcd_conf, int height = LCD_TFTHEIGHT, int width = LCD_TFTWIDTH, bool dma_en = true, int dma_word_size = 1024, int dma_chan = 1) : CEspLcd(lcd_conf, height, width, dma_en, dma_word_size, dma_chan)
    {
        /* Code here*/
		**ESP_LOGI("ST7789_ADA", "1");**
    }_
and the logging stops working completely after from anywhere in the code (between the last CEspLCD log and this one is just the background stuff of "new" operator calling a parent class and than the child class constructor - parent constructor is ok, child get's corrupted somehow and corrupts logging as well).

**Not the logging concerns me but rather the new operator - if does corrupts the stack or anything else which kicks out logging could result in unstable firmware ...**

IDF version: 3.2.2 (iot-solution cannot be compiled with 3.3)
ESP-WROOM-32/ESP32_Core_Board using ST7789 LCD. 
Tools installed with esp-idf-tools-setup-2.0.exe.

Otherwise the example sort of works, I can see content on the LCD ...

InfiniteYuan1, 2020-01-22:
Through these descriptions, it is really unclear what your problem is?

Through these descriptions, it is really unclear what your problem is? 

Did you find out that this operation will cause firmware instability during the analysis of why this project cannot run?

brownx, 2020-01-22:
Nope, cannot find the reason for it - the logging stops working after that line, since it has nothing special (new operator creates a class calling it's constructors and nothing special in those constructors either) most probably the issue is elsewhere ...