espressif/esp-iot-solution #63: Can't build eth2wifi

joel2106, 2019-11-22:
## Environment

- Development Kit: Olimex ESP32-POE
- Kit version (for WroverKit/PicoKit/DevKitC): -
- Module or chip used: ESP32-WROOM-32D
- IDF version (run ``git describe --tags`` to find it): v3.3
- Build System: CMake
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a5) 5.2.0
- Operating System: Windows
- Power Supply: USB

## Problem Description

I'm new to ESP and tried to follow the steps of the Getting Started Guide closely. I tried the ESP-IDF Tools Installer but already followed the Setup Windows Toolchain from Scratch guide as well. I'm trying to get the eth2wifi example to work but have problems building it. The hello_world example built and flashed correctly.

- hello_world example was working
- "git clone --recursive https://github.com/espressif/esp-iot-solution" to the same folder as esp-idf
- "git submodule update --init --recursive" with no error message but no visible installation process as well
- set IOT_SOLUTION_PATH=C:/Users/Public/esp-iot-solution
- copying the example eth2wifi folder to a separate location
- idf.py menuconfig is working as expected. I changed the PHY settings and Pins.
- idf.py build:

C:\Users\Public\esp_ex\eth2wifi>idf.py build
Setting IDF_PATH environment variable: C:\Users\Public\esp
Checking Python dependencies...
Python requirements from C:\Users\Public\esp\requirements.txt are satisfied.
Running ninja in directory C:\Users\Public\esp_ex\eth2wifi\build
Executing "ninja all"...
[0/1] Re-running CMake...
-- Building empty lvgl component due to configuration
-- Building empty ugfx component due to configuration
-- Building empty aws_iot component due to configuration
-- Component names: soc log heap xtensa-debug-module app_trace freertos vfs newlib esp_ringbuf driver esp_event ethernet lwip tcpip_adapter partition_table app_update spi_flash mbedtls micro-ecc bootloader_support nvs_flash pthread smartconfig_ack wpa_supplicant esp32 cxx components __config dac_audio infrared touchpad ulp ulp_monitor esp_adc_cal adc base_class button debugs led light ota param power_meter relay weekly_timer iot_blufi iot_espnow smart_config wifi_conn epaper lcd xpt2046 i2c_bus apds9960 bh1750 bme280 ft5x06 hdc2010 hts221 lis2dh12 mpu6050 mvh3004d veml6040 at24c02 ch450 ht16c21 is31fl3xxx mcp23017 ssd1306 nghttp mqtt tcp udp json openssl alink a4988 servo_motor lowpower_framework lcd_common ili9806 nt35510 lvgl_gui ugfx_gui gdrivers asio jsmn aws_iot bootloader bt coap console esp-tls tcp_transport esp_http_client esp_http_server esp_https_ota esptool_py expat wear_levelling sdmmc fatfs freemodbus idf_test libsodium mdns protobuf-c protocomm spiffs wifi_provisioning main
-- Component paths: C:/Users/Public/esp-iot-solution/submodule/esp-idf/components/soc;C:/Users/Public/esp-iot-solution/submodule/esp-idf/components/log;C:/Users/Public/esp-iot-solution/submodule/esp-idf/components/heap;C:/Users/Public/esp-iot-solution/submodule/esp-idf/components/xtensa-debug-module;C:/Users/Public/esp-iot-solution/submodule/esp-idf/components/app_trace;C:/Users/Public/esp-iot-solution/submodule/esp-idf/components/freertos;C:/Users/Public/esp-iot-solution/submodule/esp-idf/components/vfs;C:/Users/Public/esp-iot-solution/submodule/esp-idf/components/newlib;C:/Users/Public/esp-iot-solution/submodule/esp-idf/components/esp_ringbuf;C:/Users/Public/esp-iot-solution/submodule/esp-idf/components/driver;C:/Users/Public/esp-iot-solution/submodule/esp-idf/components/esp_event;C:/Users/Public/esp-iot-solution/submodule/esp-idf/components/ethernet;C:/Users/Public/esp-iot-solution/submodule/esp-idf/components/lwip;C:/Users/Public/esp-iot-solution/submodule/esp-idf/components/tcpip_adapter;C:/Users/Public/esp-iot-solution/submodule/esp-idf/components/partition_table;C:/Users/Public/esp-iot-solution/submodule/esp-idf/components/app_update;C:/Users/Public/esp-iot-solution/submodule/esp-idf/components/spi_flash;C:/Users/Public/esp-iot-solution/submodule/esp-idf/components/mbedtls;C:/Users/Public/esp-iot-solution/submodule/esp-idf/components/micro-ecc;C:/Users/Public/esp-iot-solution/submodule/esp-idf/components/bootloader_support;C:/Users/Public/esp-iot-solution/submodule/esp-idf/components/nvs_flash;C:/Users/Public/esp-iot-solution/submodule/esp-idf/components/pthread;C:/Users/Public/esp-iot-solution/submodule/esp-idf/components/smartconfig_ack;C:/Users/Public/esp-iot-solution/submodule/esp-idf/components/wpa_supplicant;C:/Users/Public/esp-iot-solution/submodule/esp-idf/components/esp32;C:/Users/Public/esp-iot-solution/submodule/esp-idf/components/cxx;C:/Users/Public/esp-iot-solution/components;C:/Users/Public/esp-iot-solution/components/__config;C:/Users/Public/esp-iot-solution/components/features/dac_audio;C:/Users/Public/esp-iot-solution/components/features/infrared;C:/Users/Public/esp-iot-solution/components/features/touchpad;C:/Users/Public/esp-iot-solution/submodule/esp-idf/components/ulp;C:/Users/Public/esp-iot-solution/components/features/ulp_monitor;C:/Users/Public/esp-iot-solution/submodule/esp-idf/components/esp_adc_cal;C:/Users/Public/esp-iot-solution/components/general/adc;C:/Users/Public/esp-iot-solution/components/general/base_class;C:/Users/Public/esp-iot-solution/components/general/button;C:/Users/Public/esp-iot-solution/components/general/debugs;C:/Users/Public/esp-iot-solution/components/general/led;C:/Users/Public/esp-iot-solution/components/general/light;C:/Users/Public/esp-iot-solution/components/general/ota;C:/Users/Public/esp-iot-solution/components/general/param;C:/Users/Public/esp-iot-solution/components/general/power_meter;C:/Users/Public/esp-iot-solution/components/general/relay;C:/Users/Public/esp-iot-solution/components/general/weekly_timer;C:/Users/Public/esp-iot-solution/components/wifi/iot_blufi;C:/Users/Public/esp-iot-solution/components/wifi/iot_espnow;C:/Users/Public/esp-iot-solution/components/wifi/smart_config;C:/Users/Public/esp-iot-solution/components/wifi/wifi_conn;C:/Users/Public/esp-iot-solution/components/spi_devices/epaper;C:/Users/Public/esp-iot-solution/components/spi_devices/lcd;C:/Users/Public/esp-iot-solution/components/spi_devices/xpt2046;C:/Users/Public/esp-iot-solution/components/i2c_devices/i2c_bus;C:/Users/Public/esp-iot-solution/components/i2c_devices/sensor/apds9960;C:/Users/Public/esp-iot-solution/components/i2c_devices/sensor/bh1750;C:/Users/Public/esp-iot-solution/components/i2c_devices/sensor/bme280;C:/Users/Public/esp-iot-solution/components/i2c_devices/sensor/ft5x06;C:/Users/Public/esp-iot-solution/components/i2c_devices/sensor/hdc2010;C:/Users/Public/esp-iot-solution/components/i2c_devices/sensor/hts221;C:/Users/Public/esp-iot-solution/components/i2c_devices/sensor/lis2dh12;C:/Users/Public/esp-iot-solution/components/i2c_devices/sensor/mpu6050;C:/Users/Public/esp-iot-solution/components/i2c_devices/sensor/mvh3004d;C:/Users/Public/esp-iot-solution/components/i2c_devices/sensor/veml6040;C:/Users/Public/esp-iot-solution/components/i2c_devices/others/at24c02;C:/Users/Public/esp-iot-solution/components/i2c_devices/others/ch450;C:/Users/Public/esp-iot-solution/components/i2c_devices/others/ht16c21;C:/Users/Public/esp-iot-solution/components/i2c_devices/others/is31fl3xxx;C:/Users/Public/esp-iot-solution/components/i2c_devices/others/mcp23017;C:/Users/Public/esp-iot-solution/components/i2c_devices/others/ssd1306;C:/Users/Public/esp-iot-solution/submodule/esp-idf/components/nghttp;C:/Users/Public/esp-iot-solution/components/network/mqtt;C:/Users/Public/esp-iot-solution/components/network/tcp;C:/Users/Public/esp-iot-solution/components/network/udp;C:/Users/Public/esp-iot-solution/submodule/esp-idf/components/json;C:/Users/Public/esp-iot-solution/submodule/esp-idf/components/openssl;C:/Users/Public/esp-iot-solution/components/platforms/alink;C:/Users/Public/esp-iot-solution/components/motor/stepper/a4988;C:/Users/Public/esp-iot-solution/components/motor/servo/servo_motor;C:/Users/Public/esp-iot-solution/components/framework/lowpower_framework;C:/Users/Public/esp-iot-solution/components/i2s_devices/lcd_common;C:/Users/Public/esp-iot-solution/components/i2s_devices/ili9806;C:/Users/Public/esp-iot-solution/components/i2s_devices/nt35510;C:/Users/Public/esp-iot-solution/components/hmi/lvgl_gui;C:/Users/Public/esp-iot-solution/components/hmi/ugfx_gui;C:/Users/Public/esp-iot-solution/components/hmi/gdrivers;C:/Users/Public/esp-iot-solution/submodule/esp-idf/components/asio;C:/Users/Public/esp-iot-solution/submodule [... cut ...]
-- Building for target esp32
CMake Error at C:/Users/Public/esp-iot-solution/submodule/esp-idf/tools/cmake/project.cmake:57 (idf_set_global_variables):
  Unknown CMake command "idf_set_global_variables".
Call Stack (most recent call first):
  CMakeLists.txt:20 (project)

-- Configuring incomplete, errors occurred!
[206/848] Building C object mbedtls/CMakeFiles/mbedtls.dir/mbedtls/library/ecp_curves.c.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

Many Thanks in advance!

zztiger123, 2019-12-24:
Hi, joel2106

I think the cause of your problem is that the esp-idf version is higher than 3.2.

Solution: Change the esp-idf version. Use the following commands to modify the version of esp-idf:
```
cd C:/esp-iot-solution/submodule/esp-idf
git checkout v3.2.2
git submodule update --init --recursive
```
and then you can set the new path:
```
set IDF_PATH=C:/esp-iot-solution/submodule/esp-idf
```
Now you can build successfully. If the error is still the same, you may need to delete the build folder under the current `eth2wifi` project, and then build again.

Best regards,

Xiehanxin, 2020-07-27:
@joel2106  hi, If your issue is now resolved, can you please close the issue.  If not, please let us konw. Thank you.

closed 2020-07-27 by joel2106