espressif/esp-iot-solution #65: A4988 driver issue with ledc_set_freq

chegewara, 2019-12-13:
Hi,
i am trying to get working A4988 and i can make it working with 2 issues.
1) bug in ledc.c i already opened issue in esp-idf repository,
2) there is problem with `ledc_set_freq` this exact line:
https://github.com/espressif/esp-iot-solution/blob/master/components/motor/stepper/a4988/a4988.cpp#L255

This line return ESP_OK, but motor is no longer working. When i comment out this line everything seems to works fine.

Thanks for code and help.

chegewara, 2019-12-16:
Hi,
is there possible to backport fix for this to v3.3 or at least to have patch i can use wiith AFR?

@projectgus any help please

projectgus, 2019-12-16:
Hi @chegewara,

This looks like an issue in the IDF driver not  the IoT solution, so I'll reply to you on that issue: https://github.com/espressif/esp-idf/issues/4476

chegewara, 2019-12-16:
Yes, i realized it after finding core issue, but didnt want to open another issue.
Thanks

closed 2019-12-16 by chegewara

projectgus, 2019-12-16:
@chegewara The fix in this driver is to choose LEDC_REF_TICK if you plan to set low frequency PWM speeds (approx <1MHz?), and LEDC_APB_CLK otherwise. But it should produce an error instead of silently turning off the PWM output.