espressif/esp-iot-solution #68: Unable to clone ugfx submodule

InfiniteYuan1, 2019-12-19:
You may encounter this error when updating submodule of esp-iot-solution.

```
Cloning into '/home/infiniteyuan/MyDocuments/Picture/temporary/esp-iot-solution/components/hmi/ugfx_gui/ugfx'...
fatal: unable to access 'https://git.ugfx.io/ugfx/ugfx.git/': SSL certificate problem: certificate has expired
fatal: clone of 'https://git.ugfx.io/ugfx/ugfx.git' into submodule path '/home/infiniteyuan/MyDocuments/Picture/temporary/esp-iot-solution/components/hmi/ugfx_gui/ugfx' failed
Failed to clone 'components/hmi/ugfx_gui/ugfx'. Retry scheduled
```

Solution:

```
env GIT_SSL_NO_VERIFY=true git submodule update --init --recursive
```

zztiger123, 2019-12-24:
Hi, InfiniteYuan1 

Thank you for your support, I would appreciate it!

best regards,

wibsea, 2019-12-24:
Brave decision! :) Maybe wait a bit?

carson2017, 2019-12-25:
Great! Thanks a lot, I have been puzzled by this problem for a long time

Alexlizx, 2020-05-08:
I now get a slightly different error:
`Cloning into 'C:/Users/Documents/ESP-IDF/esp-iot-solution/components/hmi/ugfx_gui/ugfx'...
fatal: unable to access 'https://git.ugfx.io/ugfx/ugfx.git/': error:1408F10B:SSL routines:ssl3_get_record:wrong version number`

I'm working on a Windows environment so "env GIT_SSL_NO_VERIFY=true" does not work. I tried other ways to disable ssl verification but always come back to the same error.

Xiehanxin, 2020-07-27:
@InfiniteYuan  If this issue is now resolved, please close it. Thank you.

MrityunjaiKumar, 2020-12-06:
Its a SSL issue, the git server for this dependency got ssl certificate issue.

In [.gitmodules](https://github.com/espressif/esp-iot-solution/blob/master/.gitmodules), I changed  

https://git.ugfx.io/ugfx/ugfx.git
to
http://git.ugfx.io/ugfx/ugfx.git 

there after use following command.

`git submodule sync`

It worked for me!

closed 2023-07-10 by leeebo