espressif/esp-iot-solution #69: snprintf input parameter error in lcd

gengyuchao, 2019-12-25:
Hello, I found that '% d' directive output may be truncated writing between 2 and 12 bytes (Consider symbols and string terminator '\ 0') into a region of size 10, So the `char tmp [10];` should be `char tmp [12];`

https://github.com/espressif/esp-iot-solution/blob/ab1723d42515e7e6562f5166b828391eb8a294aa/components/spi_devices/lcd/iot_lcd.cpp#L575-L580

https://github.com/espressif/esp-iot-solution/blob/ab1723d42515e7e6562f5166b828391eb8a294aa/components/spi_devices/lcd/iot_lcd.cpp#L507-L512

InfiniteYuan1, 2020-01-22:
Of course, Please carefully about string terminator.

Xiehanxin, 2020-07-27:
@gengyuchao  hi,If your issue is now resolved, can you please close the issue. Thank you.

gengyuchao, 2020-08-12:
@Xiehanxin This problem is not resolved, I have shown you the solution. Please take a look, thank you.

leeebo, 2023-07-10:
@gengyuchao Thanks for your contribution, as we have dropped the support for `spi_devices/lcd`, the code will not be merged. for the new LCD project, please use the new LCD driver https://github.com/espressif/esp-idf/tree/master/components/esp_lcd

closed 2023-07-10 by leeebo