espressif/esp-iot-solution #71: Issues with iot_i2s_lcd_write_cmd in 8-bit LCD mode

csobrinho, 2020-01-15:
INSTRUCTIONS
============

## Environment

- Development Kit:      Hitec WiFi Kit 32
- Module or chip used:  ESP32
- IDF version (run ``git describe --tags`` to find it): 
    v4.1-dev-1935-g647cb628a
- Build System:        CMake
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0
- Operating System:     macOS
- Power Supply:         USB

## Problem Description

Having some issues with the `iot_i2s_lcd_write_cmd` component in 8 bit mode. Also the clock `I2S[i2s_num]->clkm_conf.clkm_div_num = 2;` doesn't seem to work reliably with 8 bit mode. Had to use `4`.

### Expected Behavior

`iot_i2s_lcd_write_cmd` (that uses the FIFO) should work when a DMA is also underway. My impression is that in 8 bit mode, the `I2S[i2s_num]->state.tx_idle` doesn't get updated quickly and the second write of the LOW byte cancels the previous one. I believe that adding a small us delay fixed it but my memory is not very sure here. 

### Actual Behavior
A few issues that I noticed and validated with the MSO1074z logic analyzer:
 
* `iot_i2s_lcd_write_cmd` doesn't seem to work properly. I can see the RS line going down and up but no data is written. I confirmed this with the LA. Replacing it with the `i2s_lcd8_write_data` and toggling the RS line manually fixed this issue (https://github.com/csobrinho/hydroponics/blob/11b9b3b9bd3c9ae62058ade886d847e1b19deb0f/main/driver/i2s_lcd8.c#L102-L104). This problem was independent on the clock and was the same for `clkm_div_num = {2|4|8}`

* `I2S[i2s_num]->clkm_conf.clkm_div_num = 2` should be `3` or `4` for 8 bit mode. From what I could understand from the RTM, from the source code comments and from the forum, for 8 bits the clock is divided by two and must be `3` or `4`. With `3` I got some issues during reboot where in some cases the byte was transmitted and in other cases it wasn't. I also remember reading some info regarding 8bit mode and I2S1 being better than I2S0 but I can't find my reference.

This board has the CPUs set at 240 MHz so not sure if that was the reason but it was strange. If I defined a bus with of 16 I saw the data changing but I missing half of the data.

### Steps to reproduce

1. call `iot_i2s_lcd_write_cmd`.
2. call `iot_i2s_lcd_write_data`. Only the rs toggle and the iot_i2s_lcd_write_data was properly transmitted.

From https://github.com/csobrinho/hydroponics/blob/master/main/display/lcd.c#L18-L32:

```c
static i2s_lcd8_dev_t dev = {
        .base = {
                .i2s_lcd_conf = {
                        .data_width = 8,
                        .data_io_num = {
                                GPIO_NUM_23, GPIO_NUM_18, GPIO_NUM_14, GPIO_NUM_27,
                                GPIO_NUM_26, GPIO_NUM_21, GPIO_NUM_22, GPIO_NUM_19,
                        },
                        .ws_io_num = GPIO_NUM_17,
                        .rs_io_num = GPIO_NUM_12,
                },
                .i2s_port = I2S_NUM_1,
        },
        .rst_io_num = OLED_RESET,
};
```

### Code to reproduce this issue
I have made a wrapper around the iot_i2s_lcd component here: https://github.com/csobrinho/hydroponics/blob/master/main/driver/i2s_lcd8.c

## Debug Logs
N/A but can provide upon request.

## Other items if possible

[image]

There are other issues (#19) and (#26) with LCD in 8 bit mode but I believe they are not related to this particular issue.

I'm using a RM68090 display and it accepts 8 bit commands sent as 0b00000000{cmd} bits so no need to change the function to only send one byte.

- [x] sdkconfig file (attach the sdkconfig file from your project folder)
https://github.com/csobrinho/hydroponics/blob/master/sdkconfig

- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

InfiniteYuan1, 2020-01-22:
This component has not been adapted for other IDF versions.

nopnop2002, 2021-02-07:
@csobrinho 

I don't know your TFT driver, but give it a try.

It's work using esp-idf version:v4.3-dev-2586

https://github.com/nopnop2002/esp-idf-parallel-tft

csobrinho, 2021-02-08:
Thanks nopnop2002. I still have my I2S LCD driver working but I'm moving to a MHS40 screen with an 80 MHz SPI interface so I'll soon deprecate that driver. I did manage to get really fast speeds though.

leeebo, 2023-07-10:
From ESP-IDF v5.0, we suggest using the new general LCD driver https://github.com/espressif/esp-idf/tree/master/examples/peripherals/lcd

closed 2023-07-10 by leeebo