espressif/esp-iot-solution #73: eth2wifi error

chenxiuqimr, 2020-01-24:
开发板：ESP32-Ethernet-Kit V1.1

git clone --recursive https://github.com/espressif/esp-iot-solution
git submodule update --init --recursive
export IOT_SOLUTION_PATH=~/esp/esp-iot-solution

一切正常，但是运行 esp-iot-solution的eth2wifi示例，插上网线时出现错误 emac rx buf err   
我们产品需要用到wifi转网线功能给带网口的设备上网，查找了网上很多资料没有找到解决办法，非常着急，能否给些提示如何解决该问题，谢谢指导！

-----------make menuconfig 配置如下------------------------
#
# Compiler options
#
CONFIG_OPTIMIZATION_LEVEL_RELEASE=y
#
# IoT Example - Ethernet to WiFi Demo Configuration
#
CONFIG_PHY_IP101=y
CONFIG_PHY_ADDRESS=1
CONFIG_PHY_USE_POWER_PIN=y
CONFIG_PHY_POWER_PIN=5
CONFIG_PHY_SMI_MDC_PIN=23
CONFIG_PHY_SMI_MDIO_PIN=18
CONFIG_ETH_TO_STATION_MODE=y
#
# Wi-Fi
#
CONFIG_ESP32_WIFI_STATIC_RX_BUFFER_NUM=16
CONFIG_ESP32_WIFI_DYNAMIC_RX_BUFFER_NUM=64
CONFIG_ESP32_WIFI_DYNAMIC_TX_BUFFER_NUM=64
CONFIG_ESP32_WIFI_TX_BA_WIN=16
CONFIG_ESP32_WIFI_RX_BA_WIN=16
#
# ESP32-specific
#
CONFIG_ESP32_DEFAULT_CPU_FREQ_240=y
CONFIG_ESP32_DEFAULT_CPU_FREQ_MHZ=240
#
# Ethernet
#
CONFIG_EMAC_TASK_PRIORITY=5
#
# FreeRTOS
#
CONFIG_FREERTOS_HZ=1000

-----------插上网线可以上网但是速度很慢，错误信息如下------------------------------------------
MONITOR
--- idf_monitor on COM74 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6500
load:0x40078000,len:9484
load:0x40080400,len:7008
entry 0x40080744
I (29) boot: ESP-IDF v3.2.2 2nd stage bootloader
I (29) boot: compile time 19:09:37
I (29) boot: Enabling RNG early entropy source...
I (33) qio_mode: Enabling default flash chip QIO
I (39) boot: SPI Speed      : 40MHz
I (43) boot: SPI Mode       : QIO
I (47) boot: SPI Flash Size : 4MB
I (51) boot: Partition Table:
I (54) boot: ## Label            Usage          Type ST Offset   Length
I (62) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (69) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (77) boot:  2 factory          factory app      00 00 00010000 00100000
I (84) boot: End of partition table
I (88) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x13aec ( 80620) map
I (121) esp_image: segment 1: paddr=0x00023b14 vaddr=0x3ffb0000 size=0x02dcc ( 11724) load
I (125) esp_image: segment 2: paddr=0x000268e8 vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at D:/Project/esp/msys32/home/dev/esp/esp-iot-solution/submodule/esp-idf/components/freerto
s/xtensa_vectors.S:1779

I (128) esp_image: segment 3: paddr=0x00026cf0 vaddr=0x40080400 size=0x09320 ( 37664) load
I (149) esp_image: segment 4: paddr=0x00030018 vaddr=0x400d0018 size=0x5e5cc (386508) map
0x400d0018: _stext at ??:?

I (263) esp_image: segment 5: paddr=0x0008e5ec vaddr=0x40089720 size=0x05aa0 ( 23200) load
0x40089720: disableAllWdts at D:/Project/esp/msys32/home/dev/esp/esp-iot-solution/submodule/esp-idf/components/esp32/pan
ic.c:707
 (inlined by) panicHandler at D:/Project/esp/msys32/home/dev/esp/esp-iot-solution/submodule/esp-idf/components/esp32/pan
ic.c:309

I (280) boot: Loaded app from partition at offset 0x10000
I (280) boot: Disabling RNG early entropy source...
I (281) cpu_start: Pro cpu up.
I (284) cpu_start: Starting app cpu, entry point is 0x40081018
0x40081018: call_start_cpu1 at D:/Project/esp/msys32/home/dev/esp/esp-iot-solution/submodule/esp-idf/components/esp32/cp
u_start.c:246

I (0) cpu_start: App cpu up.
I (294) heap_init: Initializing. RAM available for dynamic allocation:
I (301) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (308) heap_init: At 3FFB85B8 len 00027A48 (158 KiB): DRAM
I (314) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (320) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (327) heap_init: At 4008F1C0 len 00010E40 (67 KiB): IRAM
I (333) cpu_start: Pro cpu start user code
I (126) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (185) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (190) emac: emac reset done
I (191) wifi: wifi driver task: 3ffc9dd8, prio:23, stack:3584, core=0
I (193) wifi: wifi firmware version: 693c7b6
I (197) wifi: config NVS flash: enabled
I (201) wifi: config nano formating: disabled
I (205) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (215) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (238) wifi: Init dynamic tx buffer num: 64
I (239) wifi: Init data frame dynamic rx buffer num: 64
I (239) wifi: Init management frame dynamic rx buffer num: 64
I (242) wifi: Init management short buffer num: 32
I (247) wifi: Init static rx buffer size: 1600
I (251) wifi: Init static rx buffer num: 16
I (254) wifi: Init dynamic rx buffer num: 64
SYSTEM_EVENT_ETH_CONNECTED
I (4845) phy: phy_version: 4008, 544f89f, Jan 24 2019, 14:54:06, 0, 0
I (4846) wifi: mode : sta (84:0d:8e:d3:67:84)
SYSTEM_EVENT_STA_START
E (4847) emac: emac rx buf err
SYSTEM_EVENT_STA_START
E (4991) emac: emac rx buf err
E (5290) emac: emac rx buf err
E (5790) emac: emac rx buf err
E (5991) emac: emac rx buf err
I (6174) wifi: n:11 2, o:1 0, ap:255 255, sta:11 2, prof:1
I (6734) wifi: state: init -> auth (b0)
I (6738) wifi: state: auth -> assoc (0)
I (6744) wifi: state: assoc -> run (10)
I (6758) wifi: connected with XQ_WLAN0, channel 11
I (6758) wifi: pm start, type: 1

SYSTEM_EVENT_STA_CONNECTED
E (6991) emac: emac rx buf err
E (7290) emac: emac rx buf err
E (7315) emac: emac rx buf err
E (7315) emac: emac rx buf err
E (7316) emac: emac rx buf err
E (7317) emac: emac rx buf err
E (7336) emac: emac rx buf err
E (7373) emac: emac rx buf err
E (7373) emac: emac rx buf err
E (7373) emac: emac rx buf err
E (7790) emac: emac rx buf err

chenxiuqimr, 2020-02-04:
make menuconfig 把Component config--->Ethernet下的Enable received buffers be copied to Layer3 from Layer2的勾去掉，后，不在报emac：emac rx buf err这个错误。
但是速度还是很慢，在网络使用过程一直会出现以下提示：
I (0) cpu_start: App cpu up.
I (242) heap_init: Initializing. RAM available for dynamic allocation:
I (243) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (244) heap_init: At 3FFB85C8 len 00027A38 (158 KiB): DRAM
I (244) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (245) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (246) heap_init: At 4008F1BC len 00010E44 (67 KiB): IRAM
I (247) cpu_start: Pro cpu start user code
I (33) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (100) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (104) emac: emac reset done
I (106) wifi: wifi driver task: 3ffc9e68, prio:23, stack:3584, core=0
I (106) wifi: wifi firmware version: 693c7b6
I (107) wifi: config NVS flash: enabled
I (107) wifi: config nano formating: disabled
I (107) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (108) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (125) wifi: Init dynamic tx buffer num: 64
I (126) wifi: Init data frame dynamic rx buffer num: 64
I (126) wifi: Init management frame dynamic rx buffer num: 64
I (126) wifi: Init management short buffer num: 32
I (127) wifi: Init static rx buffer size: 1600
I (127) wifi: Init static rx buffer num: 16
I (128) wifi: Init dynamic rx buffer num: 64
SYSTEM_EVENT_ETH_CONNECTED
I (4168) phy: phy_version: 4008, 544f89f, Jan 24 2019, 14:54:06, 0, 0
I (4170) wifi: mode : sta (84:0d:8e:d3:67:84)
SYSTEM_EVENT_STA_START
SYSTEM_EVENT_STA_START
I (5502) wifi: n:11 2, o:1 0, ap:255 255, sta:11 2, prof:1
I (6063) wifi: state: init -> auth (b0)
I (6066) wifi: state: auth -> assoc (0)
I (6077) wifi: state: assoc -> run (10)
I (6105) wifi: connected with XQ_WLAN0, channel 11
I (6105) wifi: pm start, type: 1

SYSTEM_EVENT_STA_CONNECTED
I (17716) wifi: state: run -> init (6c0)
I (17717) wifi: pm stop, total sleep time: 4105430 us / 11611231 us

I (17717) wifi: n:11 0, o:11 2, ap:255 255, sta:11 2, prof:1
SlYSTEM_EVENT_STA_DISCONNECTED
I (17840) wifi: n:11 2, o:11 0, ap:255 255, sta:11 2, prof:1
I (17841) wifi: state: init -> auth (b0)
I (17845) wifi: state: auth -> assoc (0)
I (17853) wifi: state: assoc -> run (10)
I (17866) wifi: connected with XQ_WLAN0, channel 11
I (17867) wifi: pm start, type: 1

SYSTEM_EVENT_STA_CONNECTED
I (18120) wifi: state: run -> init (6c0)
I (18121) wifi: pm stop, total sleep time: 95167 us / 254380 us

I (18122) wifi: n:11 0, o:11 2, ap:255 255, sta:11 2, prof:1
SlYSTEM_EVENT_STA_DISCONNECTED
I (18244) wifi: n:11 2, o:11 0, ap:255 255, sta:11 2, prof:1
I (18246) wifi: state: init -> auth (b0)
I (18249) wifi: state: auth -> assoc (0)
I (18257) wifi: state: assoc -> run (10)
I (18269) wifi: connected with XQ_WLAN0, channel 11
I (18270) wifi: pm start, type: 1

SYSTEM_EVENT_STA_CONNECTED
I (30195) wifi: state: run -> init (6c0)
I (30196) wifi: pm stop, total sleep time: 3389065 us / 11926243 us

I (30197) wifi: n:11 0, o:11 2, ap:255 255, sta:11 2, prof:1
SlYSTEM_EVENT_STA_DISCONNECTED
I (30320) wifi: n:11 2, o:11 0, ap:255 255, sta:11 2, prof:1
I (30321) wifi: state: init -> auth (b0)
I (30325) wifi: state: auth -> assoc (0)
I (30331) wifi: state: assoc -> run (10)
I (30351) wifi: connected with XQ_WLAN0, channel 11
I (30352) wifi: pm start, type: 1

SYSTEM_EVENT_STA_CONNECTED
I (33982) wifi: state: run -> init (6c0)
I (33983) wifi: pm stop, total sleep time: 697516 us / 3631434 us

I (33984) wifi: n:11 0, o:11 2, ap:255 255, sta:11 2, prof:1
SlYSTEM_EVENT_STA_DISCONNECTED
I (34106) wifi: n:11 2, o:11 0, ap:255 255, sta:11 2, prof:1
I (34108) wifi: state: init -> auth (b0)
I (34121) wifi: state: auth -> assoc (0)
I (34129) wifi: state: assoc -> run (10)
I (34144) wifi: connected with XQ_WLAN0, channel 11
I (34145) wifi: pm start, type: 1

SYSTEM_EVENT_STA_CONNECTED
I (35824) wifi: state: run -> init (6c0)
I (35825) wifi: pm stop, total sleep time: 97169 us / 1680269 us

I (35825) wifi: n:11 0, o:11 2, ap:255 255, sta:11 2, prof:1
SlYSTEM_EVENT_STA_DISCONNECTED
I (35948) wifi: n:11 2, o:11 0, ap:255 255, sta:11 2, prof:1
I (35950) wifi: state: init -> auth (b0)
I (35953) wifi: state: auth -> assoc (0)
I (35960) wifi: state: assoc -> run (10)
I (35979) wifi: connected with XQ_WLAN0, channel 11
I (35980) wifi: pm start, type: 1

SYSTEM_EVENT_STA_CONNECTED
I (37360) wifi: state: run -> init (6c0)
I (37361) wifi: pm stop, total sleep time: 193348 us / 1380978 us

I (37361) wifi: n:11 0, o:11 2, ap:255 255, sta:11 2, prof:1
SlYSTEM_EVENT_STA_DISCONNECTED
I (37484) wifi: n:11 2, o:11 0, ap:255 255, sta:11 2, prof:1
I (37485) wifi: state: init -> auth (b0)
I (37489) wifi: state: auth -> assoc (0)
I (37494) wifi: state: assoc -> run (10)
I (37506) wifi: connected with XQ_WLAN0, channel 11
I (37507) wifi: pm start, type: 1

SYSTEM_EVENT_STA_CONNECTED
。。。。。。。

leeebo, 2023-07-10:
Hi @cendikj to achieve eth2wifi or any other bridge types, please refer the new esp-iot-bridge solution, https://github.com/espressif/esp-iot-bridge

closed 2023-07-10 by leeebo