espressif/esp-iot-solution #76: Feature Request: Make simpler to reuse components from this project

nicola-lunghi, 2020-03-01:
Hi,
there are a lot of nice components from this repository and could be really useful to be able to reuse those easily from an external project (instead of copying it)

Can you split up the components folder of this project into a separate project that can be included as s submodule in other projects?

My idea would be to have some sort of "library manager" where the user can specify what library needs and that it clones it in the component subfolder....

I also really like the idea of object in a bus
https://github.com/espressif/esp-iot-solution/tree/master/components/i2c_devices/i2c_bus

I think if this was in a separate project would make more easily for other users to contribute and add more i2c/spi/i2s devices

Cheers
Nick

kumekay, 2020-03-03:
Hi Nick, 
thank you for the issue - the idea is nice, 
and we are working on component manager that is currently we develop internally, 
we will update this issue when public beta will be available.

Xiehanxin, 2020-07-27:
@nicola-lunghi 
Hi Nick, If your issue is now resolved, could you please close the issue. if you still have problem, please tell us . Thank you.

nicola-lunghi, 2020-07-27:
Sorry i forgot

closed 2020-07-27 by nicola-lunghi