espressif/esp-iot-solution #78: Please upgrade the least esp-idf version

jerry-zhu93, 2020-03-31:
Please upgrade the least esp-idf version,thanks!

leeebo, 2020-08-13:
Hi @jerry-zhu93  , you can use [esp-iot-solution/tree/release/v1.1](https://github.com/espressif/esp-iot-solution/tree/release/v1.1) branch in your idf 4.0 project.

cadrjr, 2020-08-13:
Thanks so much @leeebo ! I had spent hours trying to get touch_pad_evb example to build .....and now it does!

cadrjr, 2020-08-13:
Now I have another problem with touch_pad_evb example.
I want to enable the ESP-Tuning Tool on this example.
I run idf.py menuconfig .... but I don't get the 
"IoT Solution settings > IoT Components Management > General functions > Touch Sensor" menu option which would enable me to do this.
I think this has something to do with SDKCONFIG and Kconfig.projbuild...... bear with me....I'm afraid I'm a noobie!

alic-maker, 2020-08-17:
To make the search easier, if the build problem is resolved, close it and reopen a new one for the new issue.

cadrjr, 2020-08-17:
> @cadrjr To make the search easier, if the compilation problem is resolved, close it and reopen a new one for the new issue.

Thank you. Will do.

closed 2023-07-10 by leeebo