espressif/esp-iot-solution #83: Compiling esp-iot-solution examples with esp-idf V4.0

BHAKTIDESHPANDE, 2020-05-20:
Is esp-idf V4.0 support this esp-iot-solutin library? I am trying to build example code eth2wifi with esp-idf V4.0 but getting errors. Please check log file.
[log.txt](https://github.com/espressif/esp-iot-solution/files/4658346/log.txt)

0xjakob, 2020-05-22:
Hi @BHAKTIDESHPANDE, I'm afraid currently esp-iot-solution (master branch) works only with esp-idf v3.2. But we're planning to update it in the medium term future.

BHAKTIDESHPANDE, 2020-05-22:
Thanks for the response.
Can I compile Eth2wifi example only using esp-idf ? Does it require some components from esp-iot solution library?

BHAKTIDESHPANDE, 2020-05-25:
Hi @0xjakob any comments on my query?

BHAKTIDESHPANDE, 2020-05-29:
Hi,

As you said the esp-iot-solution (master branch) works only with esp-idf v3.2, we are stuck here now. How do I run eth2wifi demo code with esp-idf v4.0? Is it possible to do it by adding required components from esp-iot-solution to esp-idf v4.0?

Waiting for your prompt reply.

Regards,
Bhakti.

leeebo, 2020-08-13:
Hi @BHAKTIDESHPANDE, you can use [esp-iot-solution/tree/release/v1.1](https://github.com/espressif/esp-iot-solution/tree/release/v1.1) branch in your idf 4.0 project.  In this branch we update the eth2wifi example.

closed 2023-07-10 by leeebo