espressif/esp-iot-solution #84: eth2wifi example with esp-idf v4.0

BHAKTIDESHPANDE, 2020-05-28:
How do I run eth2wifi code from esp-iot-solution library with only esp-idf v4.0, (As esp-iot-solution supports esp-idf v3.2 only)
Can I do it by adding required components from esp-iot-solution to esp-idf v4.0?

Thanks in advance.
Bhakti.

leeebo, 2020-07-27:
The eth2wifi example will be updated to esp-idf v4.0.1 in early Augest.

coollofty, 2020-08-04:
> The eth2wifi example will be updated to esp-idf v4.0.1 in early Augest.

 我也在关注这个eth2wifi的demo，由于近期暂时没有条件在硬件上实际运行一下，所以想问：能否实现wifi2eth呢？也就是ESP32用wifi连到上级路由器，然后用eth带着后面的电脑上网？是说只需要简单的修改一下demo中的数据传递的方向就可以实现吗？如果不是，能否提供一个wifi2eth的demo呢？

salocinx, 2020-08-04:
**Translation of coollofty's comment:**
I'm also looking at this eth2wifi demo, and since I'm not in a position to actually run it on hardware for the near future, I'd like to ask: is it possible to implement wifi2eth? That is, the ESP32 connects to the superior router with wifi, and then uses eth to bring the computer behind it online? Does that mean it can be achieved by simply modifying the direction of the data passed in the demo? If not, can you provide a demo of wifi2eth?

mbober1, 2022-07-10:
Is there any hope that the example update will be released this year?

leeebo, 2023-07-10:
@mbober1 @salocinx  We have a more general solution to support it now, please refer  https://github.com/espressif/esp-iot-bridge

closed 2023-07-10 by leeebo