espressif/esp-iot-solution #85: Failed to Clone Submodules

earosenfeld, 2020-06-01:
Submodules are currently not able to be recursively cloned. Folders are empty, despite indications showing files have been moved to the folder. 

D:\esp-iot-solution>git clone https://github.com/espressif/esp-iot-solution/ --recursive-submodules
error: unknown option `recursive-submodules'
usage: git clone [<options>] [--] <repo> [<dir>]

    -v, --verbose         be more verbose
    -q, --quiet           be more quiet
    --progress            force progress reporting
    -n, --no-checkout     don't create a checkout
    --bare                create a bare repository
    --mirror              create a mirror repository (implies bare)
    -l, --local           to clone from a local repository
    --no-hardlinks        don't use local hardlinks, always copy
    -s, --shared          setup as shared repository
    --recurse-submodules[=<pathspec>]
                          initialize submodules in the clone
    -j, --jobs <n>        number of submodules cloned in parallel
    --template <template-directory>
                          directory from which templates will be used
    --reference <repo>    reference repository
    --reference-if-able <repo>
                          reference repository
    --dissociate          use --reference only while cloning
    -o, --origin <name>   use <name> instead of 'origin' to track upstream
    -b, --branch <branch>
                          checkout <branch> instead of the remote's HEAD
    -u, --upload-pack <path>
                          path to git-upload-pack on the remote
    --depth <depth>       create a shallow clone of that depth
    --shallow-since <time>
                          create a shallow clone since a specific time
    --shallow-exclude <revision>
                          deepen history of shallow clone, excluding rev
    --single-branch       clone only one branch, HEAD or --branch
    --no-tags             don't clone any tags, and make later fetches not to follow them
    --shallow-submodules  any cloned submodules will be shallow
    --separate-git-dir <gitdir>
                          separate git dir from working tree
    -c, --config <key=value>
                          set config inside the new repository
    -4, --ipv4            use IPv4 addresses only
    -6, --ipv6            use IPv6 addresses only
    --filter <args>       object filtering

D:\esp-iot-solution>git clone https://github.com/espressif/esp-iot-solution/ --recurse-submodules
Cloning into 'esp-iot-solution'...
remote: Enumerating objects: 7537, done.
Rremote: Total 7537 (delta 0), reused 0 (delta 0), pack-reused 7537                                              | 4.18 MiB/s
Receiving objects: 100% (7537/7537), 56.98 MiB | 4.05 MiB/s, done.
Resolving deltas: 100% (3919/3919), done.
Checking out files: 100% (1437/1437), done.
Submodule 'components/hmi/lvgl_gui/lvgl' (https://github.com/littlevgl/lvgl.git) registered for path 'components/hmi/lvgl_gui/lvgl'
Submodule 'components/hmi/ugfx_gui/ugfx' (https://git.ugfx.io/ugfx/ugfx.git) registered for path 'components/hmi/ugfx_gui/ugfx'
Submodule 'components/network/mqtt/espmqtt' (https://github.com/tuanpmt/espmqtt.git) registered for path 'components/network/mqtt/espmqtt'
Submodule 'components/platforms/alink/esp32-alink' (https://github.com/espressif/esp32-alink.git) registered for path 'components/platforms/alink/esp32-alink'
Submodule 'Adafruit-GFX-Library' (https://github.com/adafruit/Adafruit-GFX-Library.git) registered for path 'components/spi_devices/lcd/Adafruit-GFX-Library'
Submodule 'submodule/esp-idf' (https://github.com/espressif/esp-idf.git) registered for path 'submodule/esp-idf'
Cloning into 'D:/esp-iot-solution/esp-iot-solution/components/hmi/lvgl_gui/lvgl'...
remote: Enumerating objects: 10, done.
remote: Counting objects: 100% (10/10), done.
remote: Compressing objects: 100% (10/10), done.
remote: Total 31931 (delta 2), reused 0 (delta 0), pack-reused 31921
Receiving objects: 100% (31931/31931), 31.98 MiB | 5.29 MiB/s, done.
Resolving deltas: 100% (25770/25770), done.
Cloning into 'D:/esp-iot-solution/esp-iot-solution/components/hmi/ugfx_gui/ugfx'...
fatal: unable to access 'https://git.ugfx.io/ugfx/ugfx.git/': Failed to connect to git.ugfx.io port 443: Timed out
fatal: clone of 'https://git.ugfx.io/ugfx/ugfx.git' into submodule path 'D:/esp-iot-solution/esp-iot-solution/components/hmi/ugfx_gui/ugfx' failed
Failed to clone 'components/hmi/ugfx_gui/ugfx'. Retry scheduled
Cloning into 'D:/esp-iot-solution/esp-iot-solution/components/network/mqtt/espmqtt'...
remote: Enumerating objects: 238, done.
remote: Counting objects: 100% (238/238), done.
remote: Compressing objects: 100% (154/154), done.
remote: Total 2212 (delta 119), reused 198 (delta 83), pack-reused 1974
Receiving objects: 100% (2212/2212), 1.46 MiB | 3.64 MiB/s, done.
Resolving deltas: 100% (1173/1173), done.
Cloning into 'D:/esp-iot-solution/esp-iot-solution/components/platforms/alink/esp32-alink'...
remote: Enumerating objects: 386, done.
remote: Total 386 (delta 0), reused 0 (delta 0), pack-reused 386
Receiving objects: 100% (386/386), 777.12 KiB | 3.15 MiB/s, done.
Resolving deltas: 100% (226/226), done.
Cloning into 'D:/esp-iot-solution/esp-iot-solution/components/spi_devices/lcd/Adafruit-GFX-Library'...
remote: Enumerating objects: 55, done.
remote: Counting objects: 100% (55/55), done.
remote: Compressing objects: 100% (46/46), done.
remote: Total 2038 (delta 29), reused 22 (delta 8), pack-reused 1983
Receiving objects: 100% (2038/2038), 1.80 MiB | 4.79 MiB/s, done.
Resolving deltas: 100% (1360/1360), done.
Cloning into 'D:/esp-iot-solution/esp-iot-solution/submodule/esp-idf'...
remote: Enumerating objects: 18, done.
remote: Counting objects: 100% (18/18), done.
remote: Compressing objects: 100% (18/18), done.
remote: Total 159844 (delta 5), reused 4 (delta 0), pack-reused 159826
Receiving objects: 100% (159844/159844), 99.99 MiB | 3.47 MiB/s, done.
Resolving deltas: 100% (117292/117292), done.
Cloning into 'D:/esp-iot-solution/esp-iot-solution/components/hmi/ugfx_gui/ugfx'...
fatal: unable to access 'https://git.ugfx.io/ugfx/ugfx.git/': Failed to connect to git.ugfx.io port 443: Timed out
fatal: clone of 'https://git.ugfx.io/ugfx/ugfx.git' into submodule path 'D:/esp-iot-solution/esp-iot-solution/components/hmi/ugfx_gui/ugfx' failed
Failed to clone 'components/hmi/ugfx_gui/ugfx' a second time, aborting

longtengj, 2020-06-24:
you can use a stable version.such as:
git clone -b v3.2.3 --recursive https://github.com/espressif/esp-idf.git

zhouli-esp, 2020-07-20:
You can try this command `git submodule update --init --recursive` to update all submodules.

`"Failed to connect to git.ugfx.io port 443: Timed out"` This error seems to be a network problem.

Xiehanxin, 2020-07-27:
hi, If your issue is now resolved, could you please close the issue. if you still have problem, please tell us . Thank you.

MrityunjaiKumar, 2020-12-06:
Its a SSL issue, the git server for this dependency got ssl certificate issue.

In .gitmodules, I changed

https://git.ugfx.io/ugfx/ugfx.git
to
http://git.ugfx.io/ugfx/ugfx.git

there after use following command.

git submodule sync

It worked for me!

alic-maker, 2020-12-07:
@MrityunjaiKumar Thanks for your suggestion

closed 2023-07-10 by leeebo