espressif/esp-iot-solution #87: Power meter on ESP8266

colesnicov, 2020-07-12:
## Environment

- Development Kit:      WEMOS D1 mini
- IDF version: v3.2-502-gda3362ec
- Build System:         Make
- Compiler version: xtensa-lx106-elf-gcc (crosstool-NG crosstool-ng-1.22.0-100-ge567ec7) 5.2.0
- Operating System:     Linux
- Power Supply:         USB

## Problem Description

Hello. I try to compile the Power_meter component but I get this error:
`power_meter.c:20:25: fatal error: driver/pcnt.h: No such file or directory`

alic-maker, 2020-08-17:
What command did you use to compile this component?

I recommend that you use  [release/v1.1](https://github.com/espressif/esp-iot-solution/tree/release/v1.1)

colesnicov, 2020-08-17:
I just type 
```
make
```

alic-maker, 2020-08-18:
You can try the following command to compile
```shell
cd tools/unit-test-app
make TEST_COMPONENTS="power_meter" -j8  && make flash monitor
```

closed 2023-07-10 by leeebo