espressif/esp-iot-solution #88: wifi2eth?

coollofty, 2020-08-06:
I'm also looking at this eth2wifi demo, and since I'm not in a position to actually run it on hardware for the near future, I'd like to ask: is it possible to implement wifi2eth? That is, the ESP32 connects to the superior router with wifi, and then uses eth to bring the computer behind it online? Does that mean it can be achieved by simply modifying the direction of the data passed in the demo? If not, can you provide a demo of wifi2eth?

leeebo, 2020-08-12:
Hi, an example of wifi2eth is already in [Ethernet to wifi station data forwarding demo](https://github.com/espressif/esp-iot-solution/tree/master/examples/eth2wifi#ethernet-to-wifi-station-data-forwarding-demo), please try it.

coollofty, 2020-08-12:
> Hi, an example of wifi2eth is already in [Ethernet to wifi station data forwarding demo](https://github.com/espressif/esp-iot-solution/tree/master/examples/eth2wifi#ethernet-to-wifi-station-data-forwarding-demo), please try it.

it is a eth->wifi, not wifi->eth

leeebo, 2020-08-12:
**Ethernet to wifi station data forwarding demo**
It's like a wireless adapter. You can connect the ESP32 with PC through ethernet, and the PC can access the Internet, if esp32 wifi station connected the router which can access the Internet.

**Ethernet to wifi softap data forwarding demo**
It's like a WiFi hotspots. You can connect the ESP32 with PC through ethernet and other wifi station can connect this wifi softap.But the PC must enable DHCP server, otherwise, the wifi station device cann't get ip.

coollofty, 2020-08-12:
So, thank you, this information is very helpful to me :)

coollofty, 2020-08-14:
> **Ethernet to wifi station data forwarding demo**
> It's like a wireless adapter. You can connect the ESP32 with PC through ethernet, and the PC can access the Internet, if esp32 wifi station [...]

I purchased the hardware of ESP32-Ethernet-Kiv V1.2, then compiled & run the eth2wifi demo, using the sta mode, ESP32 showed that the connection to the superior router was successful, and the PC was also successfully connected to the ESP32, but the PC could not access the Internet , Whether it is manually setting IP or DHCP, what are the possible reasons?

```
I (592) system_api: Base MAC address is not set
I (592) system_api: read default base MAC address from EFUSE
I (615) wifi:wifi driver task: 3ffc5fa0, prio:23, stack:6656, core=0
I (629) wifi:wifi firmware version: 517093f
I (630) wifi:wifi certification version: v7.0
I (630) wifi:config NVS flash: enabled
I (630) wifi:config nano formating: disabled
I (634) wifi:Init dynamic tx buffer num: 64
I (638) wifi:Init data frame dynamic rx buffer num: 64
I (643) wifi:Init management frame dynamic rx buffer num: 64
I (648) wifi:Init management short buffer num: 32
I (653) wifi:Init static rx buffer size: 1600
I (657) wifi:Init static rx buffer num: 16
I (661) wifi:Init dynamic rx buffer num: 64
ETHERNET_EVENT_CONNECTED
I (14776) phy: phy_version: 4181, c9a03e2, Mar 10 2020, 12:45:06, 0, 0
I (14778) wifi:mode : sta (24:0a:c4:e7:ee:70)
WIFI_EVENT_STA_START
WIFI_EVENT_STA_START
I (15506) wifi:new:<6,1>, old:<1,0>, ap:<255,255>, sta:<6,1>, prof:1
I (15979) wifi:state: init -> auth (b0)
I (15983) wifi:state: auth -> assoc (0)
I (15989) wifi:state: assoc -> run (10)
I (15999) wifi:connected with zhongjunboyan, aid = 3, channel 6, 40U, bssid = 60:83:34:85:a3:90
I (15999) wifi:security type: 3, phy: bgn, rssi: -16
I (16002) wifi:pm start, type: 1

WIFI_EVENT_STA_CONNECTED
I (16011) wifi:AP's beacon interval = 102400 us, DTIM period = 1
IP_EVENT_STA_GOT_IP
I (16582) esp_netif_handlers: sta ip: 192.168.86.125, mask: 255.255.255.0, gw: 192.168.86.1
```

TDA-2030, 2020-08-14:
When I ran this demo, I found that there was some delay, but the connection was successful and the web page was opened normally. I didn't set up IP or DHCP manually.

[image]

 You can do some checks:
- Check whether a device is added from the management panel of the wireless router
- Check if the power supply is insufficient

coollofty, 2020-08-14:
Now, there is only eth2wifi+ap mode, that is: esp32 uses ethernet to connect to the up route, and then set it to ap mode to allow other devices to access the Internet succeeded. So, unfortunately, I still did not successfully implement wifi2eth. No matter it is a direct connection or a crossover cable, the PC failed to ping the gateway *_*

TDA-2030, 2020-08-17:
@coollofty  What I mentioned above is the eth2wifi STA mode you want to implement

leeebo, 2023-07-10:
To achieve wifi2eth function, please refer the new solution https://github.com/espressif/esp-iot-bridge

closed 2023-07-10 by leeebo