espressif/esp-iot-solution #89: Compatible 16bit parallel display modules

ammaree, 2020-08-06:
Can somebody please point me to specific 16bit parallel interface display modules that can be used with the ESP32-DevKitC-VC+ESP32-LCDKit combination.
We prefer modules that are direct 1-for-1 pin compatible with 2x20 pin connector on the LCDKit.
Resolution in range 320x240 to 800x480 would be ideal.

We have searched Banggood, Amazon and Adafruit with no luck so far.

Thanks

TDA-2030, 2020-08-14:
@ammaree 
The screen model number used in the example using 16bit parallel interface screen under `examples/hmi` is NT35510. Maybe you can find the module you want from this model number.

ammaree, 2020-08-14:
@TDA-2030 
I have searched all over for NT35510, found many modules, but not 1 that is pin-compatible. What we are trying to avoid is a bunch of loose cables causing timing and reliability problems.

If you can point me towards any pin compatible modules that would be greatly appreciated.

TDA-2030, 2020-08-17:
I used this https://item.taobao.com/item.htm?spm=a1z0d.6639537.1997196601.3.106e7484FH0Sts&id=560872117142

[image]

Unfortunately, it's on Taobao. I'm not sure you can buy it from this link

ammaree, 2020-08-17:
Thanks for your feedback, much appreciated.
But, as you predicted, I have been unable to make any sense of Taobao, no English language version available.
Have also tried all the different  number being AC1568/9 and TK043F1508 but still no luck.
Maybe I should give up on the LCDKit as a a solution and look for something else?

TDA-2030, 2020-08-18:
Maybe you can design your own adapter board. I think the easiest way is to connect them with wires. After all, the goal is to make the screen display.

ammaree, 2021-04-08:
@TDA-2030 

We have at last been able to procure a number of displays using the English language interface of tbfocus.com using the following link:
[https://buy.tbfocus.com/item.php?id=560872117142#3512775208513](url)

The modules received are very similar to the one you linked to, except for the following differences:
Yours         Ours
NC             MISO       Probably not a problem, SPI slave/display to master/ESP32 communications
SDA           TCS         I read as I2C SDA, no idea what TCS indicates
NC             SCK        I2C SCK I assume, need to cut/strap or GPIO remap ?  
SCK           TE           TE I have no idea, but my boards also have a solder strap option for TE

Since I have no documentation of any form I have little idea what the implications are, if anybody can help with documentation and/or advise on the changes required, it will be much appreciated.

Thanks

André

Non matching pins:
[image]

TE jumper:
[image]

leeebo, 2023-07-10:
Hi @ammaree, From ESP-IDF v5.0, ESP-IDF has a new [esp_lcd](https://github.com/espressif/esp-idf/tree/master/components/esp_lcd) component which supports both 8080/RGB/I2S/SPI interface and is well [documented](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/lcd.html), we recommend using it in your new projects.

closed 2023-07-10 by leeebo