espressif/esp-iot-solution #90: touch_pad_evb example and EST_Tuning Tool.

cadrjr, 2020-08-18:
Hi,
I'm using IDF 4.01. and esp-iot-solution V1.1.
I can build and run the touch_pad_evb example.

I now want to use the ESP-Tuning tool, which V1.1 is supposed to support.
I run idf.py menuconfig but the menuconfig menu does not contain any of the Tuning-related menu entries.

https://github.com/espressif/esp-iot-solution/blob/release/v1.1/documents/touch_pad_solution/esp_tuning_tool_user_guide_en.md says I should see the following menu: "IoT Solution settings > IoT Components Management > General functions > Touch Sensor" but I do not.

I presume this is some kind of Kconfig error ?
Any help welcome. I'm a noobie and it's very frustrating when examples don't work.....

alic-maker, 2020-08-18:
I found that there is real no such configuration item, and I will fix it as soon as possible.
Thank you for your feedback!

alic-maker, 2020-08-18:
The configuration you want to use has been added to `Component config > Touch Sensor`

cadrjr, 2020-08-19:
That works great!. Thank you so much.

closed 2020-08-19 by cadrjr