espressif/esp-iot-solution #93: Button create bug

Lisa999, 2020-09-26:
> use esp timer !!!
I (976) gpio: GPIO[35]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:3 
use esp timer !!!
E (986) gpio: gpio_install_isr_service(439): GPIO isr service already installed
I (996) gpio: GPIO[34]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:3 
use esp timer !!!

Function 'iot_button_create' line 316 in file button.c: gpio_install_isr_service(0);
Shouldn't call this function for the 2nd (or more) button create!

leeebo, 2021-02-20:
A new button framework is added, you can try https://github.com/espressif/esp-iot-solution/tree/master/components/button

Lisa999, 2021-02-22:
New button framework is working great!

closed 2021-02-22 by Lisa999