espressif/esp-iot-solution #94: CMake Error at CMakeLists.txt:20 (register_component)

exrector, 2020-11-15:
`Running cmake in directory /Users/exrector/esp-iot-solution/components/motor/servo/servo_motor/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 --warn-uninitialized -DIDF_TARGET=esp32 -DCCACHE_ENABLE=0 /Users/exrector/esp-iot-solution/components/motor/servo/servo_motor"...
Warn about uninitialized values.
CMake Warning (dev) in CMakeLists.txt:
  No project() command is present.  The top-level CMakeLists.txt file must
  contain a literal, direct call to the project() command.  Add a line of
  code such as

    project(ProjectName)

  near the top of the file, but after cmake_minimum_required().

  CMake is pretending there is a "project(Project)" command on the first
  line.
This warning is for project developers.  Use -Wno-dev to suppress it.

-- The C compiler identification is AppleClang 12.0.0.12000032
-- The CXX compiler identification is AppleClang 12.0.0.12000032
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /Library/Developer/CommandLineTools/usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /Library/Developer/CommandLineTools/usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
CMake Error at CMakeLists.txt:20 (register_component):
  Unknown CMake command "register_component".

CMake Warning (dev) in CMakeLists.txt:
  No cmake_minimum_required command is present.  A line of code such as

    cmake_minimum_required(VERSION 3.18)

  should be added at the top of the file.  The version specified may be lower
  if you wish to support older CMake versions for this project.  For more
  information run "cmake --help-policy CMP0000".
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Configuring incomplete, errors occurred!
See also "/Users/exrector/esp-iot-solution/components/motor/servo/servo_motor/build/CMakeFiles/CMakeOutput.log".
See also "/Users/exrector/esp-iot-solution/components/motor/servo/servo_motor/build/CMakeFiles/CMakeError.log".
cmake failed with exit code 1`

TDA-2030, 2020-11-17:
Which branch are you using？

exrector, 2020-11-18:
> Какую ветку вы используете？

1.1

TDA-2030, 2020-11-19:
@exrector Which version of esp-idf are you using？The esp-iot-solution v1.1 should be compiled using esp-idf v4.0.1

exrector, 2020-11-21:
I used idf4.1. Is it not working with esp-iot-solution v1.1 ?

> @exrector Какую версию esp-idf вы используете？ esp-iot-solution v1.1 должен быть скомпилирован с использованием esp-idf v4.0.1.
> Как вы использовали компонент servo_motor? Как происходит процесс компиляции и какие команды используются？

@TDA-2030 
I used idf4.1. Is it not working with esp-iot-solution v1.1 ?

exrector, 2020-11-21:
@TDA-2030 
simple /Users/exrector/esp-iot-solution/examples/hmi/lvgl_example

MacExrector:lvgl_example Exrector$ idf.py set-target esp32
Adding "set-target"'s dependency "fullclean" to list of commands with default set of options.
Executing action: fullclean
Directory '/Users/exrector/esp-iot-solution/examples/hmi/lvgl_example/build' doesn't seem to be a CMake build directory. Refusing to automatically delete files in this directory. Delete the directory manually to 'clean' it.
MacExrector:lvgl_example Exrector$ idf.py fullclean
Executing action: fullclean
Build directory '/Users/exrector/esp-iot-solution/examples/hmi/lvgl_example/build' not found. Nothing to clean.
Done
MacExrector:lvgl_example Exrector$ idf.py set-target esp32
Executing action: set-target
Set Target to: esp32, new sdkconfig created. Existing sdkconfig renamed to sdkconfig.old.
Running cmake in directory /Users/exrector/esp-iot-solution/examples/hmi/lvgl_example/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 --warn-uninitialized -DIDF_TARGET=esp32 -DCCACHE_ENABLE=0 /Users/exrector/esp-iot-solution/examples/hmi/lvgl_example"...
Warn about uninitialized values.
CMake Error at CMakeLists.txt:15 (include):
  include could not find load file:

    /esp/esp-idf/CMakeLists.txt

-- Found Git: /usr/local/bin/git (found version "2.28.0") 
-- Unexpected file in components directory: /Users/exrector/esp/esp-idf/components/.DS_Store
-- The C compiler identification is GNU 8.2.0
-- The CXX compiler identification is GNU 8.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /Users/exrector/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /Users/exrector/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /Users/exrector/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Project version: 4f5689f-dirty
-- Building ESP-IDF components for target esp32
Loading defaults file /Users/exrector/esp-iot-solution/examples/hmi/lvgl_example/sdkconfig.defaults...
/Users/exrector/esp-iot-solution/examples/hmi/lvgl_example/sdkconfig.defaults:102 CONFIG_TIMER_TASK_STACK_DEPTH was replaced with CONFIG_FREERTOS_TIMER_TASK_STACK_DEPTH
/var/folders/vv/rxlmj03s5t91qb238wqsshd40000gn/T/confgen_tmpHbSor8:39 line was updated to CONFIG_BIT_MODE_8BIT=n
/var/folders/vv/rxlmj03s5t91qb238wqsshd40000gn/T/confgen_tmpHbSor8:45 line was updated to CONFIG_UGFX_GUI_ENABLE=n
/var/folders/vv/rxlmj03s5t91qb238wqsshd40000gn/T/confgen_tmpHbSor8:46 line was updated to CONFIG_SELECT_NONE=n
/var/folders/vv/rxlmj03s5t91qb238wqsshd40000gn/T/confgen_tmpHbSor8:51 line was updated to CONFIG_LVGL_USE_CUSTOM_DRIVER=n
/var/folders/vv/rxlmj03s5t91qb238wqsshd40000gn/T/confgen_tmpHbSor8:55 line was updated to CONFIG_LVGL_LCD_DRIVER_API_MODE=n
/var/folders/vv/rxlmj03s5t91qb238wqsshd40000gn/T/confgen_tmpHbSor8:57 line was updated to CONFIG_LVGL_DRIVER_DOUBLE_BUFFER_ENABLE=n
/var/folders/vv/rxlmj03s5t91qb238wqsshd40000gn/T/confgen_tmpHbSor8:58 line was updated to CONFIG_LVGL_DISP_ROTATE_0=n
/var/folders/vv/rxlmj03s5t91qb238wqsshd40000gn/T/confgen_tmpHbSor8:60 line was updated to CONFIG_LVGL_DISP_ROTATE_180=n
/var/folders/vv/rxlmj03s5t91qb238wqsshd40000gn/T/confgen_tmpHbSor8:61 line was updated to CONFIG_LVGL_DISP_ROTATE_270=n
/var/folders/vv/rxlmj03s5t91qb238wqsshd40000gn/T/confgen_tmpHbSor8:67 line was updated to CONFIG_LVGL_DRIVER_TOUCH_FT5X06=n
/var/folders/vv/rxlmj03s5t91qb238wqsshd40000gn/T/confgen_tmpHbSor8:74 line was updated to CONFIG_LVGL_DRIVER_ILI9341=n
/var/folders/vv/rxlmj03s5t91qb238wqsshd40000gn/T/confgen_tmpHbSor8:75 line was updated to CONFIG_LVGL_DRIVER_SSD1306=n
/var/folders/vv/rxlmj03s5t91qb238wqsshd40000gn/T/confgen_tmpHbSor8:77 line was updated to CONFIG_LVGL_DRIVER_NT35510=n
/var/folders/vv/rxlmj03s5t91qb238wqsshd40000gn/T/confgen_tmpHbSor8:83 line was updated to CONFIG_LVGL_LCD_VSPI=n
-- Found PythonInterp: /Users/exrector/.espressif/python_env/idf4.1_py2.7_env/bin/python (found version "2.7.18") 
-- Found Perl: /usr/bin/perl (found version "5.28.2") 
-- Adding linker script /Users/exrector/esp-iot-solution/examples/hmi/lvgl_example/build/esp-idf/esp32/esp32_out.ld
-- Adding linker script /Users/exrector/esp/esp-idf/components/esp32/ld/esp32.project.ld.in
-- Adding linker script /Users/exrector/esp/esp-idf/components/esp32/ld/esp32.peripherals.ld
-- Adding linker script /Users/exrector/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-time.ld
-- Adding linker script /Users/exrector/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script /Users/exrector/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script /Users/exrector/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld
-- Adding linker script /Users/exrector/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld
-- Adding linker script /Users/exrector/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
[... cut ...] ctor/esp/esp-idf/components/unity /Users/exrector/esp/esp-idf/components/vfs /Users/exrector/esp/esp-idf/components/wear_levelling /Users/exrector/esp/esp-idf/components/wifi_provisioning /Users/exrector/esp/esp-idf/components/wpa_supplicant /Users/exrector/esp/esp-idf/components/xtensa
-- Configuring incomplete, errors occurred!
See also "/Users/exrector/esp-iot-solution/examples/hmi/lvgl_example/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1
MacExrector:lvgl_example Exrector$

closed 2020-11-25 by exrector