espressif/esp-iot-solution #99: ESP32-S2 Touch Sensor Application Note 404s

lowtolerance, 2021-02-25:
Every link to the “Touch Sensor Application Note” for the esp32-s2 is 404ing now

leeebo, 2021-02-26:
@lowtolerance Hi you can find the history code and docs in an archived branch [release/v1.0](https://github.com/espressif/esp-iot-solution/tree/release/v1.0) 

For the invalid URL you can simply modify `blob/master` to `tree/release/v1.0` to fix.

https://github.com/espressif/esp-iot-solution/tree/release/v1.0/documents/touch_pad_solution/touch_sensor_design_en.md

lowtolerance, 2021-02-26:
That link points to the touch sensor application note for the esp32, not the esp-s2, which has a significantly different touch sensor implementation. I’m certain I’ve seen this document before, and it is referenced in the esp32-s2 data sheet, hardware reference and I believe the API reference as well.

leeebo, 2021-03-01:
@lowtolerance Thanks for the report. We have notified internally of the relevant changes.

closed 2023-07-10 by leeebo